micro-ROS/micro_ros_arduino #1067: Long delay on fini functions

v-lopez, 2022-07-04:
## Issue template

- Hardware description: Teensy4
- RTOS: Arduino
- Version or commit hash: galactic 2.0.2

#### Steps to reproduce the issue
Start with https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L104

Change the last lines to a blinking led every 100ms for instance with:
```
 static bool on = false;
  if (!on) {
    digitalWrite(LED_PIN, 255);
    on = true;
  } else {
    digitalWrite(LED_PIN, 0);
    on = false;
  }
```

Run the program, the led blinks, (not at the specified rate due to the `rmw_uros_ping_agent` but that's kind of expected).
Launch an agent, the led blinks at the specified rate.
Kill the agent, the led stops blinking for over 10 seconds.

#### Expected behavior
The led keeps blinking at the specified rate + the delays introduced by the  `rmw_uros_ping_agent`.

#### Actual behavior
Extremely long delay.

#### Additional information
On the application where I am facing this, with the debugger, I found it gets stuck here:
```
#0  usb_serial_read (buffer=buffer@entry=0x20009974 <custom_sessions+532>, size=size@entry=5)
    at ~/arduino-1.8.15/hardware/teensy/avr/cores/teensy4/usb_serial.c:215
#1  0x00003814 in usb_serial_class::readBytes (length=5, buffer=0x20009974 <custom_sessions+532> "\201", this=<optimized out>)
    at ~/arduino-1.8.15/hardware/teensy/avr/cores/teensy4/usb_serial.h:117
#2  arduino_transport_read (transport=<optimized out>, buf=0x20009974 <custom_sessions+532> "\201", len=5, timeout=<optimized out>, 
    errcode=0x20057d17 "") at /home/victor/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/default_transport.cpp:57
#3  0x0000efe0 in uxr_framing_read_transport ()
#4  0x0000f5ea in uxr_read_framed_msg ()
#5  0x0000d730 in recv_custom_msg ()
#6  0x0000e740 in listen_message_reliably ()
#7  0x0000e86a in uxr_run_session_until_all_status ()
#8  0x0000be4c in run_xrce_session ()
#9  0x0000b790 in rmw_destroy_subscription ()
#10 0x000099b8 in rcl_subscription_fini ()
```

This is a show stopper for me, as my application is doing time sensitive stuff that cannot be interrupted for so much time when a ping fails and the destruction objects are called.
If the connection is broken, I should not have to listen to anything (`listen_message_reliably`) to destroy the objects.

What is the proper way of doing this? I want to reconnect when the link is available, so I assume I must call fini to avoid memory leaks.

pablogs9, 2022-07-04:
You can use the timeout APIs to control this timeout: https://micro.ros.org/docs/api/rmw/#function-rmw_uros_set_context_entity_destroy_session_timeout

v-lopez, 2022-07-04:
@pablogs9 I just saw that in the example I linked it's already set to 0.

I've added the debugger to this example, and I see its waiting on:
```
#0  0x000039b6 in usb_serial_read (buffer=buffer@entry=0x2000793c <custom_sessions+532>, size=size@entry=5)
    at /home/victor/arduino-1.8.15/hardware/teensy/avr/cores/teensy4/usb_serial.c:217
#1  0x000005c4 in usb_serial_class::readBytes (length=5, buffer=0x2000793c <custom_sessions+532> "\201", this=<optimized out>)
    at /home/victor/arduino-1.8.15/hardware/teensy/avr/cores/teensy4/usb_serial.h:117
#2  arduino_transport_read (transport=<optimized out>, buf=0x2000793c <custom_sessions+532> "\201", len=5, timeout=<optimized out>, 
    errcode=0x20067ea7 "") at /home/victor/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/default_transport.cpp:57
#3  0x0000a868 in uxr_framing_read_transport ()
#4  0x0000ae72 in uxr_read_framed_msg ()
#5  0x00009038 in recv_custom_msg ()
#6  0x00009e20 in wait_session_status ()
#7  0x00009e98 in uxr_delete_session_retries ()
#8  0x00006fa4 in rmw_shutdown ()
```

pablogs9, 2022-07-04:
https://github.com/micro-ROS/micro_ros_arduino/blob/fda891038935f8425a980b08299024a90243a301/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L45-L48

pablogs9, 2022-07-04:
In any case, I'm just testing the default configuration of the library and I have no delay. The onboard LED turns off immediately when no agent is available and turns on immediately when an agent appears.

Could you share your code?

v-lopez, 2022-07-04:
Sure.

[urostest.ino.txt](https://github.com/micro-ROS/micro_ros_arduino/files/9040334/urostest.ino.txt)

v-lopez, 2022-07-04:
Related question, the `rmw_uros_ping_agent` ping takes way more time than what is passed through parameters the first time you run the application. 
To me, pings take over 1 second to return if an agent has not been available since the Teensy booted.
Once the agent is online and connected, ping is very fast obviously.
But if I kill the agent and the ping fails, the ping fails very fast, not taking 1 second like the first time you run the application.

pablogs9, 2022-07-05:
Can you try the example recompiling the library with the proposed parameters (`RMW_UXRCE_ENTITY_CREATION_DESTROY_TIMEOUT=0` and `UCLIENT_MAX_SESSION_CONNECTION_ATTEMPTS=0`)?

v-lopez, 2022-07-05:
> Can you try the example recompiling the library with the proposed parameters (`RMW_UXRCE_ENTITY_CREATION_DESTROY_TIMEOUT=0` and `UCLIENT_MAX_SESSION_CONNECTION_ATTEMPTS=0`)?

With this parameter, I still have the long ping until the first connection. (Debugger indicates waiting at `uxr_init_custom_transport`).
After the first connection, the ping is fast.

v-lopez, 2022-07-05:
Yes you are right. 
If I remove all micro ros code and leave my loop with:
```
void loop() {
  Serial.begin(115200);
 static bool on = false;
  if (!on) {
    digitalWrite(LED_PIN, 255);
    on = true;
  } else {
    digitalWrite(LED_PIN, 0);
    on = false;
  }
  delay(100);
}
```

I see huge delays as well.
I'll see how to modify the source and build them with the docker with the static bool you mention.

v-lopez, 2022-07-05:
From teensy documentation:
> Serial.begin()
Serial.begin() is optional on Teensy. USB hardware initialization is performed before setup() runs.

> The baud rate input is ignored, and only used for Arduino compatibility. USB serial communication always occurs at full USB speed. However, see the baud() function below.

> Serial.begin() may wait up to 2.5 seconds for USB serial communication to be ready. For fastest program startup, simply do not use this unnecessary function. Its only purpose is for compatibility with programs written for Arduino. The delay is intended for programs which do not test the Serial boolean. 

I may remove this altogether.

Thanks for the help @pablogs9