micro-ROS/micro_ros_arduino #1072: Cannot ping agent from client using Ethernet UDP transport

adityakamath, 2022-07-09:
I've been trying to ping the micro-ROS agent from my Arduino Portenta over Ethernet using rmw_uros_ping_agent() as described [here](https://micro.ros.org/docs/tutorials/programming_rcl_rclc/micro-ROS/). Here's the code snippet:

`
rmw_ret_t ping_result = rmw_uros_ping_agent(timeout_ms, attempts);

if (RMW_RET_OK == ping_result)
{
    // micro-ROS Agent is reachable
    ...
}
else
{
    // micro-ROS Agent is not available
    ...
}
`

the ping_result is never OK and always enters the else condition. I only see this issue with the Ethernet (UDP) transport, it works as expected with Serial and Wifi (UDP) transports. Has anyone tried this with the Ethernet transport (with or without the Portenta)? I'm not sure if it is a SW issue or something wrong with my network configuration. Pub/Sub and services work over ethernet, just the ping does not.

- Hardware description: Arduino Portenta H7
- RTOS: -
- Installation type: micro_ros_arduino, and this [reconnection example](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino)
- Version or commit hash: galactic

#### Steps to reproduce the issue
[This Reconnection example](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino). The ```set_microros_transports()``` in the setup is replaced with this:

```
byte arduino_mac[] = {0x01, 0xAB, 0x23, 0xCD, 0x45, 0xEF};
IPAddress arduino_ip(192, 168, 1, 109);
IPAddress agent_ip(192, 168, 1, 100);
set_microros_native_ethernet_udp_transports(arduino_mac, arduino_ip, agent_ip, 8888);
```

#### Expected behavior
Solid LED when the ping is successful to indicate that the agent is connected

#### Actual behavior
Flashing LED (error loop) as the ping function does not return OK

#### Additional information
I've been meaning to try this with the Teensy 4.1 + Ethernet to see if that works, but that transport is [not officially tested/supported yet](https://github.com/micro-ROS/micro_ros_arduino/issues/1065).

pablogs9, 2022-07-11:
Hello @adityakamath could you make sure that you have the very latest version of the `galactic` branch of this repo? Also, could you make sure that you are using the latest version of the micro-ROS Agent? If you are using a dockerized agent please make a `docker pull ...`

adityakamath, 2022-07-11:
I'm using the latest release (2.0.5), but I will try the latest version of this repo when I get the chance. What change do you think causes this issue?

pablogs9, 2022-07-11:
Make sure that you have an updated agent. If you have a very old version we have found some incompatibilities with the ping feature.

Jenifen, 2022-07-18:
same issue with me, i'm using Portenta H7 also, any update about this ?

Acuadros95, 2022-07-18:
@Jenifen, any chance you can debug this as the transport contributor?

Come ideas:
- `rmw_uros_ping_agent` will open and close the transport if no session is active, check if this methods are failing.
- Does the ping work if the Agent is available on the first iteration?
- Check if there is data exchange (wireshark, ...) and compare it with the working Wifi or Teensy Ethernet packets.

Jenifen, 2022-07-18:
No problem on the first iteration, 
how can i debug **rmw_uros_ping_agent**, and how and where can i modified https://github.com/micro-ROS/rmw_microxrcedds/blob/humble/rmw_microxrcedds_c/src/rmw_microros/ping.c ?

Acuadros95, 2022-07-19:
Dont worry about the behavior of `rmw_uros_ping_agent`. Check the return of the transport methods, focusing on the open and close functions: `arduino_native_ethernet_udp_transport_open` and `arduino_native_ethernet_udp_transport_close`.

They should succeed each time, otherwise the ping wont even try to send a message.

Acuadros95, 2022-07-20:
@adityakamath, could you test this PR modifications on a Teensy 4.1: https://github.com/micro-ROS/micro_ros_arduino/pull/1094?

adityakamath, 2022-07-20:
@Acuadros95, it works with Teensy 4.1 + Ethernet. Tested it with the reconnection example. Also works for the Arduino Portenta with Ethernet as expected. Thanks for the fix @Jenifen!

Acuadros95, 2022-07-20:
The fix just got merged: https://github.com/micro-ROS/micro_ros_arduino/pull/1094

Thanks @adityakamath and @Jenifen, closing!

closed 2022-07-20 by Acuadros95