micro-ROS/micro_ros_arduino #1090: ROS2 message memory allocations -- via micro_ros_utilities_create_*_message_memory() APIs -- seemingly causing data corruption (buffer overflow?)

alikureishy, 2022-07-19:
## Issue template
I am encapsulating the microros rcl_service_t object within a C++ wrapper class, and I utilize micro_ros_utilities_create_static_message_memory() (and/or micro_ros_utilities_create_message_memory()) to setup the message memory etc within the object's constructor.

Strangely, these allocation calls seem to be corrupting the memory of contiguous instance variables after the message field.

- Hardware description: esp32
- RTOS: arduino
- Installation type: git clone
- Version or commit hash: humble

#### Steps to reproduce the issue
My code is a bit more complex, so I didn't prune all that code yet. But I was able to narrow it down to the scenario described in the snippets below. It will not compile, but hoping it might provide some rough clues as a starting point. If this is insufficient and you need a more specific repro, please let me know and I'll put one together.

***Service.h***
```
template <class RequestType, class ResponseType> 
class ServiceWrapper<RequestType, ResponseType> 
{ 
public:    
   ServiceWrapper(
         const std::string& name, 
         rcl_node_t* node, 
         rosidl_service_type_support_t * service_type_support, 
         rosidl_message_type_support_t * request_type_support, 
         rosidl_message_type_support_t * response_type_support, 
         ServiceCallback callback) 
  : name(name) {         // <------- Name argument is copied over via this member-initialization-list

       RCCHECK(rclc_service_init_default(&this->service_data, node, service_type_support, name.c_str()));

       Serial.printf("'Name: %s'\n", this->name.c_str());               // <----------- Prints is as expected

       // ~~~~~~~ TROUBLE CAUSING CODE (either or both calls) ~~~~~~~~~~~
       assert(micro_ros_utilities_create_message_memory(
                        request_type_support,
                        &this->request_msg, 
                        micro_ros_utilities_memory_conf_default));
       assert(micro_ros_utilities_create_message_memory(
                        response_type_support,
                        &this->response_msg, 
                        micro_ros_utilities_memory_conf_default));

       Serial.printf("'Name: %s'\n", this->name.c_str());              // <------------- The name variable gets corrupted by now
   }

private:
   RequestType    request_msg;    
   ResponseType  response_msg;
   std::string          name;              // <--------- Gets corrupted
   rcl_service_t     service_data; 
};
```

The Request/Response types should probably be something complex, such as:

***RequestType***
```
string<=150             name
string<=64[<=20]        some_other_strings
float64[<=20]           values
```

***ResponseType***
```
string<=250            status
```

***ServiceType***
```
RequestType          request
---
ResponseType       response
```

***Calling code:***
```
ServiceWrapper<RequestType, ResponseType> service(
      std::string("TestName"),
      node,
      ROSIDL_GET_SVC_TYPE_SUPPORT(my_package, srv, ServiceType),
      ROSIDL_GET_MSG_TYPE_SUPPORT(my_package, srv, RequestType),
      ROSIDL_GET_MSG_TYPE_SUPPORT(my_package, srv, ResponseType),
      some_callback
);
```

#### Expected behavior

The name should remain the same between the 1st and 2nd Serial.print() invocations:

```
Name: 'TestName'
Name: 'TestName'
```

#### Actual behavior

The 1st invocation of name output:
```
Name: 'TestName'
Name: ''                     <======== EMPTY STRING
```

#### Additional information

If I pad the ```request_msg``` and ```response_msg``` fields with sufficiently sized message buffers, as below:
```
   RequestType        request_msg;
   uint8_t            __request_overflow_buffer__[128];    <== Haven't yet investigated the precise length required.
   ResponseType       response_msg;
   uint8_t            __response_overflow_buffer__[128];   <== Haven't yet investigated the precise length required.
   std::string        name;              
   rcl_service_t      service_data;   
```

... the ```name``` field does *not* get corrupted, suggesting there is some sort of buffer overflow happening. Note that the padding lengths attempted above are arbitrary ... I didn't narrow it down to an exact amount yet, but can do so if needed.

pablogs9, 2022-07-19:
I'm going to take a look

pablogs9, 2022-07-19:
Hello @alikureishy can you check your use case with this branch: https://github.com/micro-ROS/micro_ros_utilities/pull/48?

alikureishy, 2022-07-20:
Thanks @pablogs9 ! Glad the message definition helped!
I will try it out shortly. Had to switch over to something else at the moment, but am looking forward to getting back to verifying this asap this week. Will post an update then.

pablogs9, 2022-07-20:
As soon as you test the PR, we will merge

alikureishy, 2022-07-21:
@pablogs9 , since the change is in the ```micro_ros_utilities``` repository, how do I include those into the static libmicroros generated by ```micro_ros_arduino```? The former does not seem to be a *submodule* of the latter either. Is there a ```micro_ros_arduino``` commit/branch that I could pull that would include those references/fixes internally?

pablogs9, 2022-07-22:
You can do a quick test by doing a checkout of the branch [here](https://github.com/micro-ROS/micro_ros_arduino/blob/c15853d2580c6f1a2ad77abf426e4fb6c7a3e1c2/extras/library_generation/library_generation.sh#L47) and [rebuilding the library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library).

Genozen, 2022-07-22:
Hi guys, I'm working on this as well. Just a side question I'm getting:
`src/main.cpp:8:10: fatal error: micro_ros_utilities/type_utilities.h: No such file or directory
 #include <micro_ros_utilities/type_utilities.h>`

Built using the MicroROS platformIO version, where can I find this library?

alikureishy, 2022-07-22:
> You can do a quick test by doing a checkout of the branch [...]

I understand the part about rebuilding the microros static library. However, I don't follow the first part of your comment. Are you suggesting that the script should clone+copy the code from the ```micro_ros_utilities``` repo and ***overwrite*** the same code files (```string_utilities.h``` and ```type_utilities.h```) under ```micro_ros_arduino/src/micro_ros_utilities``` (before doing the libmicroros build)? Or should this repo be cloned fully under the ```micro_ros_arduino/extras/library_generation/extra_packages``` folder instead (before the libmicroros build)?

pablogs9, 2022-07-26:
@Genozen which distro of micro-ROS are you using?

@alikureishy just replace the whole folder for this test.

alikureishy, 2022-07-27:
Thanks, @pablogs9, for your prompt responses. Since I am not familiar with the internals of micro-ros builds, I am not yet clear on these steps. In the interest of avoiding additional back-and-forth here, and more moving variables -- and if it isn't too much trouble for you -- would you mind providing a temporary verification branch on the ```micro_ros_arduino``` repository directly, with the ```library_generation.sh``` changes needed to pull all the modified dependencies?

pablogs9, 2022-07-28:
Hello @alikureishy I have rebuilt the library for you here: https://github.com/micro-ROS/micro_ros_arduino/tree/test/1090

Only for ESP32, as you can see [here](https://github.com/micro-ROS/micro_ros_arduino/commit/252b45f37ee38a32b7bc312ec5ce0483adc9a4ce):
- `micro_ros_utilities` in `feature/update_bounded`
- All target libs removed
- ESP32 library updated

Please check out this branch and report if this solves your issue.

pablogs9, 2022-08-04:
Any update on this?

Genozen, 2022-08-04:
> 

Sry for the late reply. I was using Linux 20.04 Foxy + PlatformIO with Arduino Framework ESP32.

pablogs9, 2022-08-04:
@Genozen open a new issue in the repo you are using.

@alikureishy any update on this?

alikureishy, 2022-08-05:
Sorry for the delay in responding, @pablogs9 . And, many thanks for the special branch for verifying the fix! It was invaluable.

The fix appears to be working *for the use case I had mentioned above*! However, I have a follow-up question. I've read the [tutorial about memory management](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/) for micro-ros, but am trying to better understand how the ***contiguous*** memory buffer is ***subdivided*** for storing the message tree (including nesting of complex types etc), during allocation by ```micro_ros_utilities_create_message_memory() ```. Is there any place other than the code, where this is described in more detail (perhaps even visually)? I ask because I might be facing another buffer overflow error, but before I log an issue I want to make sure I'm navigating across the allocated memory properly.

pablogs9, 2022-08-05:
Hello @alikureishy, `micro_ros_utilities_create_message_memory` iterates recursively through the type and allocates memory when it discovers a sequence, a string, or a type that requires memory. It is kind of difficult to follow the execution sequence, but step by step debugging can be a solution.

I'm merging https://github.com/micro-ROS/micro_ros_utilities/pull/48 and closing this issue due it solves your initial issue. **But please open a new one if you found a corner case where `micro_ros_utilities` misbehaves.**

closed 2022-08-05 by pablogs9