micro-ROS/micro_ros_arduino #1091: rcl_executor_spin_some() returns error code 1 on microros service server during CLI invocation

alikureishy, 2022-07-19:
I am running a custom ROS2 service on an ESP32 using the micro_ros_arduino library. The service shows up in the shell via ```ros2 service list``` as well. On the micro-ros side, I am initializing the service and message data objects using a memory buffer via the ```micro_ros_utilities_create_static_message_memory()``` API. This might be a side-effect of issue #1090 but I am not sure, hence this bug report.

- Hardware description: esp32
- RTOS: arduino
- Installation type: git clone + docker build
- Version or commit hash: humble

#### Actual behavior

When I invoke this service from the CLI, I see this initial output on the shell:
```
requester: making request: my_custom_svc.srv.MyRead_Request(queries=> [my_custom_svc.msg.MyReadQuery(name='something', aspects=['something-else'])])
```
.. and then nothing. It hangs waiting for a response from the server, which never returns.

On the server, I get an error return code (```value 1```) from the ```rcl_executor_spin_some()``` invocation.

At the exact same time, here's what I see as the output from the micro-ros agent (running with verbosity = 6):
```
[1657929970.659653] debug    | Replier.cpp        | read_fn           | [==>> DDS <<==]        | client_key: 0x00000000, len: 64, data:  0000: 01 0F 2E 32 1A A6 5D 76 01 00 00 00 00 00 12 03 00 00 00 00 01 00 00 00 01 00 00 00 0C 00 00 00 0020: 75 6C 6E 61 72 5F 6A 6F 69 6E 74 00 01 00 00 00 09 00 00 00 70 6F 73 69 74 69 6F 6E 00 00 00 00

[1657929970.659886] debug    | UDPv4AgentLinux.cpp | send_message     | [** <<UDP>> **]        | client_key: 0x196442D4, len: 76, data:  0000: 81 80 04 00 09 01 44 00 00 0C 00 08 01 0F 2E 32 1A A6 5D 76 01 00 00 00 00 00 12 03 00 00 00 00 0020: 01 00 00 00 01 00 00 00 0C 00 00 00 75 6C 6E 61 72 5F 6A 6F 69 6E 74 00 01 00 00 00 09 00 00 00 0040: 70 6F 73 69 74 69 6F 6E 00 00 00 00 

[1657929970.686920] debug    | UDPv4AgentLinux.cpp | recv_message     | [==>> UDP <<==]        | client_key: 0x196442D4, len: 13, data:  0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
```

#### Additional information

- What are some general causes for error code # 1 from this method?
- Is there any tool I could use to interpret the hex data from the agent output above?
- On looking at the ```rcl_executor_spin_some()``` code, there are a lot of ```PRINT_RCLC_ERROR``` statements, but I don't see any error logs via the serial port. How should one go about enabling and viewing such microros error outputs via the serial port?

pablogs9, 2022-07-19:
Provide code for replicating

Acuadros95, 2022-07-19:
We should probably fix https://github.com/micro-ROS/micro_ros_arduino/issues/1090 first if its related to the problem.
Then, we will take a look if the problem persist.

alikureishy, 2022-08-07:
Closing for the time being since I haven't been able to reproduce this since #1090 was fixed. Will re-open if that changes.

closed 2022-08-07 by alikureishy