micro-ROS/micro_ros_arduino #1105: Failed to communication when I changed serial port baudrate

z2010012286, 2022-07-24:
## Issue template

- Hardware description: M5 stack Atom Lite (esp32 core)
- Installation type: micro_ros_arduino
- Version or commit hash: ros2 foxy + ubuntu 20.04

#### Steps to reproduce the issue
1. Change the serial port baudrate (250000) instead of the default baudrate (115200).  The file path is : src\default_transport.cpp
[image]
2. Build and download the example code "micro_ros_publisher.ino"
3. ros2 run micro_ros_agent in the ROS2 foxy, and there is not topic visible.
4. When I used the default baudrate (115200), the topic is visible.

#### Expected behavior
Serial baudrate (115200), and the output as below.
[image]

#### Actual behavior
Serial baudrate (250000), and the output as below. I'm sure restart the esp32, but it is still incorrect. 
[image]

#### Additional information
I'm beginner for micro_ros, I'm not sure if the serial port baudrate can be modified.

Genozen, 2022-07-25:
I tried it with ESP32 Heltec LoRa v2, no issue for me on 250000 baudrate. Does your board support this? (I'm also new to microROS, just trying to see if I can help :) )

pablogs9, 2022-07-26:
Not sure, but maybe not related to micro-ROS itself. Have you tried to run a basic serial "Hello World" using this baudrate?

z2010012286, 2022-07-26:
> Not sure, but maybe not related to micro-ROS itself. Have you tried to run a basic serial "Hello World" using this baudrate?

Thanks for your reply.
I have tested the serial port communication using Windows Serial Monitor, and it works well using baudrate 250000. So I think the hardware should support this baudrate(250000).

z2010012286, 2022-07-26:
> I tried it with ESP32 Heltec LoRa v2, no issue for me on 250000 baudrate. Does your board support this? (I'm also new to microROS, just trying to see if I can help :) )

Thanks for your reply.
I am sure my board support 250000 baudrate. But the difference is that the core of my hardware is esp32-pico-d4 instead of esp32. I don't know if this diference cause the communication problem.

pablogs9, 2022-07-26:
Not sure, but test it in your target hardware before trying to run micro-ROS. I'm closing, please reopen if you find a problem in the integration between your transport and micro-ROS.

closed 2022-07-26 by pablogs9