micro-ROS/micro_ros_arduino #1107: Float32MultiArray causes Wifi: Association

Genozen, 2022-07-25:
#1062

- Hardware description: ESP32 Heltec LoRa v2
- RTOS: microROS platformIO
- Installation type: microROS platformIO
- Version or commit hash: Linux 20.04 + Foxy ROS2.

#### Steps to reproduce the issue
Similar to #1062 
Whenever I added `msg2.data.data[0] = 0.1;` causes `Wifi:Association refused temporarily, comeback time 299 mSec.` in the terminal

#### Expected behavior
stores multiple float data in the array

Genozen, 2022-07-25:
I should mention the board resets:
```
[1658775353.731866] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x02A35511, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1658775353.731987] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x02A35511, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 05 00 00
[1658775353.732062] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x02A35511, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1658775353.753863] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x02A35511, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 03 00 03 00 80
[1658775353.765599] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x02A35511, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1658775353.774613] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x02A35511, len: 16, data: 
0000: 81 01 00 00 07 01 08 00 00 0E 00 05 00 00 00 00
[1658775353.774955] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 4, data: 
0000: 00 00 00 00
```

Full Serial Terminal Message:
```
Initialized...
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d2b62  PS      : 0x00060e30  A0      : 0x80139cf8  A1      : 0x3ffb2730  
A2      : 0x3f800000  A3      : 0x00000000  A4      : 0x00150ab8  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb2710  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00ffffff  
A14     : 0x01000000  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40089648  LEND    : 0x4008965e  LCOUNT  : 0xffffffff  

Backtrace:0x400d2b5f:0x3ffb27300x40139cf5:0x3ffb2750 0x40137d4f:0x3ffb2790 0x401381f9:0x3ffb27b0 0x401382ad:0x3ffb27e0 0x400d2b8c:0x3ffb2800 0x400dd959:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
E (613) wifi:Association refused temporarily, comeback time 299 mSec
E (918) wifi:Association refused temporarily, comeback time 299 mSec
E (1222) wifi:Association refused temporarily, comeback time 299 mSec
```

pablogs9, 2022-07-26:
Is `msg2.data.data[0] = 0.1;` really related to `Wifi:Association refused temporarily, comeback time 299 mSec.`? Could you verify that removing this line everything works?

Some comments:
- There is no `msg2` variable in your code.
- `msg_array_float` is not being initialized, so  `msg_array_float.data.data[0] = 0.1;` might be a segmentation fault. Please read this tutorial: https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/

Genozen, 2022-07-27:
Thank you. The issue has been solved:
Error is: `Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.`
Solution is to initialized the fields:
```
 msg2.data.capacity = 10;
  msg2.data.data = (float*)infoBuffer;
  msg2.data.size = 10;
  infoBuffer = {0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0};
```

Then update the `infoBuffer`

See example from: [https://github.com/husarion/heat-follower-robot/blob/main/mcu/src/main.cpp](url)

closed 2022-07-27 by Genozen