micro-ROS/micro_ros_arduino #1113: Error while building microros arduino library

dsubhasish09, 2022-07-27:
I wanted to rebuild from source because I have to modify a few parameters in colcon.meta (DRMW_UXRCE_MAX_HISTORY). I followed the instructions given at https://github.com/micro-ROS/micro_ros_arduino/tree/foxy-
1) docker pull microros/micro_ros_static_library_builder:foxy
2) docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy

However my build fails with the error for most of the packages- Package 'rcutils' exports library 'dl' which couldn't be found. Please let me know if I am missing something here?

Acuadros95, 2022-07-27:
Please double check that `Package 'rcutils' exports library 'dl' which couldn't be found` is a warning and not an error, as those warnings are expected and harmless as explained on this comment: https://github.com/micro-ROS/freertos_apps/issues/37#issuecomment-727769065

If your build still fails, copy the actual build log here so we can check the actual error.

dsubhasish09, 2022-07-27:
Here is my terminal output-
$ docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p teensy36
dos2unix: converting file /project/extras/library_generation/library_generation.sh to Unix format...
Get:1 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:3 http://packages.ros.org/ros2/ubuntu focal InRelease [4679 B]
Get:4 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:5 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
Get:6 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Get:7 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:8 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Get:9 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [1093 kB]
Get:10 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:11 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [885 kB]
Get:12 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1164 kB]
Get:13 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [27.5 kB]
Get:14 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [30.2 kB]
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [2484 kB]
Get:16 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [2035 kB]
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1513 kB]
Get:18 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [1401 kB]
Get:19 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [54.2 kB]
Get:20 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [27.1 kB]
Fetched 24.1 MB in 3s (8967 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
36 packages can be upgraded. Run 'apt list --upgradable' to see them.
Creating firmware for generate_lib platform generic
.......
=== ./ament/ament_cmake (git) ===
Cloning into '.'...
=== ./ament/ament_index (git) ===
Cloning into '.'...
=== ./ament/ament_lint (git) ===
Cloning into '.'...
=== ./ament/ament_package (git) ===
Cloning into '.'...
=== ./ament/googletest (git) ===
Cloning into '.'...
=== ./ament/uncrustify_vendor (git) ===
Cloning into '.'...
=== ./ros2/ament_cmake_ros (git) ===
Cloning into '.'...
List of repositories is empty
#All required rosdeps installed successfully
..............
=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
Cloning into '.'...
=== ./ros2/rcl_interfaces (git) ===
Cloning into '.'...
=== ./ros2/rcl_logging (git) ===
Cloning into '.'...
=== ./ros2/rcpputils (git) ===
Cloning into '.'...
=== ./ros2/rmw (git) ===
Cloning into '.'...
=== ./ros2/rmw_implementation (git) ===
Cloning into '.'...
=== ./ros2/rosidl (git) ===
Cloning into '.'...
=== ./ros2/rosidl_dds (git) ===
Cloning into '.'...
=== ./ros2/rosidl_defaults (git) ===
[... 979 lines cut ...]
Starting >>> stereo_msgs
Starting >>> sensor_msgs_py
Finished <<< sensor_msgs_py [1.26s]
--- stderr: stereo_msgs
CMake Warning at /uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rmw/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rmw/cmake/rmwConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_microxrcedds_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_microxrcedds_c/cmake/rosidl_typesupport_microxrcedds_cConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

---
Finished <<< stereo_msgs [3.31s]
Starting >>> common_interfaces
Finished <<< common_interfaces [0.57s]

Summary: 63 packages finished [1min 38s]
  40 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces control_msgs diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_noop rclc rclc_lifecycle rclc_parameter rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs tf2_msgs trajectory_msgs unique_identifier_msgs visualization_msgs

dsubhasish09, 2022-07-27:
Here is the terminal output-
`$ docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p teensy36
dos2unix: converting file /project/extras/library_generation/library_generation.sh to Unix format...
Get:1 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:3 http://packages.ros.org/ros2/ubuntu focal InRelease [4679 B]
Get:4 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:5 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
Get:6 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Get:7 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:8 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Get:9 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [1093 kB]
Get:10 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:11 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [885 kB]
Get:12 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1164 kB]
Get:13 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [27.5 kB]
Get:14 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [30.2 kB]
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [2484 kB]
Get:16 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [2035 kB]
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1513 kB]
Get:18 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [1401 kB]
Get:19 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [54.2 kB]
Get:20 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [27.1 kB]
Fetched 24.1 MB in 3s (8967 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
36 packages can be upgraded. Run 'apt list --upgradable' to see them.
Creating firmware for generate_lib platform generic
.......
=== ./ament/ament_cmake (git) ===
Cloning into '.'...
=== ./ament/ament_index (git) ===
Cloning into '.'...
=== ./ament/ament_lint (git) ===
Cloning into '.'...
=== ./ament/ament_package (git) ===
Cloning into '.'...
=== ./ament/googletest (git) ===
Cloning into '.'...
=== ./ament/uncrustify_vendor (git) ===
Cloning into '.'...
=== ./ros2/ament_cmake_ros (git) ===
Cloning into '.'...
List of repositories is empty
#All required rosdeps installed successfully
..............
=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
Cloning into '.'...
=== ./ros2/rcl_interfaces (git) ===
Cloning into '.'...
=== ./ros2/rcl_logging (git) ===
Cloning into '.'...
=== ./ros2/rcpputils (git) ===
Cloning into '.'...
=== ./ros2/rmw (git) ===
Cloning into '.'...
=== ./ros2/rmw_implementation (git) ===
Cloning into '.'...
=== ./ros2/rosidl (git) ===
Cloning into '.'...
=== ./ros2/rosidl_dds (git) ===
Cloning into '.'...
=== ./ros2/rosidl_defaults (git) ===
[... 979 lines cut ...]
Starting >>> stereo_msgs
Starting >>> sensor_msgs_py
Finished <<< sensor_msgs_py [1.26s]
--- stderr: stereo_msgs
CMake Warning at /uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rmw/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rmw/cmake/rmwConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_microxrcedds_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_microxrcedds_c/cmake/rosidl_typesupport_microxrcedds_cConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

---
Finished <<< stereo_msgs [3.31s]
Starting >>> common_interfaces
Finished <<< common_interfaces [0.57s]

Summary: 63 packages finished [1min 38s]
  40 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces control_msgs diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_noop rclc rclc_lifecycle rclc_parameter rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs tf2_msgs trajectory_msgs unique_identifier_msgs visualization_msgs`

dsubhasish09, 2022-07-27:
Here is my terminal output-
[terminal_output.txt](https://github.com/micro-ROS/micro_ros_arduino/files/9197544/terminal_output.txt)

Acuadros95, 2022-07-27:
Your build looks OK, please check that your library has been update.

Closing, reopen if library has not been updated.

closed 2022-07-27 by Acuadros95

dsubhasish09, 2022-07-28:
How to check if the library was updated. I wanted to rebuild to allow for longer Float64MultiArray to be published and / or subscribed. I tried to publish a longer message and it complains about insufficient buffer yet again.