micro-ROS/micro_ros_arduino #1114: Undefined reference to `__ctype_ptr__' (again..)

zharifzubaidi, 2022-07-28:
Hi micro-ROS team,
I really appreciate the effort you all put on making microROS possible. I have a question from issue #315. In the steps that you mentioned, can you please explain in the detail the following:

1) How can I know the local path of my desired compiler? I install gcc-arm-none-eabi using sudo apt-get from this guide https://installati.one/ubuntu/20.04/gcc-arm-none-eabi/  I got 2019 version (arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599])

2) Step 3 involves calling docker to build the precompiled library. However, I'm not using a docker. My setup is Arduino Due, Jetson Nano with Xubuntu 20.04, ROS2 Foxy, microros Foxy 2.0.5. So how do I build the library without using the docker?

Hopefully, you could help with this. Thank you.

"Hello,

As you can see on the [lib generation script,](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/library_generation.sh) we use the same compiler version as the Arduino IDE for each board.

Your board is not currently supported by the library generator, but you can modify the script to use your own gcc version:

1. Clone this repo and modify this line to: `export TOOLCHAIN_PREFIX=/toolchain/bin/arm-none-eabi-`
https://github.com/micro-ROS/micro_ros_arduino/blob/94cb206d03cfb19aa3f1404f0d9d9cbd6b7bac6d/extras/library_generation/library_generation.sh#L88

2. Set the local path of the desired compiler:
`export GCC_PATH=/local_path/gcc-arm-none-eabi-10-2020-q4-major` (Use your own local path)

3. Call the docker generator from the repo folder:
`docker run -it --rm -v ${GCC_PATH}:/toolchain -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p cortex_m3`

3. The generated library should be on `src/cortex-m3/libmicroros.a`:
```
~/workspace/micro_ros_arduino$ strings src/cortex-m3/libmicroros.a | grep "GCC: ("
$ GCC: (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
```

_Originally posted by @Acuadros95 in https://github.com/micro-ROS/micro_ros_arduino/issues/315#issuecomment-857472762_ "

pablogs9, 2022-07-29:
> So how do I build the library without using the docker?

If you do not want to use docker you have instructions for building micro-ROS libraries here: https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/

In this tutorial you will be asked for colcon.meta and toolchain.cmake files, you can use the ones [here](https://github.com/micro-ROS/micro_ros_arduino/tree/humble/extras/library_generation) for reference.

> How can I know the local path of my desired compiler? 

check [which](https://linux.die.net/man/1/which)

zharifzubaidi, 2022-07-29:
Sorry, I'm really really new to micro ros. When I git clone the repo, where should I git clone it to? Is it inside the microros_ws ? 
I'll explain what I have done but still failed.
1) First, I download the v2.0.5-foxy to my Downloads file https://github.com/micro-ROS/micro_ros_arduino/releases
2) And then, I followed @Acuadros95 Step 1 & 2 to edit my library_generation.sh . Below is what I written for Arduino Due part

```
######## Build for SAM (e.g. Arduino Due) ########
if [[ " ${PLATFORMS[@]} " =~ " cortex_m3 " ]]; then
    rm -rf firmware/build

    #export TOOLCHAIN_PREFIX=/uros_ws/gcc-arm-none-eabi-4_8-2014q1/bin/arm-none-eabi-
    export TOOLCHAIN_PREFIX=/toolchain/bin/arm-none-eabi-
    export GCC_PATH=/usr/bin/arm-none-eabi-gcc
    ros2 run micro_ros_setup build_firmware.sh /project/extras/library_generation/cortex_m3_toolchain.cmake /project/extras/library_generation/colcon_lowmem.meta

    find firmware/build/include/ -name "*.c"  -delete
    cp -R firmware/build/include/* /project/src/

    mkdir -p /project/src/cortex-m3
    cp -R firmware/build/libmicroros.a /project/src/cortex-m3/libmicroros.a
fi
```
3) Note that my arm-none-eabi-gcc is in this path /usr/bin/arm-none-eabi-gcc. I install this compiler using 
`sudo apt-get -y install gcc-arm-none-eabi`

4) I find the tutorial is a bit confusing so I just go along and use docker what Step 3 wants me to do but i came across this error during the build

```
find: cannot delete ‘/project/src/’: Directory not empty
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /project/extras/library_generation/colcon_lowmem.meta
Starting >>> tinydir_vendor
Starting >>> rosidl_adapter
Starting >>> rosidl_typesupport_interface                                         
Starting >>> microcdr                                                                                                   
--- stderr: microcdr                                                                              
CMake Error at cmake/SuperBuild.cmake:19 (enable_language):
  The CMAKE_C_COMPILER:

    /toolchain/bin/arm-none-eabi-gcc

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:54 (include)

---
Failed   <<< microcdr [0.69s, exited with code 1]
Aborted  <<< tinydir_vendor [1.14s]                                                                          
Aborted  <<< rosidl_typesupport_interface [1.74s]                                                                      
Aborted  <<< rosidl_adapter [1.97s]                            

Summary: 0 packages finished [3.55s]
  1 package failed: microcdr
  3 packages aborted: rosidl_adapter rosidl_typesupport_interface tinydir_vendor
  2 packages had stderr output: microcdr tinydir_vendor
  59 packages not processed
find: ‘firmware/build/include/’: No such file or directory
cp: cannot stat 'firmware/build/include/*': No such file or directory
cp: cannot stat 'firmware/build/libmicroros.a': No such file or directory
```

zharifzubaidi, 2022-07-29:
Hi everyone,

I re-tried again to follow the instructions. So is what I tried:
1) git clone the repo to my home folder and then switch the repo to foxy branch by using git checkout foxy after get into the repo. The repo path is now 

> /home/prostrain/micro_ros_arduino/

2) I did the modification and now this in my library_generation.sh
```
######## Build for SAM (e.g. Arduino Due) ########
if [[ " ${PLATFORMS[@]} " =~ " cortex_m3 " ]]; then
    rm -rf firmware/build

    #export TOOLCHAIN_PREFIX=/uros_ws/gcc-arm-none-eabi-4_8-2014q1/bin/arm-none-eabi-
    export TOOLCHAIN_PREFIX=/toolchain/bin/arm-none-eabi-
    ros2 run micro_ros_setup build_firmware.sh /project/extras/library_generation/cortex_m3_toolchain.cmake /project/extras/library_generation/colcon_lowmem.meta

    find firmware/build/include/ -name "*.c"  -delete
    cp -R firmware/build/include/* /project/src/

    mkdir -p /project/src/cortex-m3
    cp -R firmware/build/libmicroros.a /project/src/cortex-m3/libmicroros.a
fi
```
3) This set the local path to desired compiler is tricky. You mentioned that I should use which to check my compiler but I'm not sure if I did it right or not. Below is what I did:
`which arm-none-eabi-gcc`
and i got 
`/usr/bin/arm-none-eabi-gcc`
So I assume this is the compiler. Not really sure but I proceed to continue to do this in terminal instead of in the script as my previous post.
`export GCC_PATH=/usr/bin/arm-none-eabi-gcc`

4) Since I'm not really sure how to do the build thing, I just follow your docker instructions. Hence, I proceed do this in terminal:
`sudo docker run -it --rm -v ${GCC_PATH}:/toolchain -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p cortex_m3`

However, I got this error
`docker: Error response from daemon: error while creating mount source path '/usr/bin/arm-none-eabi-gcc': mkdir /usr/bin/arm-none-eabi-gcc: read-only file system.`

So yeah, now I'm stuck here. Please help..

zharifzubaidi, 2022-08-01:
Hi @pablogs9,

I have tried to follow your recommendations but I having some trouble here. Below is what I have done:

1) I git clone the micro_ros_arduino repo into the src folder of my microros_ws that I make by following this guide https://micro.ros.org/docs/tutorials/core/first_application_linux/ 
2) I then use git checkout foxy to switch to foxy branch because I'm using ROS2 Foxy
3) I change the library_generation.sh line 88
from:
`export TOOLCHAIN_PREFIX=/uros_ws/gcc-arm-none-eabi-4_8-2014q1/bin/arm-none-eabi-`
to:
`export TOOLCHAIN_PREFIX=/toolchain/bin/arm-none-eabi-`

4) Note that when I search for my desired compiler, I did this:
`which arm-none-eabi-gcc`
Result:
`/usr/bin/arm-none-eabi-gcc`
When I check my gcc --version, result is 
```
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
```

5) Since I try to not use docker, I followed the instructions to build the micro-ROS libraries https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/ . As for the colcon.meta nad toolchain.cmake files, I didn't change it. I just used the ones in micro_ros_arduino/extras/library_generation folder.

6) However, when I tried to run build firmware as per below instructions, I get an error about -mthumb
What I input in terminal
```
cd microros_ws
ros2 run micro_ros_setup build_firmware.sh /home/prostrain/microros_ws/src/micro_ros_arduino/extras/library_generation/cortex_m3_toolchain.cmake /home/prostrain/microros_ws/src/micro_ros_arduino/extras/library_generation/colcon.meta
```
Error that I got:
```
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /home/prostrain/microros_ws/src/micro_ros_arduino/extras/library_generation/colcon.meta
Starting >>> tinydir_vendor
Starting >>> rosidl_adapter
Starting >>> rosidl_typesupport_interface                                         
Starting >>> microcdr
--- stderr: microcdr                                                                                                              
gcc: error: unrecognized command line option ‘-mthumb’
make[5]: *** [CMakeFiles/microcdr.dir/build.make:66: CMakeFiles/microcdr.dir/src/c/common.c.obj] Error 1
make[5]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command line option ‘-mthumb’
make[5]: *** [CMakeFiles/microcdr.dir/build.make:105: CMakeFiles/microcdr.dir/src/c/types/array.c.obj] Error 1
gcc: error: unrecognized command line option ‘-mthumb’
make[5]: *** [CMakeFiles/microcdr.dir/build.make:79: CMakeFiles/microcdr.dir/src/c/types/basic.c.obj] Error 1
gcc: error: unrecognized command line option ‘-mthumb’
make[5]: *** [CMakeFiles/microcdr.dir/build.make:92: CMakeFiles/microcdr.dir/src/c/types/string.c.obj] Error 1
make[4]: *** [CMakeFiles/Makefile2:79: CMakeFiles/microcdr.dir/all] Error 2
make[3]: *** [Makefile:133: all] Error 2
make[2]: *** [CMakeFiles/ucdr.dir/build.make:115: ucdr-prefix/src/ucdr-stamp/ucdr-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/ucdr.dir/all] Error 2
make: *** [Makefile:87: all] Error 2
---
Failed   <<< microcdr [3.00s, exited with code 2]
Aborted  <<< tinydir_vendor [3.23s]                                                                            
Aborted  <<< rosidl_typesupport_interface [3.16s]
Aborted  <<< rosidl_adapter [3.22s]
                                 
Summary: 0 packages finished [5.05s]
  1 package failed: microcdr
  3 packages aborted: rosidl_adapter rosidl_typesupport_interface tinydir_vendor
  1 package had stderr output: microcdr
  57 packages not processed
```
How can I solve this error? Thank you.

pablogs9, 2022-08-01:
It seems that your toolchain is using `gcc` instead your `arm-none-eabi-gcc` That's why `-mthumb` is not recognized.

Check that the [cortex_m3_toolchain.cmake you are using ](https://github.com/micro-ROS/micro_ros_arduino/blob/39107e4c81affd714db98d7b63655b843905b99a/extras/library_generation/cortex_m3_toolchain.cmake#L5) is using `$ENV{TOOLCHAIN_PREFIX}`. That means that you are taking the environment variable. This is valid for the micro-ROS Arduino build system.

Please check your toolchain.cmake and use something like `set(CMAKE_C_COMPILER [YOUR PATH TO THE XCOMPILER]gcc)`

zharifzubaidi, 2022-08-01:
I see. I tried your recommendation and it can be compiled now but I got a 38 packages had stderr output. Is this acceptable error or not? 
Below are what I have done:
1) I edited cortex_m3_toolchain.cmake in /home/prostrain/microros_ws/src/micro_ros_arduino/extras/library_generation/cortex_m3_toolchain.cmake

```
SET(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_CROSSCOMPILING 1)
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)

#set(CMAKE_C_COMPILER $ENV{TOOLCHAIN_PREFIX}gcc)
#set(CMAKE_CXX_COMPILER $ENV{TOOLCHAIN_PREFIX}g++)

set(CMAKE_C_COMPILER /usr/bin/arm-none-eabi-gcc)
set(CMAKE_CXX_COMPILER /usr/bin/arm-none-eabi-g++)

SET(CMAKE_C_COMPILER_WORKS 1 CACHE INTERNAL "")
SET(CMAKE_CXX_COMPILER_WORKS 1 CACHE INTERNAL "")

set(FLAGS "-O2 -ffunction-sections -fdata-sections -fno-exceptions -mcpu=cortex-m3 -nostdlib -DARDUINO=10813 -mthumb --param max-inline-insns-single=500 -DF_CPU=84000000L -D'RCUTILS_LOG_MIN_SEVERITY=RCUTILS_LOG_MIN_SEVERITY_NONE'" CACHE STRING "" FORCE)

set(CMAKE_C_FLAGS_INIT "-std=c11 ${FLAGS} -DCLOCK_MONOTONIC=0 -D'__attribute__(x)='" CACHE STRING "" FORCE)
set(CMAKE_CXX_FLAGS_INIT "-std=c++11 ${FLAGS} -fno-rtti -DCLOCK_MONOTONIC=0 -D'__attribute__(x)='" CACHE STRING "" FORCE)

set(__BIG_ENDIAN__ 0)
```

2) In terminal, I did this:
`cd microros_ws`

`ros2 run micro_ros_setup build_firmware.sh /home/prostrain/microros_ws/src/micro_ros_arduino/extras/library_generation/cortex_m3_toolchain.cmake /home/prostrain/microros_ws/src/micro_ros_arduino/extras/library_generation/colcon.meta`

3) The result that I get:

```
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /home/prostrain/microros_ws/src/micro_ros_arduino/extras/library_generation/colcon.meta
Starting >>> tinydir_vendor
Starting >>> rosidl_adapter
Starting >>> rosidl_typesupport_interface                     
Starting >>> microcdr
Finished <<< rosidl_typesupport_interface [2.79s]                   
Starting >>> rmw_implementation_cmake
Finished <<< rosidl_adapter [3.22s]                                 
Starting >>> rosidl_parser
Finished <<< rmw_implementation_cmake [3.38s]                           
Starting >>> tracetools
Finished <<< rosidl_parser [3.83s]                                      
Starting >>> rosidl_cmake
Finished <<< microcdr [7.64s]                                           
Starting >>> microxrcedds_client
Finished <<< tinydir_vendor [11.2s]                                        
Starting >>> rcutils
Finished <<< rosidl_cmake [4.38s]                                     
Starting >>> tracetools_trace
Finished <<< tracetools [7.38s]                                       
Starting >>> test_interface_files
Finished <<< test_interface_files [2.58s]                                      
Starting >>> tracetools_read                                                   
Finished <<< tracetools_trace [6.41s]                                        
--- stderr: microxrcedds_client                                               
/home/prostrain/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/util/time.c: In function 'uxr_nanos':
/home/prostrain/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/util/time.c:60:5: warning: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]
   60 |     clock_gettime(CLOCK_REALTIME, &ts);
      |     ^~~~~~~~~~~~~
---
Finished <<< microxrcedds_client [10.2s]
Starting >>> tracetools_launch
Starting >>> libyaml_vendor                                        
--- stderr: rcutils                                                 
[... 721 lines cut ...]
Starting >>> sensor_msgs_py
Finished <<< sensor_msgs_py [4.61s]                                     
--- stderr: stereo_msgs                                      
CMake Warning at /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rmw/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rmw/cmake/rmwConfig.cmake:41 (include)
  /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_microxrcedds_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_microxrcedds_c/cmake/rosidl_typesupport_microxrcedds_cConfig.cmake:41 (include)
  /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /home/prostrain/microros_ws/firmware/mcu_ws/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

---
Finished <<< stereo_msgs [11.6s]
Starting >>> common_interfaces
Finished <<< common_interfaces [2.39s]                          
                             
[... cut ...]
  38 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_noop rclc rclc_lifecycle rclc_parameter rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs trajectory_msgs unique_identifier_msgs visualization_msgs
```

pablogs9, 2022-08-01:
That's ok. Some toolchains produces some verbose warning that are taken as stderr by colcon.

zharifzubaidi, 2022-08-01:
Okay alright. After doing this, I tried to add the new micro_ros_arduino library inside my arduino.
1) Open Arduino IDE, go to Add ZIP library and then I add the micro_ros_arduino as library. The library location is at /home/prostrain/microros_ws/src/micro_ros_arduino

2) I noticed that in Arduino terminal, there is this output when i add the library. So I confirmed that the new library is added.
`Library added to your libraries, Check "Include library" menu.`

3) I then proceed trying to just Verify the example code but this error still come out:

```
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/prostrain/Arduino/libraries/micro_ros_arduino/src/cortex-m3
/home/prostrain/Arduino/libraries/micro_ros_arduino/src/cortex-m3/libmicroros.a(librmw-validate_full_topic_name.c.obj): In function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xf8): undefined reference to `__ctype_ptr__'
/home/prostrain/Arduino/libraries/micro_ros_arduino/src/cortex-m3/libmicroros.a(librcl-validate_topic_name.c.obj): In function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x1cc): undefined reference to `__ctype_ptr__'
/home/prostrain/Arduino/libraries/micro_ros_arduino/src/cortex-m3/libmicroros.a(librmw-validate_node_name.c.obj): In function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x88): undefined reference to `__ctype_ptr__'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Due (Programming Port).
```

pablogs9, 2022-08-01:
The generating library procedure will generate a `libmicroros.a` and a `include` folder. You shall integrate them in the micro-ROS for Arduino package. Here you have the procedure: https://github.com/micro-ROS/micro_ros_arduino/blob/fcab35de945fc81e8d28ee2b9bf79509da508a7e/extras/library_generation/library_generation.sh#L94-L99

- Remove all .c from the generated include folder.
- Copy the content of the include folder in this repo `src` folder.
- Copy the generated `libmicroros.a` in `src/cortex-m3/libmicroros.a`

zharifzubaidi, 2022-08-01:
Okay I did this
1) I got to microros_ws/firmware/build/include and then copy all the contents in here except the .c file. There was no .c file anyway.
2) I then paste the contents in the repo which located at microros_ws/src/micro_ros_arduino/src/ . I just select replace all. Note that there are still few cpp files here like default transport, native ethernet etc. I didn't disturbed this.
3) From this folder microros_ws/firmware/build/ , I then copy the microros.a and paste at microros_ws/src/micro_ros_arduino/src/cortex-m3/libmicroros.a and select replace option.
4)  Then, I proceed to the library by using Include library/Add .ZIP Library and select this folder /home/prostrain/microros_ws/src/micro_ros_arduino/
5) I tried to Verify the example publisher code and it was compiled successfully
6) I also tried to upload the code to my Arduino Due and it was uploaded successfully.

No words can describe the happiness that I feel right now. hahah Now, I can focus on developing my robot.

Thank you @pablogs9 for the guidance! Keep up the good work with microros arduino!

closed 2022-08-01 by pablogs9

VOCA-L, 2023-02-14:
Hi, @zharifzubaidi , I face same problem. 
I follow your solution but I can't find workspacce/src/micro_ros_arduino/cortex-m3
How can I install?

zharifzubaidi, 2023-02-20:
Hi @VOCA-L , have you follow the micro_ros_arduino main installation step first? If you have, you should have that folder.