micro-ROS/micro_ros_arduino #1141: Compilation failed

metanav, 2022-08-13:
- Hardware description: Arduino Nano RP2040 Connect
- RTOS: Mbed OS
- Installation type: Arduino Library
- Version or commit hash: Humble

I am getting following error while compilation.

```
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_introspection_c-gripper_command__type_support.c.obj): In function `rosidl_typesupport_introspection_c__get_message_type_support_handle__control_msgs__action__GripperCommand_Goal':
gripper_command__type_support.c:(.text.rosidl_typesupport_introspection_c__get_message_type_support_handle__control_msgs__action__GripperCommand_Goal+0x2): undefined reference to `rosidl_typesupport_introspection_c__get_message_type_support_handle__control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_introspection_c-gripper_command__type_support.c.obj): In function `rosidl_typesupport_introspection_c__get_message_type_support_handle__control_msgs__action__GripperCommand_SendGoal_Request':
gripper_command__type_support.c:(.text.rosidl_typesupport_introspection_c__get_message_type_support_handle__control_msgs__action__GripperCommand_SendGoal_Request+0xa): undefined reference to `rosidl_typesupport_introspection_c__get_message_type_support_handle__control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_introspection_c-gripper_command__type_support.c.obj): In function `rosidl_typesupport_introspection_c__get_service_type_support_handle__control_msgs__action__GripperCommand_SendGoal':
gripper_command__type_support.c:(.text.rosidl_typesupport_introspection_c__get_service_type_support_handle__control_msgs__action__GripperCommand_SendGoal+0x32): undefined reference to `rosidl_typesupport_introspection_c__get_message_type_support_handle__control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_microxrcedds_c-gripper_command__type_support_c.c.obj): In function `get_serialized_size_control_msgs__action__GripperCommand_Goal':
gripper_command__type_support_c.c:(.text.get_serialized_size_control_msgs__action__GripperCommand_Goal+0x8): undefined reference to `get_serialized_size_control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_microxrcedds_c-gripper_command__type_support_c.c.obj): In function `_GripperCommand_Goal__max_serialized_size':
gripper_command__type_support_c.c:(.text._GripperCommand_Goal__max_serialized_size+0xe): undefined reference to `max_serialized_size_control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_microxrcedds_c-gripper_command__type_support_c.c.obj): In function `_GripperCommand_Goal__get_serialized_size':
gripper_command__type_support_c.c:(.text._GripperCommand_Goal__get_serialized_size+0xa): undefined reference to `get_serialized_size_control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_microxrcedds_c-gripper_command__type_support_c.c.obj): In function `_GripperCommand_Goal__cdr_deserialize':
gripper_command__type_support_c.c:(.text._GripperCommand_Goal__cdr_deserialize+0xc): undefined reference to `rosidl_typesupport_microxrcedds_c__get_message_type_support_handle__control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_microxrcedds_c-gripper_command__type_support_c.c.obj): In function `_GripperCommand_Goal__cdr_serialize':
gripper_command__type_support_c.c:(.text._GripperCommand_Goal__cdr_serialize+0xc): undefined reference to `rosidl_typesupport_microxrcedds_c__get_message_type_support_handle__control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_microxrcedds_c-gripper_command__type_support_c.c.obj): In function `get_serialized_size_control_msgs__action__GripperCommand_SendGoal_Request':
gripper_command__type_support_c.c:(.text.get_serialized_size_control_msgs__action__GripperCommand_SendGoal_Request+0x1a): undefined reference to `get_serialized_size_control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_microxrcedds_c-gripper_command__type_support_c.c.obj): In function `_GripperCommand_SendGoal_Request__get_serialized_size':
gripper_command__type_support_c.c:(.text._GripperCommand_SendGoal_Request__get_serialized_size+0x1a): undefined reference to `get_serialized_size_control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_typesupport_microxrcedds_c-gripper_command__type_support_c.c.obj): In function `_GripperCommand_SendGoal_Request__max_serialized_size':
gripper_command__type_support_c.c:(.text._GripperCommand_SendGoal_Request__max_serialized_size+0x1c): undefined reference to `max_serialized_size_control_msgs__msg__GripperCommand'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_generator_c-gripper_command__functions.c.obj): In function `control_msgs__action__GripperCommand_Goal__init':
gripper_command__functions.c:(.text.control_msgs__action__GripperCommand_Goal__init+0xa): undefined reference to `control_msgs__msg__GripperCommand__init'
gripper_command__functions.c:(.text.control_msgs__action__GripperCommand_Goal__init+0x18): undefined reference to `control_msgs__msg__GripperCommand__fini'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_generator_c-gripper_command__functions.c.obj): In function `control_msgs__action__GripperCommand_Goal__fini':
gripper_command__functions.c:(.text.control_msgs__action__GripperCommand_Goal__fini+0x6): undefined reference to `control_msgs__msg__GripperCommand__fini'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_generator_c-gripper_command__functions.c.obj): In function `control_msgs__action__GripperCommand_SendGoal_Request__init':
gripper_command__functions.c:(.text.control_msgs__action__GripperCommand_SendGoal_Request__init+0x1a): undefined reference to `control_msgs__msg__GripperCommand__init'
gripper_command__functions.c:(.text.control_msgs__action__GripperCommand_SendGoal_Request__init+0x28): undefined reference to `control_msgs__msg__GripperCommand__fini'
gripper_command__functions.c:(.text.control_msgs__action__GripperCommand_SendGoal_Request__init+0x34): undefined reference to `control_msgs__msg__GripperCommand__fini'
/Users/naveen/Documents/Arduino/libraries/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libcontrol_msgs__rosidl_generator_c-gripper_command__functions.c.obj): In function `control_msgs__action__GripperCommand_SendGoal_Request__fini':
gripper_command__functions.c:(.text.control_msgs__action__GripperCommand_SendGoal_Request__fini+0xe): undefined reference to `control_msgs__msg__GripperCommand__fini'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Nano RP2040 Connect.

```

Below is the function call that causes compilation errors.

```
  RCCHECK(
    rclc_action_server_init_default(
      &action_server,
      &node,
      &support,
      ROSIDL_GET_ACTION_TYPE_SUPPORT(control_msgs, GripperCommand),
      "/robot/gripper_command"
    ));
```

By the way the code below compiles successfully.

```
  RCCHECK(
    rclc_action_server_init_default(
      &action_server_follow_trajectory,
      &node,
      &support,
      ROSIDL_GET_ACTION_TYPE_SUPPORT(control_msgs, FollowJointTrajectory),
      "/robot/follow_joint_trajectory"
    ));
```

metanav, 2022-08-24:
What could be the reason for this issue? Please let me know where should I investigate.

Acuadros95, 2022-08-24:
I think the issue is caused because the `GripperCommand` action uses a message with the same name. Detail:
- [GripperCommand.action](https://github.com/ros-controls/control_msgs/blob/galactic-devel/control_msgs/action/GripperCommand.action)
- [GripperCommand.msg](https://github.com/ros-controls/control_msgs/blob/galactic-devel/control_msgs/msg/GripperCommand.msg)

Sadly we don't have time to fix this issue at this moment, you can probably add a custom msg type or create your own fork for `control_msgs` modifying the `GripperCommand` message name. What is your use case?

metanav, 2022-08-24:
I am creating an action server which expects `GripperCommand` message. I am OK with modifying the code locally. I will try to change the message name and recompile again.

metanav, 2022-08-24:
I have forked the `control_msgs` repo and renamed the `control_msgs/GripperCommand.msg` to  `control_msgs/GripperCmd.msg`. Also, I have modified following files and changed `GripperCommand` to `GripperCmd`
- control_msgs/CMakeLists.txt
- control_msgs/action/GripperCommand.action

After recompilation and using precompiled library, errors are gone. 
I will check later if the action server receives the message correctly or not.

metanav, 2022-09-01:
I can confirm it is working.

closed 2022-09-01 by metanav