micro-ROS/micro_ros_arduino #1145: Handling large action request gets failed

metanav, 2022-08-17:
- Hardware description: Arduino Nano RP2040 Connect
- RTOS: Mbed OS
- Installation type: Arduino Library
- Version or commit hash: Humble

I am trying to create a follow_joint_trajectory action server with a joint_states publisher. I am able to publish to the join_states topic. Also, I can receive trajectory goal at the follow_joint_trajectory server.  But It seems the action server cannot handle large goal request. The Arduino Nano RP2040 Connect has 264KB memory so i believe it can handle large messages around 10 to 20 KB easily.

Successful request:

```
$ ros2 action send_goal /arm/follow_joint_trajectory control_msgs/action/FollowJointTrajectory "trajectory: {header: {frame_id: "me"}, joint_names: ['this_is_joint_1', 'this_is_joint_2', 'this_is_joint_3', 'this_is_joint_4', 'this_is_joint_5', 'this_is_joint_6'], points: [{positions: [1, 2, 3, 4, 5, 5]}, {positions: [1, 2, 3, 4, 5, 5]}, {positions: [1, 2, 3, 4, 5, 5]} , {positions: [1, 2, 3, 4, 5, 5]}, {positions: [1, 2, 3, 4, 5, 5]}, ] }" 
Waiting for an action server to become available...
Sending goal:
     trajectory:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: me
  joint_names:
  - this_is_joint_1
  - this_is_joint_2
  - this_is_joint_3
  - this_is_joint_4
  - this_is_joint_5
  - this_is_joint_6
  points:
  - positions:
    - 1.0
    - 2.0
    - 3.0
    - 4.0
    - 5.0
    - 5.0
    velocities: []
    accelerations: []
    effort: []
    time_from_start:
      sec: 0
      nanosec: 0
  - positions:
    - 1.0
    - 2.0
    - 3.0
    - 4.0
    - 5.0
    - 5.0
    velocities: []
    accelerations: []
    effort: []
    time_from_start:
      sec: 0
      nanosec: 0
  - positions:
    - 1.0
    - 2.0
    - 3.0
    - 4.0
    - 5.0
    - 5.0
    velocities: []
    accelerations: []
    effort: []
    time_from_start:
      sec: 0
      nanosec: 0
  - positions:
    - 1.0
    - 2.0
    - 3.0
[... 403 lines cut ...]
    rclc_executor_add_action_server(
      &executor,
      &action_server,
      MAX_GOALS,
      ros_goal_request,
      sizeof(control_msgs__action__FollowJointTrajectory_SendGoal_Request),
      goal_callback,
      cancel_callback,
      (void *) &action_server));
  Serial1.println("rclc_executor_add_action_server");

  rclc_executor_add_timer(&executor, &timer);

  th.start(goal_handler_thread);
  Serial1.println("thread started");

}

void loop()
{
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

```

metanav, 2022-08-18:
It works If I remove the publisher implementation from the code. So why publisher does cause the issue? Since I need the publisher in this code, the issue is still unsolved.

Acuadros95, 2022-08-19:
Consider adding an sleep on your `void goal_handler_thread()` thread and reducing the sleep on the main loop, as your timer has a period of 10 ms and you are sleeping 100 ms on each executor iteration.

> So why the publisher (or timer?) does cause the issue?

Did you modify anything else? What is the output of your serial debug port?

metanav, 2022-08-19:
Reducing the sleep ( `delay(1)` )  on the` loop() `, adding a sleep 50ms (`delay(50)`) on the `goal_handler_thread()` and  increasing the timer timeout to 500ms (2Hz) which allows to receive `goal.trajectory.points` up to ~20.  I have changed the rules to accept 50 points to see how much we can achieve ` {"goal.trajectory.points", 50}`.  I think there is not much room to adjust sleep so I think we will have to think other ways.

Acuadros95, 2022-08-22:
From the agent help:
```
Usage: 'micro_ros_agent <udp4|udp6|tcp4|tpc6|canfd|serial|multiserial|pseudoterminal> <<args>>'

Available arguments (per transport):
  * COMMON
    -h/--help.
    -m/--middleware <value> (ced, dds, rtps) [default: 'dds'].
    -r/--refs <value>.
    -v/--verbose <value> ( - ) [default: ''].
    -d/--discovery <value> [default: '7400'].
  * IPvX (udp4, udp6, tcp4, tcp6)
    -p/--port <value>.
  * SERIAL (serial, multiserial, pseudoterminal)
    -b/--baudrate <value> [default: '115200'].
    -D/--dev <value>.  * CAN FD (canfd)
    -D/--dev <value>.
```

You can modify the baudrate using the flag `-b  <value>` on the agent arguments.
On Arduino side, you can modify this value on your library: https://github.com/micro-ROS/micro_ros_arduino/blob/humble/src/default_transport.cpp#L36

**Anyway, we just discovered a bug regarding services reliability on the agent side. This directly affects action server communication, will keep you updated when the fix is ready.**

Acuadros95, 2022-08-24:
@metanav The fix is added, could you update your agent and test this again?

metanav, 2022-08-28:
Hi @Acuadros95,

I have updated the agent and tested it but it does not fix the issue.

metanav, 2022-08-30:
Hi @Acuadros95,

Is there any way to pause the timer when goal request comes in and restart after goal completed?

metanav, 2022-09-01:
It seems working after increasing RMW_UXRCE_STREAM_HISTORY and regenerating precompiled library,

closed 2022-09-01 by metanav

madgrizzle, 2023-01-07:
@metanav I've run into the same issue.  You mentioned earlier on that you tried 32 for the RMW_UXRCE_STREAM HISTORY and it didn't seem to have an effect, but your last post said you increased it AND regenerated the precompiled library.  Did you stick with 32 and it didn't take effect until you regenerated the precompiled library, or did you end up going beyond 32, or both?

madgrizzle, 2023-01-07:
To answer my own question, yes, both need to be done.  Only make sure if you are using platform.io that you do not use a tab when editing or creating custom meta files as it will result in the meta file being ignored when you manage to force micro-ros to be rebuilt... as I discovered over the last 6 hours.

metanav, 2023-01-14:
@madgrizzle I do not remember exactly but I guess it worked after doing both. By the way which manipulator/microcontroller are you using?

madgrizzle, 2023-01-14:
Yeah, both are needed as well as a meta file without tabs in it :) I'm using a teensy 4.1 and was able to get well over 100 points sent with 8 joints per point.  That part works well at least now.

metanav, 2023-01-14:
Yeah, Teensy 4.1 has larger memory to handle it. I was working with Nano RP2040 Connect which has only 1/4 of that.

madgrizzle, 2023-05-12:
@metanav is there any chance you can share your code for implementing the followjointtrajectory action using micro-ros?   previously I just published a joint trajectory to a topic but now I'm trying to shift to using an action and for some reason it never calls the handle_goal function.  It gets called when I use a simple action definition (fibonacci) but not something more complex like followjointtrajectory.