micro-ROS/micro_ros_arduino #1159: Fail to build using micro_ros_static_library_builder docker

madgrizzle, 2022-09-02:
I'm not entirely sure this is a micro_ros_arduino issue, but I can't seem rebuild the library to add a custom message (and don't think it has to do with the custom message).  I'm following the directions from README.MD, but near the end I get rsync errors trying to change directory:

> Setting up libpopt0:amd64 (1.16-14) ...
Setting up rsync (3.1.3-8ubuntu0.4) ...
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
Processing triggers for libc-bin (2.31-0ubuntu9.9) ...
rsync: change_dir "/project/src/action_msgs/action_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
rsync: change_dir "/project/src/actionlib_msgs/actionlib_msgs" failed: No such file or directory (2)

This is the command I used:
`docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:galactic -p esp32`

I did notice this earlier on in the output regarding the project directory:

>=== ./control_msgs (git) ===
Cloning into '.'...
find: cannot delete ‘/project/src/’: Directory not empty

full pastebin:
https://pastebin.com/JuJ00Bsk

Any help would be.. helpful :)

Acuadros95, 2022-09-02:
Two pints here:
- Are you using arduino galactic branch? Make sure the version used on the `micro_ros_static_library_builder` matches the repository branch.
- Those errors can happen on a successful build. I understand that it can be confusing, did you check if the library was actually rebuild?

madgrizzle, 2022-09-02:
I'll check to see what branch is being rebuilt.  It's possible its not galactic (though I did try the same process using the humble docker version.. hard to believe its not one of those two).

And no, I didn't check as I just assumed it didn't work.  But I will!

Thanks.

madgrizzle, 2022-09-02:
@Acuadros95 Thank you for the help.  Turns out it did work after all.  I just assumed all that red text was bad.

closed 2022-09-02 by madgrizzle