micro-ROS/micro_ros_arduino #1164: Handling [RTPS_READER_HISTORY Error]

avielbr, 2022-09-13:
## Issue template

- Hardware description: 
- Installation type: 
- Version or commit hash: 

#### Steps to reproduce the issue

#### Expected behavior

I expect to see the message echoed to the terminal (as it did in the past -- don't know what changed...)  Like this:
```
label: test
value: 0.5
```

#### Actual behavior
Receiving following error upon every publish:
```
...
2022-09-13 22:03:02.565 [RTPS_READER_HISTORY Error] Change payload size of '24' bytes is larger than the history payload size of '11' bytes and cannot be resized. -> Function can_change_be_added_nts
2022-09-13 22:03:02.768 [RTPS_READER_HISTORY Error] Change payload size of '24' bytes is larger than the history payload size of '11' bytes and cannot be resized. -> Function can_change_be_added_nts
2022-09-13 22:03:02.970 [RTPS_READER_HISTORY Error] Change payload size of '24' bytes is larger than the history payload size of '11' bytes and cannot be resized. -> Function can_change_be_added_nts
2022-09-13 22:03:03.173 [RTPS_READER_HISTORY Error] Change payload size of '24' bytes is larger than the history payload size of '11' bytes and cannot be resized. -> Function can_change_be_added_nts
2022-09-13 22:03:03.375 [RTPS_READER_HISTORY Error] Change payload size of '24' bytes is larger than the history payload size of '11' bytes and cannot be resized. -> Function can_change_be_added_nts
...
```

#### Additional information

Output from agent:
```
...
[1663096070.949240] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D9CF9BB, len: 32, data: 
0000: 81 80 47 0A 07 01 18 00 0A 51 00 05 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096070.949367] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 20, data: 
0000: 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096070.949618] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D9CF9BB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 48 0A 00 00 80
[1663096071.151371] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D9CF9BB, len: 32, data: 
0000: 81 80 48 0A 07 01 18 00 0A 52 00 05 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096071.151513] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 20, data: 
0000: 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096071.151660] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D9CF9BB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 49 0A 00 00 80
[1663096071.353919] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D9CF9BB, len: 32, data: 
0000: 81 80 49 0A 07 01 18 00 0A 53 00 05 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 6D 00 00 00 3F
[1663096071.354062] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 20, data: 
0000: 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 6D 00 00 00 3F
[1663096071.354206] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D9CF9BB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 4A 0A 00 00 80
[1663096071.556267] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D9CF9BB, len: 32, data: 
0000: 81 80 4A 0A 07 01 18 00 0A 54 00 05 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096071.556407] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 20, data: 
0000: 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096071.556538] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D9CF9BB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 4B 0A 00 00 80
[1663096071.758436] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D9CF9BB, len: 32, data: 
0000: 81 80 4B 0A 07 01 18 00 0A 55 00 05 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096071.758606] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 20, data: 
0000: 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096071.758709] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D9CF9BB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 4C 0A 00 00 80
[1663096071.960814] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D9CF9BB, len: 32, data: 
0000: 81 80 4C 0A 07 01 18 00 0A 56 00 05 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096071.960972] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 20, data: 
0000: 0B 00 00 00 74 65 73 74 5F 6C 61 62 65 6C 00 00 00 00 00 3F
[1663096071.961117] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D9CF9BB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 4D 0A 00 00 80
...
```

pablogs9, 2022-09-14:
How are you subscribing to the topic in the ROS 2 side?

avielbr, 2022-09-14:
> How are you subscribing to the topic in the ROS 2 side?

I haven't yet, I'm trying to get `ros2 topic echo` working first.

pablogs9, 2022-09-14:
Make sure that the type you are using is the same `.msg` on both sides. Rebuild if neccessary.

avielbr, 2022-09-14:
> Make sure that the type you are using is the same `.msg` on both sides. Rebuild if neccessary.

What's the most surefire way to check this on the Arduino side?

pablogs9, 2022-09-14:
Checking that the struct inside the header folder is the same as the one you have in the install folder in your ROS 2 workspace

avielbr, 2022-09-14:
> Make sure that the type you are using is the same `.msg` on both sides. Rebuild if neccessary.

So on the ROS2 side:
`ros2 interfaces show ei_interfaces/msg/EIClassification` shows:
```
string label
float32 value
```

And in `Arduino/libraries/micro_arduino-2.0.5-humble/src/ei_interfaces/msg/detail/ei_classification__struct.h`:

```
...
// Struct for a sequence of ei_interfaces__msg__EIClassification.
typedef struct ei_interfaces__msg__EIClassification__Sequence
{
  ei_interfaces__msg__EIClassification * data;
  /// The number of valid items in data
  size_t size;
  /// The number of allocated items in data
  size_t capacity;
} ei_interfaces__msg__EIClassification__Sequence;
...
```

Aside from it saying `float` instead of `float32` (which might be the same thing in this case?) I think they match. I confirmed this was the actual file being used by changing its name and checking if compilation failed (it did).

pablogs9, 2022-09-14:
Does it work if you use an `int` instead of `float`? Just for the sake of testing...

avielbr, 2022-09-14:
> Does it work if you use an `int` instead of `float`? Just for the sake of testing...
> 
> If it does, I will take a look on that to check what's going on.

I'll check this out. I'm trying to rebuild now for testing, and then I'll try with `int`. 

Perhaps worth noting that when I change the length of the string (`"test_message"`) I notice that the value `X` is affected in the error message:

```
2022-09-13 22:03:03.375 [RTPS_READER_HISTORY Error] Change payload size of 'X' bytes is larger than the history payload size of '11' bytes and cannot be resized. -> Function can_change_be_added_nts
```

When I pass an empty string `""` the value shrinks to 16 bytes. So maybe the issue lies somewhere with the `string label` and not with the `float32 value`.

avielbr, 2022-09-14:
Ok, cleaning out `ros2_ws` and rebuilding solved the issue. No idea what changed considering `ros2 interfaces show` was showing the right format -- but thanks anyway for the response! I should have tried this first.

closed 2022-09-14 by avielbr