micro-ROS/micro_ros_arduino #1167: Service example: The passed service type is invalid

wuhanstudio, 2022-09-17:
Hi,

I tried the service example `micro-ros_addtwoints_service.ino` on Arduino

https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino

The node and service was registered, but received an error after calling the service:

```
$ ros2 service call /add_two_ints example_interfaces/srv/AddTwoInts '{a: 1, b: 2}'
The passed service type is invalid
```

## Environments
- Board: STM32F103RCT6

- ROS2 Versioin: Galactic
- micro_ros Version: [v2.0.5-galactic](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v2.0.5-galactic)

- Arduino Version: 2.0.0
- GCC Version: arm-none-eabi-gcc/10.3.1-2.3 (from Arduino)
- Static Library: I regenerated the static library `libmicroros.a` using the above toolchain.

## Logs

micro_ros_agent:
```
$ docker run -it -v /dev:/dev --privileged microros/micro-ros-agent:galactic serial --dev /dev/ttyUSB0
[1663458202.047773] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1663458202.048243] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1663458227.566860] info     | Root.cpp           | create_client            | create                 | client_key: 0x01BDDB77, session_id: 0x81
[1663458227.567031] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x01BDDB77, address: 0
[1663458227.587226] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x01BDDB77, participant_id: 0x000(1)
[1663458227.612137] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x01BDDB77, requester_id: 0x000(7), participant_id: 0x000(1)
```

ROS2 (inside the container):
```
root@a9d5d8a1afc9:/uros_ws# ros2 node list
/add_twoints_client_rclc
root@a9d5d8a1afc9:/uros_ws# ros2 service list
/addtwoints
root@a9d5d8a1afc9:/uros_ws# ros2 service type /addtwoints
example_interfaces/srv/AddTwoInts
root@a9d5d8a1afc9:/uros_ws# ros2 service call /add_two_ints example_interfaces/srv/AddTwoInts '{a: 1, b: 2}'
The passed service type is invalid
```

Acuadros95, 2022-09-19:
You need to install the message types on your ros2 environment:
`apt install ros-$ROS_DISTRO-example-interfaces`

wuhanstudio, 2022-09-19:
Thanks for your help. I installed the message types on my PC, but forgot to install it inside the container.

```
root@e963ef5f3a08:/uros_ws# ros2 service call /addtwoints example_interfaces/srv/AddTwoInts '{a: 1, b: 2}'
requester: making request: example_interfaces.srv.AddTwoInts_Request(a=1, b=2)

response:
example_interfaces.srv.AddTwoInts_Response(sum=3)
```

closed 2022-09-19 by wuhanstudio