micro-ROS/micro_ros_arduino #1171: Fail to build the precompiled library by following the instructions in the package

marklee56826, 2022-09-25:
Issue: Fail to build the precompiled library by following the instructions in the package micor_ros_arduino. Please help me.
Hardware and software description: 
Intel NUC i5 CPU, esp32 board, ubuntu 22.04, ros2 humble, arduino 1.8.19
Package my_custom_message used and this package was downloaded from  micro-ROS/micro_ros_platformio 
THE DOCKER COMMAND:
$: ~/Arduino/libraries/micro_ros_arduino-humble$
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:humble
THE TERMINAL OUTPUT:
Get:1 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Get:2 http://packages.ros.org/ros2/ubuntu jammy InRelease [4673 B]               
Get:3 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]
…
Summary: 61 packages finished [39.6s]
  18 packages had stderr output: ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint domain_coordinator
#All required rosdeps installed successfully
Cloning into 'geometry2'...
remote: Enumerating objects: 15047, done.
remote: Counting objects: 100% (2397/2397), done.
remote: Compressing objects: 100% (518/518), done.
remote: Total 15047 (delta 2113), reused 2003 (delta 1874), pack-reused 12650
Receiving objects: 100% (15047/15047), 3.41 MiB | 2.76 MiB/s, done.
Resolving deltas: 100% (10505/10505), done.
=== ./control_msgs (git) ===
Cloning into '.'...
find: cannot delete ‘/project/src/’: Directory not empty
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /project/extras/library_generation/colcon.meta
Starting >>> builtin_interfaces
Starting >>> unique_identifier_msgs
Starting >>> rosidl_cli
Starting >>> rcutils
--- stderr: rosidl_cli                                                  
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< rosidl_cli [2.19s]
Starting >>> rosidl_adapter
--- stderr: unique_identifier_msgs                                      
CMake Error at CMakeLists.txt:17 (find_package):
  By not providing "Findrosidl_core_generators.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "rosidl_core_generators", but CMake did not find one.
  Could not find a package configuration file provided by
  "rosidl_core_generators" with any of the following names:
    rosidl_core_generatorsConfig.cmake
    rosidl_core_generators-config.cmake
  Add the installation prefix of "rosidl_core_generators" to
  CMAKE_PREFIX_PATH or set "rosidl_core_generators_DIR" to a directory
  containing one of the above files.  If "rosidl_core_generators" provides a
  separate development package or SDK, be sure it has been installed.
---
Failed   <<< unique_identifier_msgs [2.28s, exited with code 1]
Aborted  <<< builtin_interfaces [2.31s]
Aborted  <<< rcutils [2.84s]                                            
Aborted  <<< rosidl_adapter [0.96s]      
Summary: 1 package finished [3.74s]
  1 package failed: unique_identifier_msgs
  3 packages aborted: builtin_interfaces rcutils rosidl_adapter
  3 packages had stderr output: builtin_interfaces rosidl_cli unique_identifier_msgs
  67 packages not processed
[ros2run]: Process exited with failure 1
……
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
rsync is already the newest version (3.2.3-8ubuntu3).
0 upgraded, 0 newly installed, 0 to remove and 204 not upgraded.
rsync: [sender] change_dir "/project/src/action_msgs/action_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1333) [sender=3.2.3]
……End

pablogs9, 2022-09-26:
Try with a `docker pull microros/micro_ros_static_library_builder:humble` before the docker command

marklee56826, 2022-09-26:
Thank you for your reply. I did that before the docker command.

pablogs9, 2022-09-26:
I have just ran 

```
git clone -b humble https://github.com/micro-ROS/micro_ros_arduino
cd micro_ros_arduino
docker pull microros/micro_ros_static_library_builder:humble
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:humble
```

and it has worked as expected. Could you retry?

marklee56826, 2022-09-27:
Thank you for your kind reply.  I tried to run the commands on my machine and the result was much better: more packages were processed even if most of the  packages had stderr output. Finally, it failed after several loops.

closed 2022-09-27 by marklee56826

marklee56826, 2022-09-27:
There may be a problem with my system, not the package. Thank you, pablogs9.