micro-ROS/micro_ros_arduino #1178: How to create PoseArray

matthew-william-lock, 2022-10-05:
Hey !

The general question here is regarding the PoseArray. 

I am not completely comfortable with using pure C, and am looking for some assistance in create a PoseArray object.

There are some [examples](https://answers.ros.org/question/339453/question-related-to-posearray-and-its-implementation/) available for the normal Cpp ROS implementation, but I am unsure how to do this with micro-ROS.

Any advice is appreciated!

matthew-william-lock, 2022-10-05:
This does not work :cry:

pablogs9, 2022-10-06:
You have the whole description of how to handle message memory here: https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/

Manually it looks like:
```c
    geometry_msgs__msg__PoseArray m = {};
    m.header.stamp.sec = 0;
    m.header.stamp.nsec = 0;

    char frame_name[] = "base_link";
    m.header.frame_id.data = frame_name;
    m.header.frame_id.size = sizeof(frame_name);
    m.header.frame_id.capacity = sizeof(frame_name);

    geometry_msgs__msg__Pose pose_array[10];
    m.poses.data = pose_array;
    m.poses.size = 10;
    m.poses.capacity = 10;

    for(int i = 0; i < 10; i++){
        pose_array[i].position.x = 0;
        pose_array[i].position.y = 0;
        pose_array[i].position.z = 0;
        pose_array[i].orientation.x = 0;
        pose_array[i].orientation.y = 0;
        pose_array[i].orientation.z = 0;
        pose_array[i].orientation.w = 0;
    }
```

You also can use `micro_ros_utils` for doing this, the whole info in the link above.

closed 2022-10-06 by pablogs9