micro-ROS/micro_ros_arduino #1181: Error compiling teensy code

samprt-heracles, 2022-10-12:
## Issue template

- Hardware description: Teensy 4.1
- Installation type: micro_ros_setup
- Version or commit hash: humble

#### Steps to reproduce the issue
I am trying to compile a sketch for a Teensy 4.1 using arduino CLI (version 0.27.1). I added the Teensy support to the CLI and the patch to the platform.txt file. However, I get the following error when performing the following command
  ```
$ arduino-cli compile --fqbn teensy:avr:teensy41 <sketch_name>
  Used platform Version      Path                                                          
  teensy:avr    1.58.0-beta2 /home/sam/.arduino15/packages/teensy/hardware/avr/1.58.0-beta2
  
  Error during build: fork/exec /home/sam/.arduino15/packages/teensy/hardware/avr/../tools/arm/bin/arm-none-eabi-g++: no such file or directory
```

#### Expected behavior
Sketch compiles

#### Actual behavior
Error while compiling

#### Additional information
- I have tried to uninstall the teensy files and reinstall them but the behavior is the same. 
- The tools/ folder contains :
`teensy-compile  teensy-discovery  teensy-monitor  teensy-tools`
but not the arm/bin/... folders from the error message.

Any help would be much appreciated !

Acuadros95, 2022-10-13:
How did you install Teensy support? Can you compile a simple blink example?

You can check our CI for a working arduino CLI install: [CI](https://github.com/micro-ROS/micro_ros_arduino/blob/humble/.github/workflows/ci.yml)

Focus on the Teensy sections:
```
# INSTALLING TEENSY SUPPORT
wget https://downloads.arduino.cc/arduino-1.8.19-linux64.tar.xz
tar -xf arduino-1.8.19-linux64.tar.xz
wget https://www.pjrc.com/teensy/td_156/TeensyduinoInstall.linux64
chmod 755 TeensyduinoInstall.linux64
./TeensyduinoInstall.linux64 --dir=arduino-1.8.19
```

samprt-heracles, 2022-10-13:
I installed the Teensy support by adding the URL in the arduino-cli.yaml config file and then used the `arduino-cli core install teensy:avr` command to install the teensy files.
After that I could compile the Teensy sketches fine, but when I patch the platform.txt file, I was not able to compile anymore, not even the Blink example.

Acuadros95, 2022-10-14:
Did you try to compile a micro-ROS example without the patch step?

Anyway, micro-ROS support for Teensy is [Based on Teensyduino](https://www.pjrc.com/teensy/td_download.html), and the patch is mean for this add-on which is our recommended approach.

Another option is to use [micro_ros_platformio](https://github.com/micro-ROS/micro_ros_platformio), which integrates micro-ROS on the platformIO enviroment, with support for teensy boards. But this goes out of the Arduino IDE scope.

samprt-heracles, 2022-10-14:
Yes I tried and it gives undefined references for all rcl functions and variables.
I will try the Teensyduino approach now.

Acuadros95, 2022-10-20:
Any update on this?

samprt-heracles, 2022-10-24:
No luck, sorry. I also tried using platformIO but did not manage either. I lack the experience to get microROS to work on my system.
Feel free to close the issue as it seems I am the only one with this problem.
Thanks for your help !

AyaZeidan, 2022-12-07:
I have the same issue. I tried using PlatformIO, but I get lots of issues when I attempt to build, such as ".pio/libdeps/due/micro_ros_platformio/libmicroros/include/rmw/types.h:392:49: error: expected '}' before '__attribute__' ".

PR #1261 (2023-02-01): Fix teensy compile and board discovery for arduino-cli
Closes #1181 #1192 

Fixes out-of-date paths for locations of compiler toolchain and additional teensy tools for arduino-cli. In addition, pulls upstream changes for board discovery from the default platform.txt. I am unsure of the impact this has on IDE users but I would assume the paths are updated for 2.0.0 as well.