micro-ROS/micro_ros_arduino #1192: Running into issues with Teensy 4.1

avielbr, 2022-10-26:
Hi! Hoping to get some help getting my Teensy 4.1 running the publisher example.

Using micro-ROS Arduino on Ubuntu 22.04 running Humble.

#### Steps to reproduce the issue
- Install Teensyduino for Arduino 1.8.19 (Blink example uploads successfully)
- Try to upload "publisher" example from micro-ROS Arduino examples

#### Expected behavior
- This example works fine using the Portenta H7, so I expected it to work the same with the Teensy. Maybe I need to change something though first..?

#### Actual behavior
- Seeing the following error output:
```
Arduino: 1.8.19 (Linux), TD: 1.57, Board: "Teensy 4.1, Serial, 600 MHz, Faster, US English"

WARNING: library micro_ros_arduino-2.0.5-humble claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino-2.0.5-humble has been declared precompiled:
Precompiled library in "/root/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/imxrt1062/fpv5-d16-hard" not found
Precompiled library in "/root/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/imxrt1062" not found
The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.
/tmp/arduino_build_130983/sketch/sketch_oct19b.ino.cpp.o: In function `timer_callback(rcl_timer_s*, long long)':
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:36: undefined reference to `rcl_publish'
/tmp/arduino_build_130983/sketch/sketch_oct19b.ino.cpp.o: In function `set_microros_transports':
/root/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/micro_ros_arduino.h:40: undefined reference to `rmw_uros_set_custom_transport'
/tmp/arduino_build_130983/sketch/sketch_oct19b.ino.cpp.o: In function `setup':
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:49: undefined reference to `rcutils_get_default_allocator'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:52: undefined reference to `rclc_support_init'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:55: undefined reference to `rclc_node_init_default'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:58: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:58: undefined reference to `rclc_publisher_init_default'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:66: undefined reference to `rclc_timer_init_default'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:73: undefined reference to `rclc_executor_init'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:74: undefined reference to `rclc_executor_add_timer'
/tmp/arduino_build_130983/sketch/sketch_oct19b.ino.cpp.o: In function `loop':
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 4.1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Happy to provide any additional information as needed. Thanks!

avielbr, 2022-10-26:
Ok, did some digging and it appears that somehow `/imxrt1062` had gotten deleted from `/src`. I returned it and the error is now showing:

```
Arduino: 1.8.19 (Linux), TD: 1.57, Board: "Teensy 4.1, Serial, 600 MHz, Faster, US English"

WARNING: library micro_ros_arduino-2.0.5-humble claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino-2.0.5-humble has been declared precompiled:
Using precompiled library in /root/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/imxrt1062/fpv5-d16-hard
The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.
/tmp/arduino_build_130983/sketch/sketch_oct19b.ino.cpp.o: In function `timer_callback(rcl_timer_s*, long long)':
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:36: undefined reference to `rcl_publish'
/tmp/arduino_build_130983/sketch/sketch_oct19b.ino.cpp.o: In function `set_microros_transports':
/root/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/micro_ros_arduino.h:40: undefined reference to `rmw_uros_set_custom_transport'
/tmp/arduino_build_130983/sketch/sketch_oct19b.ino.cpp.o: In function `setup':
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:49: undefined reference to `rcutils_get_default_allocator'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:52: undefined reference to `rclc_support_init'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:55: undefined reference to `rclc_node_init_default'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:58: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:58: undefined reference to `rclc_publisher_init_default'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:66: undefined reference to `rclc_timer_init_default'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:73: undefined reference to `rclc_executor_init'
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:74: undefined reference to `rclc_executor_add_timer'
/tmp/arduino_build_130983/sketch/sketch_oct19b.ino.cpp.o: In function `loop':
/home/avi/Downloads/sketch_oct19b/sketch_oct19b.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 4.1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

And now this seems to be the same issue as addressed [in this thread](https://github.com/micro-ROS/micro_ros_arduino/issues/819). Problem is I'm having trouble understanding from the answers there what I have to change -- or even if it is relevant since I'm using a different board.

pablogs9, 2022-10-26:
@Acuadros95 can you take a quick look?

avielbr, 2022-10-26:
@Acuadros95 before you put time into this I am going to try a fresh install of the micro-ROS Arduino library as per https://github.com/micro-ROS/micro_ros_arduino/issues/684#issuecomment-1012570369 and then update.

fraxker, 2023-01-23:
From what I have figured out, a number of paths in the `platform.txt` are out of date mostly related to compilation/tools locations. In addition, I was having issues using pluggable discover with the current `platform.txt` and updated code from the current one that is pulled when the board that is installed. I plan on opening a PR that will hopefully fix these issues but will need some more testing.

Acuadros95, 2023-02-01:
Hi! This should be solved by the following PR: https://github.com/micro-ROS/micro_ros_arduino/pull/1261

Please check it out, reopen if the problem persist.

closed 2023-02-01 by Acuadros95

avielbr, 2023-02-01:
I'll test this, thank you! Funny, you posted this just as I was thinking of revisiting this project...
Will report back

avielbr, 2023-02-17:
Hi again!

Beyond simply updating the repo on my end are there any other changes I need to perform?

I just tried compiling the example publisher again for Teensy 4.1 and see the following:

```
Linking everything together...
/home/avi/.arduino15/packages/teensy/tools/teensy-compile/5.4.1/arm/bin/arm-none-eabi-gcc -O2 -Wl,--gc-sections,--relax -T/home/avi/.arduino15/packages/teensy/hardware/avr/1.57.2/cores/teensy4/imxrt1062_t41.ld -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/sketch_feb17a.ino.elf /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/sketch/sketch_feb17a.ino.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/micro_ros_arduino-humble/default_transport.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/micro_ros_arduino-humble/native_ethernet_transport.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/micro_ros_arduino-humble/wifi_transport.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/NativeEthernet/NativeDns.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/NativeEthernet/NativeEthernet.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/NativeEthernet/NativeEthernetClient.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/NativeEthernet/NativeEthernetServer.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/NativeEthernet/NativeEthernetUdp.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/NativeEthernet/NativeMdns.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/NativeEthernet/Nativesocket.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/cpu/fnet_cpu.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/cpu/mimxrt/fnet_mimxrt.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/cpu/mimxrt/fnet_mimxrt_eth.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/cpu/mimxrt/fnet_mimxrt_isr.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/cpu/mimxrt/fnet_mimxrt_isr_inst.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/cpu/mimxrt/fnet_mimxrt_serial.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/cpu/mimxrt/fnet_mimxrt_serial.cpp.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/fnet_usb.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/fnet_usb_config.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/port/netif/fec/fnet_fec.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/autoip/fnet_autoip.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/azure/fnet_azure_lock.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/azure/fnet_azure_platform.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/azure/fnet_azure_socketio.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/azure/fnet_azure_threadapi.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/azure/fnet_azure_tickcounter.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/azure/fnet_azure_tlsio.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/azure/fnet_azure_tlsio_socketio.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/bench/fnet_bench_cln.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/bench/fnet_bench_srv.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/dhcp/fnet_dhcp.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/dhcp/fnet_dhcp_cln.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/dhcp/fnet_dhcp_srv.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/dns/fnet_dns.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/flash/fnet_flash.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/fnet_service.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/fs/fnet_fs.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/fs/fnet_fs_rom.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/fs/fnet_fs_root.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/http/fnet_http_cln.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/http/fnet_http_srv.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/http/fnet_http_srv_auth.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/http/fnet_http_srv_cgi.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/http/fnet_http_srv_get.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/http/fnet_http_srv_post.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/http/fnet_http_srv_ssi.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/link/fnet_link.c.o /tmp/arduino-sketch-EFA9F4024431563F2882A0E1C601D9EF/libraries/FNET/service/llmnr/fnet_llmnr.c.o [... cut ...]
/tmp/.arduinoIDE-unsaved2023117-135796-1h5wni6.xky4/sketch_feb17a/sketch_feb17a.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status

Using library micro_ros_arduino at version 2.0.5-humble in folder: /home/avi/Arduino/libraries/micro_ros_arduino-humble 
Using library NativeEthernet at version 1.0.5 in folder: /home/avi/.arduino15/packages/teensy/hardware/avr/1.57.2/libraries/NativeEthernet 
Using library FNET at version 0.1.3 in folder: /home/avi/.arduino15/packages/teensy/hardware/avr/1.57.2/libraries/FNET 
exit status 1

Compilation error: exit status 1
```

Acuadros95, 2023-02-17:
You need to reapply the [Patch teensyduino](https://github.com/micro-ROS/micro_ros_arduino#patch-teensyduino) steps.

Also, make sure that you installed  Teensyduino with the current supported approach: [link](https://www.pjrc.com/arduino-ide-2-0-0-teensy-support/).

JaYNits, 2023-07-10:
I am also facing the same issue, any solution?