micro-ROS/micro_ros_arduino #1197: microros build : could not determine ref type,  find: cannot delete ‘/project/src/’: Directory not empty

Jenifen, 2022-10-31:
## Issue template

- Hardware description: Portenta H7 machine control 
- Installation type: Docker
- Version or commit hash: foxy

I'm trying to build the precompiled library, i did the following:
1. git clone -b foxy https://github.com/micro-ROS/micro_ros_arduino
2. sudo docker pull microros/micro_ros_static_library_builder:foxy
3. sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p portenta-m7

However, it seems that it cannot overwrite files present and the tracetool fail to clone : 
[image]

pablogs9, 2022-11-02:
Can you paste here the whole console output?

Jenifen, 2022-11-03:
Somehow, the problem was from my network connection!  
Thanks

closed 2022-11-03 by Jenifen