micro-ROS/micro_ros_arduino #1221: Unusual publication duration: publishing with rcl_publish() takes 100~200ms

z2010012286, 2022-11-27:
## Issue template

- Hardware description: <--esp32 dev -->
- Installation type: <-- micro_ros_arduino#v2.0.5-foxy  -->
- Computer: ROS2 Foxy / Ubuntu 20.04

#### Steps to reproduce the issue

The main.cpp can be seen in https://github.com/z2010012286/Robot_car/blob/main/codes/reconfig_code/unit_test/unit_test_micro_ros/src/main.cpp
In my own code, my node publishes `nav_msgs__msg__Odometry` and `sensor_msgs__msg__Imu` message.
- the time before the function rcl_publish() with function millis(),
- the time just after rcl_publish() with function millis()
- the code is below
[image]
- I measured a "publication duration" most between 7ms and 11ms.
- **However, there is always unusual "publication duration" up to 100~200ms**
- Test results about "publication duration" as below.
```
18:03:23.626 -> imu_publisher RCSOFTCHECK duration: 7
18:03:23.638 -> odom_publisher RCSOFTCHECK duration: 11
18:03:23.651 -> imu_publisher RCSOFTCHECK duration: 7
18:03:23.664 -> odom_publisher RCSOFTCHECK duration: 11
18:03:23.677 -> imu_publisher RCSOFTCHECK duration: 8
18:03:23.696 -> imu_publisher RCSOFTCHECK duration: 8
18:03:23.716 -> odom_publisher RCSOFTCHECK duration: 20
18:03:23.729 -> imu_publisher RCSOFTCHECK duration: 7
18:03:23.959 -> imu_publisher RCSOFTCHECK duration: 218
18:03:23.977 -> odom_publisher RCSOFTCHECK duration: 13
18:03:24.201 -> imu_publisher RCSOFTCHECK duration: 219
18:03:24.219 -> odom_publisher RCSOFTCHECK duration: 13
18:03:24.232 -> imu_publisher RCSOFTCHECK duration: 7
18:03:24.353 -> imu_publisher RCSOFTCHECK duration: 113
18:03:24.373 -> odom_publisher RCSOFTCHECK duration: 19
18:03:24.384 -> imu_publisher RCSOFTCHECK duration: 7
18:03:24.398 -> imu_publisher RCSOFTCHECK duration: 8
18:03:24.412 -> odom_publisher RCSOFTCHECK duration: 14
18:03:24.425 -> imu_publisher RCSOFTCHECK duration: 7
18:03:24.446 -> imu_publisher RCSOFTCHECK duration: 9
18:03:24.570 -> odom_publisher RCSOFTCHECK duration: 123
18:03:24.582 -> imu_publisher RCSOFTCHECK duration: 7
18:03:24.596 -> odom_publisher RCSOFTCHECK duration: 13
18:03:24.609 -> imu_publisher RCSOFTCHECK duration: 7
18:03:24.729 -> odom_publisher RCSOFTCHECK duration: 118
18:03:24.747 -> odom_publisher RCSOFTCHECK duration: 13
18:03:24.760 -> imu_publisher RCSOFTCHECK duration: 7
18:03:24.781 -> odom_publisher RCSOFTCHECK duration: 20
18:03:24.798 -> imu_publisher RCSOFTCHECK duration: 8
18:03:24.809 -> odom_publisher RCSOFTCHECK duration: 12
18:03:24.821 -> imu_publisher RCSOFTCHECK duration: 8
18:03:24.836 -> imu_publisher RCSOFTCHECK duration: 9
18:03:24.851 -> imu_publisher RCSOFTCHECK duration: 10
18:03:24.973 -> odom_publisher RCSOFTCHECK duration: 121
18:03:24.990 -> odom_publisher RCSOFTCHECK duration: 12
18:03:25.005 -> imu_publisher RCSOFTCHECK duration: 9
18:03:25.133 -> odom_publisher RCSOFTCHECK duration: 123
18:03:25.146 -> imu_publisher RCSOFTCHECK duration: 8
18:03:25.156 -> odom_publisher RCSOFTCHECK duration: 13
18:03:25.169 -> imu_publisher RCSOFTCHECK duration: 7
18:03:25.182 -> odom_publisher RCSOFTCHECK duration: 12
18:03:25.201 -> imu_publisher RCSOFTCHECK duration: 9
18:03:25.213 -> odom_publisher RCSOFTCHECK duration: 12
18:03:25.331 -> imu_publisher RCSOFTCHECK duration: 113
18:03:25.344 -> odom_publisher RCSOFTCHECK duration: 12
18:03:25.359 -> imu_publisher RCSOFTCHECK duration: 9
18:03:25.372 -> odom_publisher RCSOFTCHECK duration: 11
18:03:25.384 -> imu_publisher RCSOFTCHECK duration: 7
18:03:25.397 -> imu_publisher RCSOFTCHECK duration: 7
18:03:25.412 -> odom_publisher RCSOFTCHECK duration: 11
18:03:25.426 -> imu_publisher RCSOFTCHECK duration: 8
18:03:25.447 -> imu_publisher RCSOFTCHECK duration: 7
18:03:25.453 -> imu_publisher RCSOFTCHECK duration: 8
```

#### Additional information
- the serial baudrate is set to 921600.
- publication frequency of `nav_msgs__msg__Odometry` message is set to 20Hz . Qos use `rclc_publisher_init_default`
- publication frequency of `sensor_msgs__msg__Imu` message is set to 50Hz . Qos use `rclc_publisher_init_default`
- Due to the unusual "publication duration", the publication frequency is not correct.
- Using `rclc_publisher_init_best_effort` can increase publication frequency of `sensor_msgs__msg__Imu` message, however, unusual "publication duration" is still existed.
- It seems to be error when using `rclc_publisher_init_best_effort` to publish `nav_msgs__msg__Odometry` message. Maybe because the  size of `nav_msgs__msg__Odometry` message is about 700 bytes that more than 500 bytes(UCLIENT_SERIAL_TRANSPORT_MTU).

Looking forward to your reply. Thank you very much!

pablogs9, 2022-11-28:
- For those rates, use `rclc_publisher_init_best_effort` for both publishers
- [Rebuild the library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library) with [UCLIENT_SERIAL_TRANSPORT_MTU](https://github.com/micro-ROS/micro_ros_arduino/blob/9ff226f5a115f902fa50873164c412f1b1d148f8/extras/library_generation/colcon.meta#L30-L38) set to something that fits all your messages (750 B ?)
- This is not related to micro-ROS, but PlatformIO + Arduino + ESP32 uses FreeRTOS under the hood. We must be sure that there is no other task with a higher priority blocking this one.

Report your results after these two approaches have been taken.

z2010012286, 2022-11-29:
@pablogs9 Thanks for your reply.
I'm trying to rebuild the libraby by following steps:
```
git clone -b foxy https://github.com/micro-ROS/micro_ros_arduino
cd micro_ros_arduino
docker pull microros/micro_ros_static_library_builder:foxy
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p esp32
```
However, it is not sucessful. 
```
dos2unix: converting file /project/extras/library_generation/library_generation.sh to Unix format...
Get:1 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]     
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]     
Get:3 http://packages.ros.org/ros2/ubuntu focal InRelease [4685 B]             
Get:4 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [1156 kB]  
Get:5 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [27.5 kB]
Get:6 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [967 kB]
Get:7 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]        
Get:8 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]      
Get:9 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:10 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:11 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [2315 kB]
Get:12 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [1712 kB]
Get:13 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]    
Get:14 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1268 kB]
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [30.2 kB]
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [2786 kB]
Get:18 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1829 kB]
Get:19 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [27.4 kB]
Get:20 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [55.2 kB]
Fetched 25.6 MB in 1min 11s (363 kB/s)                                         
Reading package lists... Done
Building dependency tree       
Reading state information... Done
227 packages can be upgraded. Run 'apt list --upgradable' to see them.
Creating firmware for generate_lib platform generic
/project/extras/library_generation/library_generation.sh: line 41: pushd: firmware/mcu_ws: No such file or directory
Cloning into 'geometry2'...
remote: Enumerating objects: 15267, done.
remote: Counting objects: 100% (2620/2620), done.
remote: Compressing objects: 100% (558/558), done.
remote: Total 15267 (delta 2289), reused 2210 (delta 2051), pack-reused 12647
Receiving objects: 100% (15267/15267), 3.48 MiB | 4.42 MiB/s, done.
Resolving deltas: 100% (10684/10684), done.
cp: cannot create directory 'ros2/tf2_msgs': No such file or directory
=== ./control_msgs (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/ros-controls/control_msgs/': GnuTLS recv error (-54): Error in the pull function.
/project/extras/library_generation/library_generation.sh: line 55: popd: directory stack empty
find: cannot delete ‘/project/src/’: Directory not empty
Crosscompiled environment: cleaning path
/uros_ws/install/micro_ros_setup/lib/micro_ros_setup/build_firmware.sh: line 65: /uros_ws/firmware/dev_ws/install/setup.bash: No such file or directory
find: ‘firmware/build/include/’: No such file or directory
cp: cannot stat 'firmware/build/include/*': No such file or directory
cp: cannot stat 'firmware/build/libmicroros.a': No such file or directory
find: ‘firmware/mcu_ws/ros2’: No such file or directory
find: ‘firmware/mcu_ws/extra_packages’: No such file or directory
```
Addition: "git clone" is failed sometimes due to the internet fire wall. (I'm in Chinese Mainland, I can't directly access to Github, Youtube, Twitter etc ..I have VPN, but it is not stable ). However,  I think this is not the main reason for above errors.
Could you give me some advices? Thanks!

pablogs9, 2022-11-29:
No problem found on the process: https://asciinema.org/a/3U6dd5Ule1f3Fg4ewVOAjd4wr

It is related to access to Github I guess.

z2010012286, 2022-11-29:
ok, I will try it again later and will be reopen if there is new test results

z2010012286, 2022-12-02:
Hi,   @pablogs9, I have tried these advices, but there is still some problems.
- I  used  `rclc_publisher_init_best_effort` for both publishers
- I have rebuilt the library with  "`-DUCLIENT_CUSTOM_TRANSPORT_MTU=1024`" and "`-DRMW_UXRCE_MAX_HISTORY=2`"

- Luckly, it is works. However, the publication rate is still low. 
[image]

- I also tested the odom_publish duration with following code, the result can be seen in [odom_publish_duration.txt](https://github.com/z2010012286/Robot_car/blob/77b1dca0d153084025840b940e19405077c43677/codes/reconfig_code/unit_test/unit_test_micro_ros/test/odom_publish_duration.txt#L1) . The `odom_publisherCallback` did not work well. 
```
void odom_publisherCallback(rcl_timer_t *timer, int64_t last_call_time)
{
    unsigned long start = millis();
    RCLC_UNUSED(last_call_time);
    if (timer != NULL)
    {
        publishOdomData();
    }
    Serial1.print("running time: ");
    Serial1.print(start);
    Serial1.print("     ");
    unsigned long duration = millis() - start;
    Serial1.print("odom_publish duration: ");
    Serial1.println(duration);
}
```
- Part of debug information (`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -b 921600 -v6`) can be seen in [micro_ros_debug.txt](https://github.com/z2010012286/Robot_car/blob/77b1dca0d153084025840b940e19405077c43677/codes/reconfig_code/unit_test/unit_test_micro_ros/test/micro_ros_debug.txt#L1)
- The actual publication rates can  be seen in :[imu_hz.txt](https://github.com/z2010012286/Robot_car/blob/77b1dca0d153084025840b940e19405077c43677/codes/reconfig_code/unit_test/unit_test_micro_ros/test/imu_hz.txt#L1) and [odom_hz.txt](https://github.com/z2010012286/Robot_car/blob/77b1dca0d153084025840b940e19405077c43677/codes/reconfig_code/unit_test/unit_test_micro_ros/test/odom_hz.txt#L1)

- The main.cpp can be seen in [main.cpp](https://github.com/z2010012286/Robot_car/blob/77b1dca0d153084025840b940e19405077c43677/codes/reconfig_code/unit_test/unit_test_micro_ros/src/main.cpp#L1)
Could you give me some advices? Thank you very much!

> * For those rates, use `rclc_publisher_init_best_effort` for both publishers
> * [Rebuild the library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library) with [...]

pablogs9, 2022-12-05:
What happens if you remove the agent check code and keep only the spin? https://github.com/z2010012286/Robot_car/blob/77b1dca0d153084025840b940e19405077c43677/codes/reconfig_code/unit_test/unit_test_micro_ros/src/main.cpp#L168-L196

z2010012286, 2022-12-05:
Thanks for your reply, @pablogs9 
I have removed the agent check code and kept only the spin as below:
```
void loop()
{
    rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
}
```
- Baudrate is set to 921600;
- It works. The publication rate is higher than before, and it seems that the connection is more stable.
- However, it still doesn't achieve the publication rate I want. (odom:20Hz, imu:50Hz)
[image]

pablogs9, 2022-12-05:
Change both publishers to `std_msgs/Int32` and check the rates. If this works at your expected rates, maybe your message types are too bit for the throughput of your transport.

For example, check that the `sensor_msgs/IMU` has 27 `uint64_t` values 
[image]

z2010012286, 2022-12-05:
Thanks for you reply @pablogs9 
I guess I found the cause of this problem. I used the [USB to Serial Port Chip CH340](http://www.wch-ic.com/products/CH340.html), and it seems not work well in 921600bps.(I guess)
When I used the WiFi instead of serial port, everything is OK!!
`set_microros_wifi_transports("zhang", "********", "192.168.50.93", 8888);`
`ros2 run micro_ros_agent micro_ros_agent udp4 -p 8888`

[image]

closed 2022-12-05 by z2010012286