micro-ROS/micro_ros_arduino #1230: Serial RX on serial1-8 break on Teensy 4.1 when init_options are created

sknipas, 2022-12-06:
I have a created a dumb flight controller firmware for teensy 4.1 with no autopilot functions.
Serial7 reads NMEA gps messages, serial5 is used for mavlink communication , serial8 is sbus.
Then I tried to also add micro-ros to communicate with a ros2 node running on a rpi.
It worked perfectly but I then realised that all communication with gps,mavlink and sbus was broken.
Everything works as expected (Except ros2 comms) when I comment RCCHECK(rclc_support_init(&support, 0, NULL, &allocator)); line.
Looks like TX works but RX doesn't on all serial ports.
- Hardware description:  Teensy 4.1

pablogs9, 2022-12-06:
You re mixing micro-ROS transport with something else. This is not a micro-ROS issue.

Please ensure that the micro-ROS client has a transport (not shared with anything else) that communicates with the micro-ROS Agent.

closed 2022-12-06 by pablogs9

pablogs9, 2022-12-06:
So, you issue is that your Teensy or your code does not handle multiple serial ports opened at the same time?