micro-ROS/micro_ros_arduino #1241: Configuring WiFi transport for Portenta H7

avielbr, 2022-12-21:
Hi, I'm aware that I'm asking a lot of questions here. Sorry. I'm happy to organize whatever answers I receive here and make a tutorial writeup for the community.

I'm a bit stuck on getting the WiFi transport working for the Portenta board. How can I adjust the steps from [this](https://medium.com/@SameerT009/connect-esp32-to-ros2-foxy-5f06e0cc64df) guide to work with the Arduino precompiled library?

I imagine I would need to make an adjustment to the `library_generation.sh` file, but I'm not sure what.

Thanks in advance for any and all help, you guys rock!

pablogs9, 2022-12-21:
Hello @avielbr, I guess that you do not need to rebuild the library for changing the transports. Just use the correct function to set the transports: https://github.com/micro-ROS/micro_ros_arduino/blob/dc99a9f3663fab6ec5ff809a6b5310e0e40e4130/src/micro_ros_arduino.h#L119

Let us know if this works

avielbr, 2022-12-21:
Thanks @pablogs9, this does work. I switched networks and this worked right away - guess there is something weird in the configuration of the first network I tried.

Have a good one.

closed 2022-12-21 by avielbr