micro-ROS/micro_ros_arduino #1242: How to set the domain ID for the node?

larsoss15, 2022-12-22:
- Hardware description: Arduino DUE
- Operating system: Ubuntu 18.04
- Version or commit hash: foxy

Currently the node is using the default domain ID. We are using ID 103 and I am wondering how to configure the node. I found a 'solution' with rcl_node_options_t, but I have no idea how to install/implement this. Can anyone tell me the steps to set the domain ID for the node?

pablogs9, 2022-12-22:
https://github.com/micro-ROS/micro-ROS-demos/blob/45f342d4ef887f6a2e9f2dfa45d1d3388d7f6dbd/rclc/configuration_example/configured_publisher/main.c#L49

closed 2022-12-22 by pablogs9

larsoss15, 2022-12-23:
Thanks! I am able to change the domain ID. If I use the command ros2 topic info I can see the publisher on the correct domain ID. But when using the ros2 topic echo no data is shown. When I use the same code for the default domain id I get data when using the ros2 topic echo command. Do I need to change the code anywhere else, maybe in the publisher settings? Below you can finde the code:

`
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

#include <OneWire.h>
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;
rclc_executor_t executor;
rclc_support_t support;
//rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

int id_temp1 = 1;
int id_temp2 = 2;

size_t domain_id = 103;

OneWire  ds1(10);
OneWire  ds2(11);

LiquidCrystal_I2C lcd(0x27,16,2);
String txt;

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(1000);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    sendMessage(id_temp1, int(getTemp(ds1)));
    sendMessage(id_temp2, int(getTemp(ds2)));
  }
}

void setup() {
  lcd.init();
  lcd.backlight();
  Serial.begin(9600);
  lcd.backlight();
  lcd.setCursor(0,0);
  txt = String("Left   ") + (char)223 + "C" + "  Right";
  lcd.print(txt);
  
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  rcl_allocator_t allocator = rcl_get_default_allocator();

  //Define init options
  rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();

  RCCHECK(rcl_init_options_init(&init_options, allocator));
  rmw_init_options_t* rmw_options = rcl_init_options_get_rmw_init_options(&init_options);

  RCCHECK(rcl_init_options_set_domain_id(&init_options, domain_id));

  // create init_options
  RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "micro_ros_arduino_node_publisher"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
    timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  //msg.data = 0;
}

void loop() {
  delay(100);
  
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

float getTemp(OneWire ds)
{
  byte i = 0;
  byte present = 0;
  byte type_s;
  byte data[9];
  byte addr[8];
  float celsius;
  
  if ( !ds.search(addr)) {
    ds.reset_search();
    delay(250);
    return 10.0;
  }
 
  switch (addr[0]) {
    case 0x10:
      type_s = 1;
      Serial.println(type_s);
      break;
    case 0x28:
      type_s = 0;
      break;
    case 0x22:
      type_s = 0;
      break;
    default:
      Serial.println("Device is not a DS18x20 family device.");
      return 10.0;
    
  } 

  ds.reset();
  ds.select(addr);
  ds.write(0x44, 1);        // start conversion, with parasite power on at the end
  
  delay(1000);     // maybe 750ms is enough, maybe not
  
  present = ds.reset();
  ds.select(addr);    
  ds.write(0xBE);         // Read Scratchpad

  for (i = 0; i < 9; i++) {           // we need 9 bytes
    data[i] = ds.read();
  }

  int16_t raw = (data[1] << 8) | data[0];
  if (type_s) {
    raw = raw << 3;
    if (data[7] == 0x10) {
      raw = (raw & 0xFFF0) + 12 - data[6];
    }
  } else {
    byte cfg = (data[4] & 0x60);
    if (cfg == 0x00) raw = raw & ~7;  // 9 bit resolution, 93.75 ms
    else if (cfg == 0x20) raw = raw & ~3; // 10 bit res, 187.5 ms
    else if (cfg == 0x40) raw = raw & ~1; // 11 bit res, 375 ms
  }
  celsius = (float)raw / 16.0;
  return celsius;
}

void sendMessage(int id, int message)
{
  msg.data = id * 10000 + message;
  RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
  delay(1000);
}
`

pablogs9, 2022-12-23:
How are you listening to the topic? Are you setting `ROS_DOMAIN_ID` also in the `ros2 topic echo...`?

larsoss15, 2022-12-23:
Yes, in the terminal I give the commant: export ROS_DOMAIN_ID=103

pablogs9, 2022-12-23:
Does it works if you use domain id 2?

larsoss15, 2022-12-23:
No it does not work either

`
lars@ldellemann:~/Projects/driving/kosmonaut-ros2$ export ROS_DOMAIN_ID=2
lars@ldellemann:~/Projects/driving/kosmonaut-ros2$ ros2 topic info /micro_ros_arduino_node_publisher 
Type: std_msgs/msg/Int32
Publisher count: 1
Subscription count: 0
lars@ldellemann:~/Projects/driving/kosmonaut-ros2$ ros2 topic echo /micro_ros_arduino_node_publisher 
`

pablogs9, 2022-12-23:
Can you share the output of the micro ros agent using -v6 flag?

larsoss15, 2022-12-23:
Is this what you mean? (used: ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6) This gives:

`
[1671788945.559734] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1671788945.559937] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1671788948.462132] info     | Root.cpp           | create_client            | create                 | client_key: 0x5147DE24, session_id: 0x81
[1671788948.462294] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x5147DE24, address: 0
[1671788948.462522] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5147DE24, len: 19, data:
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1671788948.473902] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5147DE24, len: 56, data:
0000: 81 80 00 00 01 07 2E 00 00 0A 00 01 01 03 00 00 1F 00 00 00 00 01 08 20 17 00 00 00 6D 69 63 72
0020: 6F 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 00 00 02 00 00 00
[1671788948.499421] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x5147DE24, participant_id: 0x000(1)
[1671788948.499692] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5147DE24, len: 14, data:
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1671788948.499750] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5147DE24, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1671788948.506894] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5147DE24, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1671788948.514902] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5147DE24, len: 100, data:
0000: 81 80 01 00 01 07 5A 00 00 0B 00 02 02 03 00 00 4C 00 00 00 24 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 5F 70 75 62 6C 69 73 68 65 72 00 00 01 09 00
0040: 1C 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 00
0060: 00 01 00 00
[1671788948.515247] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x5147DE24, topic_id: 0x000(2), participant_id: 0x000(1)
[1671788948.515416] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5147DE24, len: 14, data:
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1671788948.515458] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5147DE24, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1671788948.523174] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5147DE24, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1671788948.523233] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5147DE24, len: 24, data:
0000: 81 80 02 00 01 07 10 00 00 0C 00 03 03 03 00 00 02 00 00 00 00 00 00 01
[1671788948.523472] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x5147DE24, publisher_id: 0x000(3), participant_id: 0x000(1)
[1671788948.523630] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5147DE24, len: 14, data:
0000: 81 80 02 00 05 01 06 00 00 0C 00 03 00 00
[1671788948.523690] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5147DE24, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1671788948.531648] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5147DE24, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1671788948.535200] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5147DE24, len: 36, data:
0000: 81 80 03 00 01 07 1C 00 00 0D 00 05 05 03 00 00 0E 00 00 00 00 02 01 00 03 00 01 00 0A 00 00 00
0020: 00 00 00 03
[1671788948.536182] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x5147DE24, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1671788948.536345] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5147DE24, len: 14, data:
0000: 81 80 03 00 05 01 06 00 00 0D 00 05 00 00
[1671788948.536402] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5147DE24, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1671788948.543455] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5147DE24, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80

`

pablogs9, 2022-12-23:
Your micro-ROS app is creating the entities and the publisher but it is not publishing, that is why you do not see any data in the echo command.

1. Make your `rcl_allocator_t allocator` a global static variable, not a setup stack variable
2. Do not   delay(1000) in the executor timer callback
3. Ensure that your code is reaching `rcl_publish`

larsoss15, 2022-12-23:
Nice, it is working! Thanks for you help!!

pablogs9, 2022-12-23:
What is was?

larsoss15, 2022-12-23:
Changing the rcl_allocator_t allocator to a global static variable