micro-ROS/micro_ros_arduino #1247: micro ros comunication problem

BenSwisa, 2023-01-08:
hello,
i am having a comunication problem .
i am running a launch file of 3 micro_ros_agents for 3 usb ports that i have : 2 micro_controllers publishing and 1 subscribing.
- when running the launch file than connecting all the devices at once(via a usb-splitting device) i get the following error for 2 out of the 3 ports : 
 open device error      | device: /dev/ttyACM0, errno: 13. Please re-run with superuser

*even though my user has permission to the usb ports*

- when running the launch file than connecting each device once at a time to my ports i have no problem.

for my project i need to connect all my devices at one time and i have no idea why this is happening.

would appreciate any help , thanks in advance

pablogs9, 2023-01-09:
Does your setup works if you run `ros2 micro_ros_agent micro_ros_agent serial [your serial device] -v6`?

pablogs9, 2023-01-09:
Which is the error running the command directly?

pablogs9, 2023-01-09:
Which is the ouput when you run this command `ros2 micro_ros_agent micro_ros_agent serial [your serial device] -v6` ?

pablogs9, 2023-01-09:
What about `sudo ros2 micro_ros_agent micro_ros_agent serial [your serial device] -v6`?

BenSwisa, 2023-01-10:
i am getting the following error

sudo ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyAMC0 
sudo: ros2: command not found

pablogs9, 2023-01-10:
I guess that you have permission issues with your serial ports. Maybe you can find how to ensure that your user has permission to read and write the serial port.

BenSwisa, 2023-01-10:
the thing is is i plug the devices one by one i have no user permission issues

BenSwisa, 2023-01-15:
hey I tried it and it worked , what is the reason it works this way and the other way not?

pablogs9, 2023-01-16:
Issue with your device's permissions? I really don't know.

Closing as solved.

closed 2023-01-16 by pablogs9