micro-ROS/micro_ros_arduino #1249: Micro ROS arduino Humble on RP2040

djaniel, 2023-01-09:
## Micro ROS arduino Humble on RP2040

- Hardware description: Arduino RP2040 connect
- RTOS: NA
- Installation type: Arduino
- Version or commit hash: Humble 2.0.5

#### Steps to reproduce the issue
I have a magnetometer connected to the Arduino RP2040 connect. The code running in the arduino is based on the micro-ros_publisher example included on this repository. I was doing all the testing running Ros 2 Foxy in a container and everything went well. I was getting a sensor_msgs Magnetic Field topic from the arduino. Then I connected the arduino to an embedded computer Ubuntu server 22 - Ros 2 Humble. 

During installation of the agent in the embedded computer I got a warning that I didn't see when working with foxy:

```
#ros2 run micro_ros_setup build_agent.sh 
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [4.14s]                     
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                                   
Cloning into 'xrceagent'...
Switched to a new branch 'ros2'
HEAD is now at 3eb56b5 Release v2.3.0
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< micro_ros_agent [28.8s]

Summary: 2 packages finished [33.0s]
  1 package had stderr output: micro_ros_agent
```
I think it is not big deal. Am I wrong?

Then, for testing I did:
```
ros2 run micro_ros_agent micro_ros_agent serial --dev [serial device] -v6
```
The output :
```
[1673270674.986457] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1673270674.986766] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```
And that is all! As you can see, there are not the whole list of messages we usually see when everything goes well. Apparently, the agent does not connect to the arduino nano.

Trying to solve my problem I came back to docker, this time with an image of Ros 2 Humble. I see the same warning I mentioned earlier and the agent still does not connect to  the Arduino.

Unsuccessful last resort:
```
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev /dev/ttyACM0 -v6
```
#### Expected behavior
That the connection is accomplished and the data from the magnetometer is published in ROS 2 Humble. It strikes me that everything goes well when using foxy.
Can you give some tips on how to find the answer to this dilemma?

pablogs9, 2023-01-09:
Hello @djaniel, the warning that you are seeing during compilation is not relevant.

Regarding the operation of the agent, if I have understood correctly, after building your Arduino sketch in the RP2040, it works with 

```bash
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
```

but it does not work with

```bash
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev /dev/ttyACM0 -v6
```

Is this correct?

djaniel, 2023-01-09:
My goodness! I don't get it. I connect the aduino to my computer and it does not work straight away. I reset the board, the agent notices the port disappearing and they launch perfectly. The problem is that having to reset the board everytime I launch the system is impractical. Do you know if I am missing something on the code?

pablogs9, 2023-01-10:
By default micro-ROS client does not handle reconnections with the agent. You can check an example on how to do this here: https://github.com/micro-ROS/micro_ros_arduino/tree/humble/examples/micro-ros_reconnection_example

I'm closing since the main issue is solved. Feel free to open new issues if you have more problems.

closed 2023-01-10 by pablogs9