micro-ROS/micro_ros_arduino #1251: uRos node not listing

djaniel, 2023-01-10:
- Hardware description: Rasperry PI connected to RP2040 nano connect
- RTOS: NA
- Version or commit hash: Humble

Hello again,

I am running uROS on a RP2040 connect. As usual, for testing I am using docker which launches the micro ros agent in humble:

``
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev /dev/ttyACM0 -v4
``

Everything works fine on my pc. When I go open a new bash on the image I can display the uros node and topic. I can display the contents of the message. 

Then I connected my RP2040 to an embedded computer with Ubuntu server 22, Ros2 Humble. The problem is that the launch process seems to go well, I set -v5 and I see a lot of :

``
[1673362828.144831] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 120
[1673362828.247510] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x701B8528, len: 132
[1673362828.247738] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 120
[1673362828.247881] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x701B8528, len: 16
``

But unfortunately and after launching the uros agent, the commands for listing the topics and the nodes do not display anything. 

``
$ros2 node list
$ros2 topic list 
/parameter_events
/rosout
``

Any suggestions would be greatly appreciated.

pablogs9, 2023-01-10:
how are you running the dockers where you run `ros2 node list`?

pablogs9, 2023-01-10:
Are you changing the domain id in the micro-ROS client?

pablogs9, 2023-01-11:
The micro-ROS agent does not modify the domain id. Is the client who determines which domain id will use. 

Could you check if setting your ROS 2 environment to domains id 0 it works?

djaniel, 2023-01-11:
Thank for your answer. I was wrong. Using uRos for Humble, I changed the domain id using the following:
``
  size_t domain_id = ;
  //Define init options
  rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();

  RCCHECK(rcl_init_options_init(&init_options, allocator));
  rmw_init_options_t* rmw_options = rcl_init_options_get_rmw_init_options(&init_options);

  RCCHECK(rcl_init_options_set_domain_id(&init_options, domain_id));
``

After programming the board, I don't see anymore my node on the container and luckily it works fine on my RPI. Thank you.