micro-ROS/micro_ros_arduino #1259: No ROS2 topics and nodes on example publisher

MrOCW, 2023-01-21:
## Issue template

- Hardware description: ESP32
- RTOS: NA
- Installation type: Docker
- Version or commit hash: humble
- 
#### Steps to reproduce the issue
Upload micro-ros_publisher.ino onto an ESP32
Run `docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev /dev/ttyUSB0 -v6`
Enter the container and run ros2 topic list

#### Expected behavior
Topics and Nodes appear as expected

#### Actual behavior
No communication with ros2

#### Additional information
[image]

pablogs9, 2023-01-23:
Where are you running the `ros2` command? in another docker?

MrOCW, 2023-01-23:
@pablogs9 yes, its in the exact same container running `docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev /dev/ttyUSB0 -v6`

pablogs9, 2023-01-24:
Are you setting some domain in the micro-ROS Client?

MrOCW, 2023-01-25:
no. everything is default

pablogs9, 2023-01-25:
Does it works in a local ROS 2 installation or in a separate docker running ROS 2?

MrOCW, 2023-01-25:
I have not tried a local ROS2 installation. everything is running in docker

pablogs9, 2023-01-25:
Well, it seems to be something related to Fast DDS communication between the micro-ROS Agent and ROS 2 because the communication between micro-ROS Client and Agent is correct as far as you can see outcoming DDS packets in the agent log `<<DDS>>`.

With the provided information I cannot determine which can be the cause.

pablogs9, 2023-01-25:
In any case, try to add `--ipc=host` in the docker launch, to check if it related with Shared Memory discovery between Fast DDS Instances.

Franklein20, 2023-02-01:
i run it on my host machine but i didn't see the topic.
Could you help me?

anasderkaoui, 2024-01-03:
Hello, I had the same issue but not running docker. I used the simple way and the topic and node were not showing. The workaround is to add the ros domain ID by typing : "`export ROS_DOMAIN_ID=0`"