micro-ROS/micro_ros_arduino #1285: Unable to compile for Teensy 4.1

adityakamath, 2023-02-05:
Hi, I am trying to compile some micro-ROS code using the Humble branch on Windows 11 and I'm getting some errors. These are the steps I followed:

- Downloaded Arduino IDE and Teensyduino
- Downloaded micro_ros_arduino to the libraries directory
- Added custom message types
- Build the library using the docker commands in the README (so far these steps have worked for Galactic)

I then tried to compile my code in the Arduino IDE and got the following errors:

```
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in C:\Users\adity\OneDrive\Documents\Arduino\libraries\micro_ros_arduino\src\imxrt1062\fpv5-d16-hard
C:\Users\adity\AppData\Local\Temp\arduino_build_510594\sketch\akros2_firmware.ino.cpp.o: In function `paramCallback(rcl_interfaces__msg__Parameter const*, rcl_interfaces__msg__Parameter const*, void*)':
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:223: undefined reference to `rclc_parameter_get_double'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:224: undefined reference to `rclc_parameter_get_double'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:225: undefined reference to `rclc_parameter_get_double'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:226: undefined reference to `rclc_parameter_get_double'
C:\Users\adity\AppData\Local\Temp\arduino_build_510594\sketch\akros2_firmware.ino.cpp.o: In function `destroyEntities()':
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:345: undefined reference to `rcl_context_get_rmw_context'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:346: undefined reference to `rmw_uros_set_context_entity_destroy_session_timeout'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:348: undefined reference to `rclc_executor_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:349: undefined reference to `rcl_publisher_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:350: undefined reference to `rcl_publisher_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:351: undefined reference to `rcl_publisher_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:352: undefined reference to `rclc_executor_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:353: undefined reference to `rcl_timer_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:354: undefined reference to `rcl_subscription_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:355: undefined reference to `rcl_subscription_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:356: undefined reference to `rclc_parameter_server_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:357: undefined reference to `rcl_node_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:358: undefined reference to `rclc_support_fini'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:359: undefined reference to `rcl_init_options_fini'
C:\Users\adity\AppData\Local\Temp\arduino_build_510594\sketch\akros2_firmware.ino.cpp.o: In function `publishData()':
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:504: undefined reference to `rcl_publish'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:505: undefined reference to `rcl_publish'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:506: undefined reference to `rcl_publish'
C:\Users\adity\AppData\Local\Temp\arduino_build_510594\sketch\akros2_firmware.ino.cpp.o: In function `createEntities()':
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:241: undefined reference to `rcutils_get_default_allocator'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:242: undefined reference to `rcl_get_zero_initialized_init_options'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:243: undefined reference to `rcl_init_options_init'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:244: undefined reference to `rcl_init_options_set_domain_id'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:245: undefined reference to `rclc_support_init_with_options'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:246: undefined reference to `rclc_node_init_default'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:249: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__nav_msgs__msg__Odometry'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:249: undefined reference to `rclc_publisher_init_default'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:256: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__sensor_msgs__msg(int0_t)'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:256: undefined reference to `rclc_publisher_init_default'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:263: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__akros2_msgs__msg__Feedback'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:263: undefined reference to `rclc_publisher_init_default'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:270: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__geometry_msgs__msg__Twist'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:270: undefined reference to `rclc_subscription_init_default'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:277: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__akros2_msgs__msg__Mode'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:277: undefined reference to `rclc_subscription_init_default'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:285: undefined reference to `rclc_timer_init_default'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:298: undefined reference to `rclc_parameter_server_init_with_option'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:304: undefined reference to `rclc_executor_get_zero_initialized_executor'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:305: undefined reference to `rclc_executor_init'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:306: undefined reference to `rclc_executor_add_subscription'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:312: undefined reference to `rclc_executor_add_subscription'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:318: undefined reference to `rclc_executor_add_timer'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:319: undefined reference to `rclc_executor_add_parameter_server'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:325: undefined reference to `rclc_add_parameter'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:326: undefined reference to `rclc_add_parameter'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:327: undefined reference to `rclc_add_parameter'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:328: undefined reference to `rclc_add_parameter'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:331: undefined reference to `rclc_parameter_set_double'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:332: undefined reference to `rclc_parameter_set_double'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:333: undefined reference to `rclc_parameter_set_double'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:334: undefined reference to `rclc_parameter_set_double'
D:\akros2_ws\akros2_firmware/akros2_firmware.ino:337: undefined reference to `rmw_uros_sync_session'
[... 15 lines cut ...]
D:\akros2_ws\akros2_firmware\src\imu/imu_interface.h:61: undefined reference to `micro_ros_string_utilities_set'
C:\Users\adity\AppData\Local\Temp\arduino_build_510594\sketch\src\odometry\odometry.cpp.o: In function `Odometry::Odometry()':
D:\akros2_ws\akros2_firmware\src\odometry/odometry.cpp:22: undefined reference to `micro_ros_string_utilities_set'
D:\akros2_ws\akros2_firmware\src\odometry/odometry.cpp:23: undefined reference to `micro_ros_string_utilities_set'
collect2.exe: error: ld returned 1 exit status
Error compiling for board Teensy 4.1.
```

I realized that I had forgotten to update the platform.txt file, and after doing that I get this error:

```
exec: "/arm/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Teensy 4.1.
```

I don't know what I'm doing wrong, and I am also unable to find any info about this, especially for Windows. Any tips will be great!

## Issue template

- Hardware description: Teensy 4.1
- RTOS: -
- Installation type: micro_ros_arduino with Arduino IDE on Windows 11
- Version or commit hash: humble

#### Additional information

pablogs9, 2023-02-07:
Hello @adityakamath, some questions:
- Does the library works without adding any custom message or rebuilding the library? If it does not work which is the error?
- `/arm/bin/arm-none-eabi-g++` related error appears on Windows or in Docker? In the case of having it in Windows, does it appears also when building a basic example without micro-ROS?

adityakamath, 2023-02-07:
@pablogs9, for your first question, I did not try, I'll do that today and let you know. 

For the second question, the error is on the Arduino IDE on windows. And it does not appear when building non-microros code

pablogs9, 2023-02-07:
> For the second question, the error is on the Arduino IDE on windows. And it does not appear when building non-microros code

Not sure why is micro-ROS modifying compiler patches. Any idea @Acuadros95?

adityakamath, 2023-02-07:
My bad, the error is appearing while also compiling non-microros code. 

I realized that I was trying with a different microcontroller earlier

Acuadros95, 2023-02-07:
How did you download `Teensyduino` and what version did you install? 

The current supported approach is [this one](https://www.pjrc.com/arduino-ide-2-0-0-teensy-support/).

adityakamath, 2023-02-07:
Ah, I wasn't aware of this. I used the installer from [here](https://www.pjrc.com/teensy/td_download.html) for Arduino IDE 1.8.x

I will try your link today and get back.

adityakamath, 2023-02-07:
@Acuadros95 I tried the link you shared with Arduino IDE 2.0.3 and looks like it works! I haven't uploaded it to my Teensy yet, but the code compiles and I don't see the errors anymore. Thanks! 

I'll close this issue now.

closed 2023-02-07 by adityakamath