micro-ROS/micro_ros_arduino #1300: example: micro-ros_publisher compile error in Arduino IDE

TaeUk-An, 2023-02-20:
## Issue template

- Hardware description:  OpenCR 1.0
- RTOS:  
- Installation type:  
- Version or commit hash:  galactic

#### Steps to reproduce the issue
https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v2.0.5-galactic 
`Sketch -> Include library -> Add .ZIP Library...` 
i'm test this examples : micro-ros_publisher 

#### Expected behavior
compile complete.

#### Actual behavior
example code compile error in Arduino IDE (not upload)
--------------------------------------------------------- below part of compile error  details -----------------------------------------
`Library micro_ros_arduino has been declared precompiled:
Using precompiled library in c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status__type_support.c.obj): In function `GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_init_function':
goal_status__type_support.c:(.text.GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_init_function+0x0): multiple definition of `GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_init_function'
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status__type_support.c.obj):goal_status__type_support.c:(.text.GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_init_function+0x0): first defined here
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status__type_support.c.obj): In function `GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_fini_function':
goal_status__type_support.c:(.text.GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_fini_function+0x0): multiple definition of `GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_fini_function'
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status__type_support.c.obj):goal_status__type_support.c:(.text.GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_fini_function+0x0): first defined here
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj): In function `GoalStatusArray__rosidl_typesupport_introspection_c__size_function__GoalStatus__status_list':
goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__size_function__GoalStatus__status_list+0x0): multiple definition of `GoalStatusArray__rosidl_typesupport_introspection_c__size_function__GoalStatus__status_list'
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj):goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__size_function__GoalStatus__status_list+0x0): first defined here
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj): In function `GoalStatusArray__rosidl_typesupport_introspection_c__get_function__GoalStatus__status_list':
goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__get_function__GoalStatus__status_list+0x0): multiple definition of `GoalStatusArray__rosidl_typesupport_introspection_c__get_function__GoalStatus__status_list'
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj):goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__get_function__GoalStatus__status_list+0x0): first defined here
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj): In function `GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_init_function':
goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_init_function+0x0): multiple definition of `GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_init_function'
c:\Users\user\Documents\Arduino\libraries\micro_ros_arduino-2.0.5-galactic\src\cortex-m7\fpv5-sp-d16-softfp\libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj):goal_status_array__type_support.c:
... 

#### Additional information

pablogs9, 2023-02-20:
All the examples are being built correctly in CI in Open CR: https://github.com/micro-ROS/micro_ros_arduino/blob/737e501dcf74ea1527140cdcadb59b919ec7fab4/.github/workflows/ci.yml#LL58C13-L64C140

Which version of Arduino IDE and Open CR package do you have?
Galactic is EOL, why are you using it? Is possible to move to Humble?

TaeUk-An, 2023-02-20:
OpenCR version was low. I solved it. Thank you.

ps. I am using Nvidia Jetson AGX Xavier, so I can only use it until Ubuntu 20.04.

closed 2023-02-20 by pablogs9