micro-ROS/micro_ros_arduino #1303: incompatible QoS

patrickwasp, 2023-02-24:
## Issue template

- Hardware description: ESP32
- RTOS: 
- Installation type: precompiled library using Arduino IDE 1.8.19
- Version or commit hash: v2.0.5-humble

#### Steps to reproduce the issue
1. load micro-ros_reconnection_example
2. upload to ESP32
3. start `microros/micro-ros-agent:humble` with `serial --dev /dev/ttyUSB0 -v6`
4. run `rqt`
5. subscribe to `std_msgs_msg_int32` using Topic Monitor 

#### Expected behavior
the value of the subscribed topic counting up

#### Actual behavior
```
WARN] [1677247550.012298137] [rqt_gui_py_node_664]: New publisher discovered on topic '/std_msgs_msg_Int32', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY
```

#### Additional information

pablogs9, 2023-02-27:
Closing as solved

closed 2023-02-27 by pablogs9

patrickwasp, 2023-03-01:
would it be a good idea to change the `micro-ros_reconnection_example` example so that it works out of the box?

pablogs9, 2023-03-01:
It works out of the box with ROS 2 CLI. If using another approach you need to take care of QoS