micro-ROS/micro_ros_arduino #1304: ROS2 custom message -> library, then add to Arduino IDE library (.zip)

TaeUk-An, 2023-02-24:
## Issue template

- Hardware description:  OpenCR
- RTOS:  Linux 20.04
- Installation type:  micro_ros_setup
- Version or commit hash:  galatic

#### Steps to reproduce the issue

i want to make a library about ROS2 custom messages to add Arduino IDE custom library (.zip)
so i found this page. 
https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/
I did as it was written. but error issue.

#### Expected behavior
generate library about ROS2 custom messages (ex. ros_lib)

#### Actual behavior
`avees@ubuntu:~/micro_custom_library$ ros2 run micro_ros_setup build_firmware.sh $(pwd)/my_custom_toolchain.cmake $(pwd)/my_custom_colcon.meta
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /home/avees/micro_custom_library/my_custom_colcon.meta
Starting >>> rcutils 
Starting >>> rosidl_cli
Starting >>> rosidl_typesupport_interface                              
Starting >>> microcdr
Starting >>> rmw_implementation_cmake                                      
Starting >>> tracetools
Starting >>> tracetools_trace                                              
Starting >>> test_interface_files
--- stderr: microcdr                                                             
CMake Error at /home/avees/micro_custom_library/my_custom_toolchain.cmake:1 (et):
  Unknown CMake command "et".
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:53 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
---
Failed   <<< microcdr [2.11s, exited with code 1]
Aborted  <<< rosidl_cli [2.22s]
Aborted  <<< tracetools_trace [1.99s]
Aborted  <<< test_interface_files [1.93s]
Aborted  <<< rosidl_typesupport_interface [2.17s]
Aborted  <<< tracetools [2.03s]
Aborted  <<< rcutils [2.26s]
Aborted  <<< rmw_implementation_cmake [2.08s]

Summary: 0 packages finished [3.36s]
  1 package failed: microcdr
  7 packages aborted: rcutils rmw_implementation_cmake rosidl_cli rosidl_typesupport_interface test_interface_files tracetools tracetools_trace
  6 packages had stderr output: microcdr rcutils rmw_implementation_cmake rosidl_typesupport_interface test_interface_files tracetools
  56 packages not processed
`
#### Additional information

pablogs9, 2023-02-27:
I have not been able to replicate this error. Here my session with your same commands: https://asciinema.org/a/WLZs1hm1N3DUusXRBJQmby8Wh

TaeUk-An, 2023-02-27:
This is my host pc specification.

Nvidia AGX Jetson Xavier
jetpack version: 5.0.2
ubuntu 20.04
Docker version 20.10.12, build 20.10.12-0ubuntu2~20.04.1

Even if I run on each of the two xavier units, the error comes out in the same part.

Which part should I suspect?

pablogs9, 2023-02-27:
Could you check if this works on a `amd64` computer?

TaeUk-An, 2023-02-27:
It works on an amd64 computer.

Thank you.

closed 2023-02-27 by TaeUk-An