micro-ROS/micro_ros_arduino #1328: Getting Started

cruzaviles, 2023-03-06:
Hi there!

First off thank you all so much for the work here it is amazing!

I am a sophomore in college and I am trying to use this library to connect a laptop running ros2 foxy to an arduino due in order to control 5 servo motors (for a robotic arm). 

I am just learning how to use ros2 and linux so a lot of this is new to me. So, I seem to have gotten lost following your first short tutorial. 

I have arduino 1.8.19 installed and I followed the instructions downloading the zip from github. I then included the zip library through arduino. But, when I try to run any example code I get several "undefined reference to" errors concerning the library files. 

I have tried a few things like deleting and re-installing the library, but I can't seem to find a way to resolve this and get the file to compile. 

Please just let me know if I am missing something and thank you again!

Acuadros95, 2023-03-07:
Hi @cruzaviles, Which board are you using?

cruzaviles, 2023-03-07:
Hi @Acuadros95 thank you for your response! I am using an **arduino due** currently, but I do also have an **ardiuno zero** as a backup.

Acuadros95, 2023-03-08:
You need to follow the steps detailed on the [Patch SAM](https://github.com/micro-ROS/micro_ros_arduino#patch-sam) readme section.

You can also check with the arduino zero board, which should work without this patch.

cruzaviles, 2023-03-31:
Hi there!

Since we last spoke I completed each of the steps in the Patch SAM section (I think). I know as a matter of fact that I did replace the platform.txt file, but I am still having the same error for "Undefined Reference to" and cannot use precompiled library. 

Maybe I missed a step or need to do a few more things but I would love some more help.

Thank you again!

cruzaviles, 2023-04-14:
Hi there guys! 

Since last posting I have fixed a few things, and it is going better.

But, I am trying to get the publisher example working and I am having a problem. I get to the stage where I run the micro_ros_agent command and I get the output you would expect. It says that the session is established and the topic publisher and data writer are created. But, when I check the topic list I do not see the microROS publisher.

Thanks again!

cruzaviles, 2023-04-17:
Hi there!

Thank you for reaching out! I actually found a very helpful forum from earlier that helped me solve my issue. 

I didn't end up launching from the docker like the earlier comments suggested because I am running on a due board. 

But, near the end they bring up the Domain ID which was new info for me. After reading up some more I learned how to use the export command in the terminal window to change the value of the Domain ID variable. All that I did was change the value to the same number in the two terminal windows running the agent and checking the topic list. Once I did this I was able to see the topic in the list and watch it publish with the echo function. 

Here is the link to the other forum that helped. 
[https://github.com/micro-ROS/micro_ros_arduino/issues/7](url)

cruzaviles, 2023-04-17:
Now that I have the publisher working, I tried the subscriber example and I am pretty sure that I had it working. 

So, now I would like to try and customize this code, or use it to create a new sketch, so that I can read the data from microROS to the arduino and use those value to move some servo motors. 

I understand that it is probably a pretty big leap and might take some extra explaining, but I would love to learn more about how the arduino board gets data from microROS, where those values go, and then how I would be able to use those values after they are found. 

If there are any helpful tutorials or links that I should follow to make this happen I would be happy to! 

Thank you all again for your help!

Acuadros95, 2023-04-18:
Glad to hear you got it working! I am closing the issue, feel free to ask more questions here or on our slack channel.

Here are some relevant links:
- [micro-ROS documentation](https://docs.vulcanexus.org/en/latest/rst/microros_documentation/index.html)
- [micro-ROS tutorials](https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/micro_tutorials.html)
- [micro-ROS demos](https://github.com/micro-ROS/micro-ROS-demos/)

closed 2023-04-18 by Acuadros95