micro-ROS/micro_ros_arduino #1340: reconnection example could not run properly over wifi

marklee56826, 2023-03-10:
Hi, 
I really appreciate this amazing micro-ROS project and I am learning it. 

My problem: when I tested the reconnection example: micro-ros_reconnection_example, I found that it could not run properly over wifi (serial transport OK).  

Description: Compiling smoothly, but after starting the agent with the command: docker run -it --rm --net=host microros/micro-ros-agent:humble udp4 --port 8888  (already updated), the agent could not be steadily connected. 

Expected behavior:
Solid LED when successfully connected to agent, publishing topics.

Actual behavior:
Flashing LED with nothing published:
......
[1678409815.632855] info     | Root.cpp           | create_client            | create                 | client_key: 0x76432A75, session_id: 0x81
[1678409815.632968] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x76432A75, address: 192.168.0.10:47138
[1678409815.686132] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x76432A75, participant_id: 0x000(1)
[1678409815.694138] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x76432A75, topic_id: 0x000(2), participant_id: 0x000(1)
[1678409815.701274] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x76432A75, publisher_id: 0x000(3), participant_id: 0x000(1)
[1678409815.710856] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x76432A75, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1678409816.447346] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x76432A75
[1678409816.447402] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x76432A75, address: 192.168.0.10:47138
…… beginning to loop

My system: board ESP32-DevKit, agent machine: Intel NUC i5, Ubuntu 22.04, ROS humble, Client and agent have the same domain_id. 

What’s the problem? I did a lot of searches, without finding any solution.
Please help me. Thank you.

pablogs9, 2023-03-10:
Have you tried to use https://github.com/micro-ROS/micro_ros_espidf_component ?

marklee56826, 2023-03-10:
Thank you for your kind reply, I will try it.

marklee56826, 2023-03-16:
Hi, @pablogs9 
I tried and used micro-ROS component for ESP-IDF to run its own example code int32_publisher successfully, but still, I am unable to run micro-ros_reconnection_example on this platform.
1.I used int32_publisher to hold the source file micro-ros_reconnection_example.ino and changed the name to reconnection_example.c and modified the CMakeLists.txt.
2.Put micro_ros_arduino library into the include directory of the micro_ros_espidf_component
3.Build the project with docker container.

The output as follows:
……
[100/101] Generating binary image from built executable
esptool.py v4.4
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /micro_ros_espidf_component/test_reconnection/int32_publisher/build/bootloader/bootloader.bin
[101/101] cd /micro_ros_espidf_component/test_reconnection/int32_publisher/build/bootloader/esp-idf/esptool_py && /opt/esp/python_env/idf5.0_py3.8_env/bin/python /opt/esp/idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /micro_ros_espidf_component/test_reconnection/int32_publisher/build/bootloader/bootloader.bin
Bootloader binary size 0x6710 bytes. 0x8f0 bytes (8%) free.
[840/846] Building C object esp-idf/mai...df_main.dir/microros_reconnection.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/microros_reconnection.c.obj 
ccache /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -I/micro_ros_espidf_component/test_reconnection/int32_publisher/build/config -I/opt/esp/idf/components/newlib/platform_include -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/include -I/opt/esp/idf/components/freertos/esp_additions/include/freertos -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/opt/esp/idf/components/freertos/esp_additions/include -I/opt/esp/idf/components/esp_hw_support/include -I/opt/esp/idf/components/esp_hw_support/include/soc -I/opt/esp/idf/components/esp_hw_support/include/soc/esp32 -I/opt/esp/idf/components/esp_hw_support/port/esp32/. -I/opt/esp/idf/components/esp_hw_support/port/esp32/private_include -I/opt/esp/idf/components/heap/include -I/opt/esp/idf/components/log/include -I/opt/esp/idf/components/soc/include -I/opt/esp/idf/components/soc/esp32/. -I/opt/esp/idf/components/soc/esp32/include -I/opt/esp/idf/components/hal/esp32/include -I/opt/esp/idf/components/hal/include -I/opt/esp/idf/components/hal/platform_port/include -I/opt/esp/idf/components/esp_rom/include -I/opt/esp/idf/components/esp_rom/include/esp32 -I/opt/esp/idf/components/esp_rom/esp32 -I/opt/esp/idf/components/esp_common/include -I/opt/esp/idf/components/esp_system/include -I/opt/esp/idf/components/esp_system/port/soc -I/opt/esp/idf/components/esp_system/port/include/private -I/opt/esp/idf/components/xtensa/include -I/opt/esp/idf/components/xtensa/esp32/include -I/opt/esp/idf/components/lwip/include -I/opt/esp/idf/components/lwip/include/apps -I/opt/esp/idf/components/lwip/include/apps/sntp -I/opt/esp/idf/components/lwip/lwip/src/include -I/opt/esp/idf/components/lwip/port/esp32/include -I/opt/esp/idf/components/lwip/port/esp32/include/arch -I/opt/esp/idf/components/esp_ringbuf/include -I/opt/esp/idf/components/efuse/include -I/opt/esp/idf/components/efuse/esp32/include -I/opt/esp/idf/components/driver/include -I/opt/esp/idf/components/driver/deprecated -I/opt/esp/idf/components/driver/esp32/include -I/opt/esp/idf/components/esp_pm/include -I/opt/esp/idf/components/mbedtls/port/include -I/opt/esp/idf/components/mbedtls/mbedtls/include -I/opt/esp/idf/components/mbedtls/mbedtls/library -I/opt/esp/idf/components/mbedtls/esp_crt_bundle/include -I/opt/esp/idf/components/esp_app_format/include -I/opt/esp/idf/components/bootloader_support/include -I/opt/esp/idf/components/bootloader_support/bootloader_flash/include -I/opt/esp/idf/components/esp_partition/include -I/opt/esp/idf/components/app_update/include -I/opt/esp/idf/components/spi_flash/include -I/opt/esp/idf/components/pthread/include -I/opt/esp/idf/components/esp_timer/include -I/opt/esp/idf/components/app_trace/include -I/opt/esp/idf/components/esp_event/include -I/opt/esp/idf/components/nvs_flash/include -I/opt/esp/idf/components/esp_phy/include -I/opt/esp/idf/components/esp_phy/esp32/include -I/opt/esp/idf/components/vfs/include -I/opt/esp/idf/components/esp_netif/include -I/opt/esp/idf/components/wpa_supplicant/include -I/opt/esp/idf/components/wpa_supplicant/port/include -I/opt/esp/idf/components/wpa_supplicant/esp_supplicant/include -I/opt/esp/idf/components/esp_wifi/include -I/opt/esp/idf/components/unity/include -I/opt/esp/idf/components/unity/unity/src -I/opt/esp/idf/components/cmock/CMock/src -I/opt/esp/idf/components/console -I/opt/esp/idf/components/http_parser -I/opt/esp/idf/components/esp-tls -I/opt/esp/idf/components/esp-tls/esp-tls-crypto -I/opt/esp/idf/components/esp_adc/include -I/opt/esp/idf/components/esp_adc/interface -I/opt/esp/idf/components/esp_adc/esp32/include -I/opt/esp/idf/components/esp_adc/deprecated/include -I/opt/esp/idf/components/esp_eth/include -I/opt/esp/idf/components/esp_gdbstub/include -I/opt/esp/idf/components/esp_gdbstub/xtensa -I/opt/esp/idf/components/esp_gdbstub/esp32 -I/opt/esp/idf/components/esp_hid/include -I/opt/esp/idf/components/tcp_transport/include -I/opt/esp/idf/components/esp_http_client/include -I/opt/esp/idf/components/esp_http_server/include -I/opt/esp/idf/components/esp_https_o [... cut ...]
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

4. I also tried to modify the code to remove the errors, for example:
   if changing from .c to cpp, still has "error: 'arduino_transport_open' undeclared (first use in this function)..."
   if removing the fuction:set_microros_transports, causes "error: undefined reference to `app_main' "
   ......

Please help me to find the problem or tell me how to use arduino IDE to run micro-ros_reconnection_example with wifi transport. Thank you very much.

pablogs9, 2023-03-16:
> 2.Put micro_ros_arduino library into the include directory of the micro_ros_espidf_component

Why?

marklee56826, 2023-03-16:
Hi, thank you for kind reply. 
Because if not, I got the error: Not found micro_ros_arduino.h, not such file or directiory. Then the compiling stops. I am sorry I don't know how to remove the error.

marklee56826, 2023-03-16:
Hi, @pablogs9
 I tried to modify the file micro-ros_reconnection_example.ino and got the following result, which seems much better now:
……
/opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x8): undefined reference to `app_main'
/opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
/opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:100: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
……
What should I do to get rid of the error?

marklee56826, 2023-03-17:
The errors in the esp-idf code were finally eliminated but the result seems not as expected. I don't know where is the problem.

pablogs9, 2023-03-17:
So please, provide details (preferably in the correct repo issues) about which are the not expected results.

marklee56826, 2023-03-17:
Hi,  @pablogs9 
I am sorry I did not make myself understood. 
“Not as expected” means the result of my trying to run reconnection_example over WiFi network. I tried in different dev environment such as Arduino IDE, Plamformio, ESP-IDF. Unfortunately, the micro-ROS client cannot connect to the Agent properly. No issue with serial transport as the example in the repo.
My experiments showed that the reconnection_example with wifi transport could be compiled smoothly, but when running on ubuntu 22.04, esp32 board will always in a dead loop: connect and then disconnect.
 The connection to wifi network is good, and the only problem exists in the connection to Agent. 
I am sorry I am a beginner and my question may be too basic: would you like to tell me whether the reconnection_example can run on wifi network, not just on serial port?
Micro-ROS is really a great project and I like it very much and will continue to learn it.
Thank you for your patience and kindness.

closed 2023-03-19 by marklee56826