micro-ROS/micro_ros_arduino #1343: Error in compiling micro_ros_arduino examples in Arduino IDE

nilutpolkashyap, 2023-03-12:
## Issue template

- Hardware description:  ESP32 Dev Module
- Installation type: micro_ros_arduino ZIP ([v2.0.5-humble](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v2.0.5-humble)) install in Arduino IDE 
- Version or commit hash: humble
- Arduino IDE version: 1.8.19
- ESP32 board package version: 2.0.2 

#### Steps to reproduce the issue
Install micro_ros_arduino-2.0.3-foxy into Arduino 1.8.19 IDE's libraries directory.
Go to File > Examples > micro_ros_arduino > micro-ros_publisher
Try to compile the project.

#### Expected behavior
Successful compilation in Arduino IDE

#### Actual behavior
I am getting the following error messages in Arduino IDE
```
WARNING: Spurious .images folder in 'micro_ros_arduino' library
Archiving built core (caching) in: /tmp/arduino_cache_569881/core/core_esp32_esp32_esp32_PSRAM_disabled,PartitionScheme_default,CPUFreq_240,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,LoopCore_1,EventsCore_1,DebugLevel_none_962322ed5ffc32b196f8ea9d5cf34195.a
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z14timer_callbackP11rcl_timer_sx+0x8): undefined reference to `rcl_publish'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z5setupv+0x3c): undefined reference to `rmw_uros_set_custom_transport'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z5setupv+0x40): undefined reference to `rcutils_get_default_allocator'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z5setupv+0x44): undefined reference to `rclc_support_init'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z5setupv+0x48): undefined reference to `rclc_node_init_default'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z5setupv+0x4c): undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z5setupv+0x50): undefined reference to `rclc_publisher_init_default'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z5setupv+0x54): undefined reference to `rclc_timer_init_default'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z5setupv+0x58): undefined reference to `rclc_executor_init'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z5setupv+0x5c): undefined reference to `rclc_executor_add_timer'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o:(.literal._Z4loopv+0x4): undefined reference to `rclc_executor_spin_some'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o: in function `timer_callback(rcl_timer_s*, long long)':
/home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:36: undefined reference to `rcl_publish'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o: in function `setup()':
/home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/micro_ros_arduino.h:33: undefined reference to `rmw_uros_set_custom_transport'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o: in function `setup()':
/home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:47: undefined reference to `rcutils_get_default_allocator'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:49: undefined reference to `rclc_support_init'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:52: undefined reference to `rclc_node_init_default'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:55: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:55: undefined reference to `rclc_publisher_init_default'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rclc_timer_init_default'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:66: undefined reference to `rclc_executor_init'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:73: undefined reference to `rclc_executor_add_timer'
/home/nilutpolk/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/micro-ros_publisher.ino.cpp.o: in function `loop()':
/home/nilutpolk/Arduino/libraries/micro_ros_arduino-2.0.5-humble/examples/micro-ros_publisher/micro-ros_publisher.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.
```

#### Additional information
Please help me with solving the issue.

Acuadros95, 2023-03-13:
Just tested successfully with [v2.0.5-humble](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v2.0.5-humble).

How did you install the library? And which version did you use `micro_ros_arduino-2.0.3-foxy` or `micro_ros_arduino-2.0.5-humble`?

nilutpolkashyap, 2023-03-13:
> Install micro_ros_arduino-2.0.3-foxy into Arduino 1.8.19 IDE's libraries directory.

Sorry, that was a mistake. I used micro_ros_arduino-2.0.5-humble as I am on a Ubuntu 22.04 system with ROS 2 Humble

From within the Arduino IDE, I included the library in the application by navigating to Sketch -> Include Library -> Add .ZIP Library [v2.0.5-humble.zip](https://github.com/micro-ROS/micro_ros_arduino/archive/refs/tags/v2.0.5-humble.zip)

nilutpolkashyap, 2023-03-16:
Hi @Acuadros95 Can you please look into this and help me?

nilutpolkashyap, 2023-03-21:
Hi,
I figured out that I had some errors with my Arduino IDE installation. I reinstalled my Arduino IDE and the problem was solved. 
Thanks

closed 2023-03-21 by Acuadros95