micro-ROS/micro_ros_arduino #1352: Arduino for Ping on BlueROV

cernicarlo, 2023-03-23:
Hi everyone,

- Hardware Type: Arduino DUE
- Installation type: micro_ros_setup
- Version or commit hash: galactic
- Script on Arduino DUE: [example publisher](https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_publisher/micro-ros_publisher.ino)

Premise:  I asked the same question on the BlueRobotics forum ([here](https://discuss.bluerobotics.com/t/ping-with-arduino-using-micro-ros-possible/13911)), but unfortunately, I didn't get an answer -> I am doing something wrong on the micro ROS side.

I copy here the content:

Using ROS 2 on my laptop, I want to communicate with an Arduino DUE (with micro ROS running on it) connected to the companion computer (Raspberry Pi 3 with a Pixhawk running [ArduSub](https://www.ardusub.com/)).

After setting up micro ROS on my Arduino DUE and testing the serial communication on my laptop (following [this](https://roboticsknowledgebase.com/wiki/interfacing/microros-for-ros2-on-microcontrollers/) tutorial), I confirmed that everything is working ok:

```shell
$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 115200 -v6
[1678973917.856675] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1678973917.857069] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1678973935.542156] info     | Root.cpp           | create_client            | create                 | client_key: 0x0793D525, session_id: 0x81
[1678973935.542255] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x0793D525, address: 0
[1678973935.542397] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0793D525, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
```
For what it's worth, I also tested the UDP connection in the [first application linux tutorial](https://micro.ros.org/docs/tutorials/core/first_application_linux/)

The next steps were:

1. connect the Arduino to the companion computer
2. create the serial port (baud rate _115200_) and the UDP port (_192.168.2.2:8888_) using the [routing procedure](https://www.ardusub.com/reference/companion/routing.html) recommended by them
3. communicate with the Arduino through the UDP using micro ROS on my topside laptop

But something is wrong:

```shell
$ ros2 run micro_ros_agent micro_ros_agent udp4 --ip 192.168.2.1 --port 8888
[1678974517.990334] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1678974517.990704] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1678974531.335043] error    | InputMessage.cpp   | log_error                | deserialization error  | buffer: 
0000: 7E 00 00 18 00 80 00 00 00
[1678974532.326210] error    | InputMessage.cpp   | log_error                | deserialization error  | buffer: 
0000: 7E 00
```
What were my mistakes?

Acuadros95, 2023-03-23:
The communication protocol used on Serial and UDP transport are different and cannot be mixed:

- UDP uses packet-oriented mode: the communication mechanism implemented is able to send a whole packet that includes an XRCE message.
- Serial uses stream-oriented mode: the communication mechanism implemented does not have the concept of packet. [HDLC framing](https://micro-xrce-dds.docs.eprosima.com/en/latest/transport.html?highlight=hdlc#custom-serial-transport) will be used.

Why do you need to route the serial port to a UDP port?

Acuadros95, 2023-04-03:
> I need to route because my laptop is connected to the companion computer through an ethernet cable, and the communication between devices connected to the companion computer and my laptop is done through the port provided by the ethernet cable (as explained [here](https://www.ardusub.com/developers/software-components.html)).

You can run the Agent directly on the Raspberry Pi 3 side, and the DDS data (Publications, subscriptions, ...) will be shared to your laptop automatically through the Ethernet connection. Is this an option?

Other option is to use the reverse approach on your laptop, and bind the UDP port to a serial device.