micro-ROS/micro_ros_arduino #1357: ESP32 Arduino Framework ROS_DOMAIN_ID change

DuneV, 2023-03-26:
## Issue template

- Hardware description: ESP32 
- RTOS: NO
- Installation type: PlatformIO library
- Version or commit hash: Humble
- Framework: Arduino
#### Steps to reproduce the issue
I am trying to use the publisher example  "micro-ros_publisher.ino". I copied the exact example and add this command:
```
rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
RCCHECK(rcl_init_options_set_domain_id(&init_options, domain_id));
```
This apparently change the ros domain because I defined it "size_t domain_id  = 1", but it still without appear on my topics

#### Expected behavior
The connection between the micro ros agent and the library works in both cases, the base case (with the ros domain defautl) and changing this parameter:
[image]
#### Actual behavior
Not appear the topics 
#### Additional information
I view some issues but are related with older versions of ros2 and now the class node_ops does not have the atribute domain or maybe I just did not found well the commad :) thanks if you could help me

Acuadros95, 2023-03-27:
Are you using the options on the support initialization?
`rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator);`

> Not appear the topics

Could you detail how you checked this?

Acuadros95, 2023-03-28:
> The agent appears that is not reciving the data:

Then this problem is not related to the domain id. Check that the serial port selected on the Agent side is the correct one.

> Installation type: PlatformIO library

Also, what do you mean with this?

kalaajiahmad, 2023-04-14:
Hey @DuneV , your setup() is missing initializing  `init_options`
Here's how you can set the ROS DOMAIN ID properly:
`   allocator = rcl_get_default_allocator();

    // create init_options
    init_options = rcl_get_zero_initialized_init_options();
    RCCHECK(rcl_init_options_init(&init_options, allocator));

    // Set ROS domain id
    RCCHECK(rcl_init_options_set_domain_id(&init_options, domain_id));

    // Setup support structure.
    RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));

    // create node
    RCCHECK(rclc_node_init_default(&node, NODE, "", &support));
    debugln("Created Node!");

    // create fb_publisher......`

Let me know if that helps!

Acuadros95, 2023-07-25:
Closing due to inactivity, reopen if needed.

closed 2023-07-25 by Acuadros95