micro-ROS/micro_ros_arduino #1361: Examples give WiFi compilation errors for Arduino Nano RP2040 Connect

dhood, 2023-03-29:
- Hardware description: Arduino Nano RP2040 Connec
- RTOS: none
- Installation type: Binaries in Arduino IDE
- Version or commit hash: Humble from github 2.0.5

#### Steps to reproduce the issue
Hi! I'm trying to get started with a new RP2040, but getting compilation errors trying to run e.g. micro-ros_subscriber example that comes with the installation. 

I may have the environment configured incorrectly since the examples are nested under an "incompatible" subdirectory in the Arduino IDE, where I understand this board to be supported.

#### Expected behavior
Out-of-the-box demo uploads fine.

#### Actual behavior
```
WARNING: library micro_ros_arduino-2.0.5-humble claims to run on (stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta) architecture(s) and may be incompatible with your current board which runs on (mbed_nano) architecture(s).
In file included from /tmp/arduino_modified_sketch_223375/micro-ros_subscriber.ino:1:0:
/home/dhood/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/micro_ros_arduino.h: In function 'void set_microros_wifi_transports(char*, char*, char*, uint)':
/home/dhood/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/micro_ros_arduino.h:119:2: error: 'WiFi' was not declared in this scope
  WiFi.begin(ssid, pass);
  ^~~~
/home/dhood/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/micro_ros_arduino.h:121:29: error: 'WL_CONNECTED' was not declared in this scope
 	while (WiFi.status() != WL_CONNECTED) {
                         	^~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Nano RP2040 Connect.

```

#### Additional information
I tried manually installing WiFiNina but it didn't help.

pablogs9, 2023-03-30:
Does it build in a normal sketch without micro-ROS? Are you able to build a wifi example?

dhood, 2023-05-31:
oops no I can't use the basic wifi examples either, so I'll close this until I can narrow it down to a micro-ros issue. Thanks!!

closed 2023-05-31 by dhood

OmidRezayof, 2024-06-24:
Hi @dhood.
I'm having the exact same problem. Would you mind telling me how you solved it?