micro-ROS/micro_ros_arduino #1372: Compilation error for Teensy 4.1 on Arduino IDE 2.0.4

liamhan1, 2023-04-07:
## Issue template

- Hardware description: Teensy 4.1 + [Teensyduino](https://www.pjrc.com/arduino-ide-2-0-0-teensy-support/)

#### Steps to reproduce the issue
- I've installed arduino ide 2.0.4 and used the `micro-ros-publisher.ino`. But when compiling, I'm getting the error shown below

#### Expected behavior

#### Actual behavior

[image]

```
WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /Users/liam/Documents/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard
The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/arduino/sketches/90BFB27027F4E4700FC02F20C2CF58EA/sketch/micro-ros_publisher.ino.cpp.o: In function `timer_callback(rcl_timer_s*, long long)':
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:36: undefined reference to `rcl_publish'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/arduino/sketches/90BFB27027F4E4700FC02F20C2CF58EA/sketch/micro-ros_publisher.ino.cpp.o: In function `set_microros_transports':
/Users/liam/Documents/Arduino/libraries/micro_ros_arduino/src/micro_ros_arduino.h:40: undefined reference to `rmw_uros_set_custom_transport'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/arduino/sketches/90BFB27027F4E4700FC02F20C2CF58EA/sketch/micro-ros_publisher.ino.cpp.o: In function `setup':
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:49: undefined reference to `rcutils_get_default_allocator'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:52: undefined reference to `rclc_support_init'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:55: undefined reference to `rclc_node_init_default'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rclc_publisher_init_default'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:66: undefined reference to `rclc_timer_init_default'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:73: undefined reference to `rclc_executor_init'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:74: undefined reference to `rclc_executor_add_timer'
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/arduino/sketches/90BFB27027F4E4700FC02F20C2CF58EA/sketch/micro-ros_publisher.ino.cpp.o: In function `loop':
/private/var/folders/53/fl696_gx30s57vhwhsvb_3t40000gn/T/.arduinoIDE-unsaved202337-43691-1ozsss8.p7hh/micro-ros_publisher/micro-ros_publisher.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

#### Additional information

pablogs9, 2023-04-10:
Have you applied the required patches? https://github.com/micro-ROS/micro_ros_arduino#patch-teensyduino

baslati, 2023-05-20:
Hey, I have a pretty similar problem. 

I tried a lot of things but I can't get it to work:

System: x86 64 BIT; Ubuntu 22.04; Arduino IDE 2.1.0; and different Teensyduino versions; Teensy 4.1; ROS humble

I also tried to use the old IDE (1.8.19 and 1.8.15) but I had no luck with them either. 

In this Issue I read that someone is able to use 2.0.4.

> Have you applied the required patches? https://github.com/micro-ROS/micro_ros_arduino#patch-teensyduino

 @pablogs9 Yes, I applied the patches, but I just get different Error messages. 

Thanks in advance 

IDE 2.1.0 with Teensy 1.57.2 gepached:
`WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/sebastian/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard
The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.
/tmp/arduino/sketches/529F33B5E84BECC17E3367493800628A/sketch/micro-ros_publisher.ino.cpp.o: In function `timer_callback(rcl_timer_s*, long long)':
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:36: undefined reference to `rcl_publish'
/tmp/arduino/sketches/529F33B5E84BECC17E3367493800628A/sketch/micro-ros_publisher.ino.cpp.o: In function `set_microros_transports':
/home/sebastian/Arduino/libraries/micro_ros_arduino/src/micro_ros_arduino.h:40: undefined reference to `rmw_uros_set_custom_transport'
/tmp/arduino/sketches/529F33B5E84BECC17E3367493800628A/sketch/micro-ros_publisher.ino.cpp.o: In function `setup':
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:49: undefined reference to `rcutils_get_default_allocator'
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:52: undefined reference to `rclc_support_init'
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:55: undefined reference to `rclc_node_init_default'
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rclc_publisher_init_default'
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:66: undefined reference to `rclc_timer_init_default'
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:73: undefined reference to `rclc_executor_init'
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:74: undefined reference to `rclc_executor_add_timer'
/tmp/arduino/sketches/529F33B5E84BECC17E3367493800628A/sketch/micro-ros_publisher.ino.cpp.o: In function `loop':
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1`

IDE 2.1.0 with 1.58.0 gepached:
`WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/sebastian/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard
The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.
/home/sebastian/.arduino15/packages/teensy/hardware/avr/1.58.0/libraries/NativeEthernet/src/NativeEthernetServer.cpp: In member function 'EthernetClient EthernetServer::available()':
/home/sebastian/.arduino15/packages/teensy/hardware/avr/1.58.0/libraries/NativeEthernet/src/NativeEthernetServer.cpp:119:103: warning: enum constant in boolean context [-Wint-in-bool-context]
  119 |                                            if(socket_poll.events_occurred & (FNET_SOCKET_EVENT_OUT || FNET_SOCKET_EVENT_IN)) {
      |                                                                                                       ^~~~~~~~~~~~~~~~~~~~
/home/sebastian/.arduino15/packages/teensy/hardware/avr/1.58.0/libraries/FNET/src/service/llmnr/fnet_llmnr.c:60:30: warning: 'fnet_llmnr_ip6_multicast_addr' defined but not used [-Wunused-const-variable=]
   60 | static const fnet_ip6_addr_t fnet_llmnr_ip6_multicast_addr = FNET_IP6_ADDR_INIT(0xFF, 0x02, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0x01, 0x00, 0x03);
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/sebastian/.arduino15/packages/teensy/hardware/avr/1.58.0/libraries/FNET/src/third_party/mbedtls-2.12.0/src/ssl_tls.c:1062:83: warning: argument 2 of type 'unsigned char[36]' with mismatched bound [-Warray-parameter=]
 1062 | MBEDTLSFLASHMEM void ssl_calc_verify_tls( mbedtls_ssl_context *ssl, unsigned char hash[36] )
      |                                                                     ~~~~~~~~~~~~~~^~~~~~~~
/home/sebastian/.arduino15/packages/teensy/hardware/avr/1.58.0/libraries/FNET/src/third_party/mbedtls-2.12.0/src/ssl_tls.c:483:73: note: previously declared as 'unsigned char *'
[... cut ...] r'
/home/sebastian/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/529F33B5E84BECC17E3367493800628A/sketch/micro-ros_publisher.ino.cpp.o: in function `loop':
/tmp/.arduinoIDE-unsaved2023420-4924-1r5wbm7.z4tj/micro-ros_publisher/micro-ros_publisher.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1`

baslati, 2023-05-20:
Well, 2 days of trial and error and after I finally ask someone I find a way 5min later. 

For me the problem was IDE 2.1.0, currently I am running IDE 2.0.4 patched with Teensy 1.57.2. 

In this combination I can compile the microros library. 

I still can't see the publisher-topic but I will follow that up in Issue 1392.
https://github.com/micro-ROS/micro_ros_arduino/issues/1392

Maybe my solution still helps someone.

Acuadros95, 2023-07-27:
Hi guys,

We have just updated the repository to support Teensyduino 1.58.x, deprecating the 1.57.2 support.
Please take a look to the latest commits and try them out, will add a release with the update once the feedback is positive.

closed 2023-07-27 by Acuadros95