micro-ROS/micro_ros_arduino #1377: General RP2040 support?

Chick92, 2023-04-17:
Are there any plans to support a standard RP2040? I.e Raspberry Pi Pico? I know there’s a C/C++ implementation for the RP2040 / Pico, but I would love to be able to leverage the existing library support. 

Thanks!

pablogs9, 2023-04-18:
This is the support for RPi Pico SDK https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk

closed 2023-04-18 by pablogs9

Chick92, 2023-04-18:
Hi,

You didn't read the ticket. As I said, I'm aware of the pico sdk. I asked about support for the pico in the arduino ecosystem.

pablogs9, 2023-04-18:
Hi @Chick92, yes I did. I was saying that the support that we provide for Pico SDK is the one I mentioned above.

There are no plans for integrating it in micro-ROS for Arduino, although contributions are welcomed.

In any case, usually, all Arduino environments can be substituted with Platform.IO, so maybe you can configure micro-ROS for the [Platform.IO module](https://github.com/micro-ROS/micro_ros_platformio) to have an Arduino API.

Chick92, 2023-04-18:
Right, so say that then instead of palming me off with a none answer.