micro-ROS/micro_ros_arduino #1378: Error When Compiling Teensyduino 1.58+

jtcass01, 2023-04-20:
- Hardware description: Teensy 4.1 and Teensyduino 1.58.0
- Installation type: micro_ros_setup
- Version or commit hash: Galactic, https://github.com/micro-ROS/micro_ros_arduino/commit/e7bd525cad256cd5fb36f1cae98b39cf1a6dba69

#### Steps to reproduce the issue
- Install Arduino 2.0.4.
- Add https://www.pjrc.com/teensy/package_teensy_index.json to Additional Boards Manager
- Add teensy rules to /etc/udev/rules.d/00-teensy.rules
- Replace platform.txt with https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/galactic/extras/patching_boards/platform_teensy.txt
- Attempt to compile code for Teensy 4.1

#### Expected behavior
- successful compilation

#### Actual behavior
[image]

#### Additional information
The above image is from Arduino 1.8.19. I had the same error for 2.0.4.  I was able to successfully compile when using Teensyduino 1.57.2. This seems like a problem specific to Teensyduino 1.58+.

CaseyGymnast, 2023-07-26:
Subscribing for future updates: I'm dealing with the same issue on a Teensy 3.2. I'm linking another [issue thread](https://forum.pjrc.com/threads/73230-Issues-using-Micro-ROS-s-precompiled-libraries-with-Arduino-and-Teensy?p=329134#post329134) on PJRC's website that I've been updating as I try to figure it out. Best of luck.

jtcass01, 2023-07-27:
@CaseyGymnast I solved this by using Arduino 2.0.4 and Teensyduino 1.57.2.  In arduino 2, you can choose older teensyduino versions.  It'll take more time for me to find out the micro ROS distribution that matched but I believe I used v2.0.4-galactic.

Acuadros95, 2023-07-27:
Hi guys,

We have just updated the repository to support Teensyduino 1.58.x, deprecating the 1.57.2 support.
Please take a look to the latest commits and try them out, will add a release with the update once the feedback is positive.

closed 2023-07-27 by Acuadros95

jtcass01, 2023-07-27:
@Acuadros95 that is great to hear! I will verify next week and let you know.  Thank you!

Russ76, 2023-08-02:
Yes, I'm having issues with Teensy 4.0 and Arduino IDE 2.1.1
It won't upload the precompiled libraries.

Russ76, 2023-08-02:
WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/russ/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/russ/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xbe): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/russ/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x5e): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/russ/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x46): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0xb6): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0x164): undefined reference to `__locale_ctype_ptr'
collect2: error: ld returned 1 exit status