micro-ROS/micro_ros_arduino #1379: 6th subscriber fails

yoneyoneclub, 2023-04-22:
## 6th subscriber fails
- Hardware description: Arduino Portenta H7
- RTOS: 
- Installation type: micro_ros_setup,Arduino IDE
- Version or commit hash: humble 2.0.5

#### Steps to reproduce the issue
> Include it in your project using Sketch -> Include library -> Add .ZIP Library...
> You can test micro-ROS examples located in this repo examples folder.

Follow the installation instructions to build the Adruino IDE environment
and run below command 

Write the following program 6 subscriber

```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rcl_logging_interface/rcl_logging_interface.h>

//type for sub/pub
#include <std_msgs/msg/empty.h>
#include <std_msgs/msg/string.h>
#include <std_msgs/msg/int8.h>
#include <std_msgs/msg/u_int8.h>
#include <std_msgs/msg/int16.h>
#include <std_msgs/msg/u_int16.h>
#include <std_msgs/msg/u_int32.h>
#include <std_msgs/msg/float32.h>
#include <std_msgs/msg/float64.h>
#include <std_msgs/msg/float64_multi_array.h>

#define DEBUG Serial1

#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

//sub
rcl_subscription_t sub1;//empty

rcl_subscription_t sub2;//float32
rcl_subscription_t sub3;//float32

rcl_subscription_t sub4;//Int16
rcl_subscription_t sub5;//Int16
rcl_subscription_t sub6;//Int16

//topic msg
std_msgs__msg__Empty msg1;

std_msgs__msg__Float32 msg2;
std_msgs__msg__Float32 msg3;

std_msgs__msg__Int16 msg4;
std_msgs__msg__Int16 msg5;
std_msgs__msg__Int16 msg6;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;

void Cb1(const void * msgin)
{
  DEBUG.println("1.get topic1");
}

void Cb2(const void * msgin)
{
  DEBUG.println("2.get topic2");
}

void Cb3(const void * msgin)
{
  DEBUG.println("3.get topic3");
}

void Cb4(const void * msgin)
{
  DEBUG.println("4.get topic4");
}

void Cb5(const void * msgin)
{
  DEBUG.println("5.get topic5");
}

void Cb6(const void * msgin)
{
  DEBUG.print("6.get ctrl_gear_mode");
}

void setup() {
  DEBUG.begin(115200);
  set_microros_transports();//USB経由
  
  delay(2000);

  allocator = rcl_get_default_allocator();//micro-ROSのためのメモリ管理

  //create init_options
  RCSOFTCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCSOFTCHECK(rclc_node_init_default(&node, "microros_arduino_host_node", "", &support));

  /* Subscriber */
  RCSOFTCHECK(rclc_subscription_init_best_effort(
    &sub1,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Empty),
    "topic1"));

  RCSOFTCHECK(rclc_subscription_init_best_effort(
    &sub2,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32),
    "topic2"));

  RCSOFTCHECK(rclc_subscription_init_best_effort(
    &sub3,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32),
    "topic3"));

  RCSOFTCHECK(rclc_subscription_init_best_effort(
    &sub4,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int16),
    "topic4"));

  RCSOFTCHECK(rclc_subscription_init_best_effort(
    &sub5,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int16),
    "topic5"));

  RCSOFTCHECK(rclc_subscription_init_best_effort(
    &sub6,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int16),
    "topic6"));

  // create executor
  executor = rclc_executor_get_zero_initialized_executor();
  RCSOFTCHECK(rclc_executor_init(&executor, &support.context, 6, &allocator));
  RCSOFTCHECK(rclc_executor_add_subscription(&executor, &sub1, &msg1, &Cb1, ON_NEW_DATA));
  
  RCSOFTCHECK(rclc_executor_add_subscription(&executor, &sub2, &msg2, &Cb2, ON_NEW_DATA));
  RCSOFTCHECK(rclc_executor_add_subscription(&executor, &sub3, &msg3, &Cb3, ON_NEW_DATA));
  RCSOFTCHECK(rclc_executor_add_subscription(&executor, &sub4, &msg4, &Cb4, ON_NEW_DATA));
  
  RCSOFTCHECK(rclc_executor_add_subscription(&executor, &sub5, &msg5, &Cb5, ON_NEW_DATA));
  
  RCSOFTCHECK(rclc_executor_add_subscription(&executor, &sub6, &msg6, &Cb6, ON_NEW_DATA));
}

void loop() {
  delay(10);
  rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
}
```

>$ docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev [YOUR BOARD PORT] -v6

Run the following in another command

> $ros2 topic list
```
/parameter_events
/rosout
/topic1
/topic2
/topic3
/topic4
/topic5
```
I debugged it in serail1 and it seemed to be the 6th subscriber that was causing the error. Also, when I switched the order of subscriber registrations, the error occurred in the sixth subscriber as well. I also changed the topic type, but the situation did not change.

#### Expected behavior
I have set the argument to 6 in `rclc_executor_init`, but I don't know what else could be causing this. Could you please tell me how to improve this situation?

pablogs9, 2023-04-22:
You need to [rebuild your library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library), configuring the colcon.meta with the [required memory options](https://github.com/micro-ROS/micro_ros_arduino/blob/040e91ba9da07533c0feefa2b13b3589bb787f70/extras/library_generation/colcon.meta#L45)

yoneyoneclub, 2023-04-24:
Thanks for the reply.
I have tried to build following the steps below.

```
git clone -b humble https://github.com/micro-ROS/micro_ros_arduino
cd micro_ros_arduino
docker pull microros/micro_ros_static_library_builder:humble
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:humble -p portenta-m7
```

However, I am getting the following error and cannot change the options to build.

> === ./control_msgs (git) ===
> Cloning into '.'...
> find: cannot delete ‘/project/src/’: Directory not empty
> Crosscompiled environment: cleaning path
> Building firmware for generate_lib platform generic
> .......

> Summary: 69 packages finished [35.0s]
>   50 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces control_msgs diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs micro_ros_utilities microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_interface rcl_logging_noop rclc rclc_lifecycle rclc_parameter rcutils rmw rmw_implementation rmw_microxrcedds ros2trace rosgraph_msgs rosidl_cli rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs sensor_msgs_py shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs test_tracetools_launch tf2_msgs tracetools_launch tracetools_read tracetools_test tracetools_trace trajectory_msgs unique_identifier_msgs visualization_msgs
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> rsync is already the newest version (3.2.7-0ubuntu0.22.04.2).
> 0 upgraded, 0 newly installed, 0 to remove and 21 not upgraded.
> rsync: [sender] change_dir "/project/src/common_interfaces/common_interfaces" failed: No such file or directory (2)
> rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]

I would appreciate it if you could tell me where the problem

pablogs9, 2023-04-24:
There is no problems in this log. Your library has been successfully built.

yoneyoneclub, 2023-04-24:
Thanks for the reply!
You are right, there was no problem with the build!
I was able to register my 6th subscriber!

`$ros2 topic list`
> /parameter_events
> /rosout
> /topic1
> /topic2
> /topic3
> /topic4
> /topic5
> /topic6

Thank you for your quick and thorough answer!!!!

closed 2023-04-24 by pablogs9