micro-ROS/micro_ros_arduino #1389: number of timers available

patrickwasp, 2023-05-16:
## Issue template

- Hardware description: ESP32
- RTOS: FreeRTOS?
- Installation type: platformio
- Version or commit hash: https://github.com/micro-ROS/micro_ros_platformio#1.0.0

#### Steps to reproduce the issue
```c++
#include <Arduino.h>
#include <micro_ros_platformio.h>

#define RC_CHECK(fn)             \
  {                              \
    rcl_ret_t temp_rc = fn;      \
    if ((temp_rc != RCL_RET_OK)) \
    {                            \
      rcError();                 \
    }                            \
  }

#define EXECUTE_EVERY_N_MS(MS, X)      \
  do                                   \
  {                                    \
    static volatile int64_t init = -1; \
    if (init == -1)                    \
    {                                  \
      init = uxr_millis();             \
    }                                  \
    if (uxr_millis() - init > MS)      \
    {                                  \
      X;                               \
      init = uxr_millis();             \
    }                                  \
  } while (0)

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>

#include <std_msgs/msg/int32.h>

rclc_support_t g_support;
rcl_node_t g_node;
rclc_executor_t g_executor;
static rcl_allocator_t g_allocator;
rcl_init_options_t g_init_options;

rcl_timer_t g_update_first_timer;
rcl_timer_t g_update_second_timer;
rcl_timer_t g_update_third_timer;
rcl_timer_t g_first_publisher_timer;
rcl_timer_t g_second_publisher_timer;
rcl_timer_t g_third_publisher_timer;

rcl_publisher_t g_first_publisher;
rcl_publisher_t g_second_publisher;
rcl_publisher_t g_third_publisher;

std_msgs__msg__Int32 g_first_msg;
std_msgs__msg__Int32 g_second_msg;
std_msgs__msg__Int32 g_third_msg;

float g_first_value;
float g_second_value;
float g_third_value;

[... 241 lines cut ...]
#### Additional information

If I change the order of the publishers and timers in code then different publishers start working. Is there a built-in limit in `rclc_executor_get_zero_initialized_executor()`, `rcl_get_default_allocator()`, or `rcl_get_zero_initialized_init_options()` that set the number of timers that can be run?

When it comes to timers, does the timeout_ns indicate how often it will run or how long it will wait for the callback to return? If it is just a timeout, how do you set the callback frequency? Are timer callbacks run asynchronously or are they blocking for the rest of the code?

the meta I'm using is

```
{
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}
```

Acuadros95, 2023-05-16:
The maximum number of timers is configured with the colcon parameter `RMW_UXRCE_MAX_GUARD_CONDITION` which defaults to 4 ([detail](https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/memory_management/memory_management.html#entity-creation))

> When it comes to timers, does the timeout_ns indicate how often it will run or how long it will wait for the callback to return? If it is just a timeout, how do you set the callback frequency? 

The timeout value defines the callback frequency, its not related to the callback execution time.

> Are timer callbacks run asynchronously or are they blocking for the rest of the code?

The callbacks run on the same thread that the executor spin method is called, so they the callbacks will block until they return.

You can find more info about timers and the executor here: [Executor and timers](https://docs.vulcanexus.org/en/latest/rst/microros_documentation/user_api/user_api_executor_timers.html).

patrickwasp, 2023-05-16:
Thanks for the link to vulcanexus, looks like a great resource. 

After adding `"-DRMW_UXRCE_MAX_GUARD_CONDITION=20"` to the meta file I still don't see the values of the second or third publisher changing. Is there a practical limit to the guard condition value? 

Why does `rqt` show ~48Hz for every publisher, when I have the timeouts set to different values? 

this is my platformio.ini 
``` 
[env:esp32-poe-iso]
platform = espressif32
board = esp32-poe-iso
framework = arduino
monitor_speed = 115200
board_microros_transport = serial
board_microros_distro = humble
board_microros_user_meta = config/esp32.meta
lib_deps = 
	https://github.com/micro-ROS/micro_ros_platformio#1.0.0
```

Acuadros95, 2023-05-17:
> After adding "-DRMW_UXRCE_MAX_GUARD_CONDITION=20" to the meta file I still don't see the values of the second or third publisher changing. 

Did your rebuild the library?

Also, some thoughts:
- Use the latest `micro_ros_platformio` version, not the `1.0.0`.
- Dont sleep inside the timers callback