micro-ROS/micro_ros_arduino #1392: ROS2 not detecting micro-ROS service

alvarit1, 2023-05-18:
## Issue template

- Hardware description: Arduino Nano RP2040 Connect
- RTOS: Arduino IDE 2.0.4
- Installation type: 
- Version or commit hash: Humble

I'm new to ROS2 and micro-ROS and currently going through micro-ROS tutorials trying to learn the basics of how to implement ROS2 publishers, subscribers and service servers/clients on microcontrollers.

I've managed to create publishers and subscribers that communicate with other ROS2 objects via micro-ROS agent. I used the micro-ros_publisher and micro-ros_subscriber examples available in the examples folder of this github repository.

However, when trying out the micro-ros_addtwoints_service example code, ROS2 running on my main computer does not detect any services. The example code compiles and uploads to the microcontroller, connection to the micro-ROS agent is made, but command `ros2 service list` gives empty output, even though service "/addtwoints" should be created in the example code. However, the node "add_twoints_client_rclc" created in the example code is created and command `ros2 node list` gives output `/add_twoints_client_rclc`. Any ideas what am I missing, why there is no service created?

#### Steps to reproduce the issue
Open Arduino IDE, go to File > Examples > micro_ros_arduino > micro-ros_addtwoints_service or copy paste the example from this link https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino.

Upload the code to the microcontroller and connect the microcontroller to the computer running ROS2 and micro-ROS agent with USB cable. 

In ROS2 type command `ros2 service list`

#### Expected behavior

Command `ros2 service list` should give output "/addtwoints"

#### Actual behavior

Command `ros2 service list` gives empty output. No services are detected.

#### Additional information

I installed ROS2 following this installation guide on ROS2 website: https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html

I installed micro-ROS agent following this guide on micro-ROS website: https://micro.ros.org/docs/tutorials/core/first_application_linux/

I'm happy to provide any extra information if I'm missing something in the description.

pablogs9, 2023-05-18:
You need to modify the colcon.meta and rebuild the library:

https://github.com/micro-ROS/micro_ros_arduino/blob/b7d5b962ed7c3bf1c9ef8c633c082f1ba0e990e2/extras/library_generation/colcon_verylowmem.meta#L46

https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library

baslati, 2023-05-21:
I also can't detect a micro-ROS service. 
I am using a Teensy 4.1, I am not able to use micro-ros_publisher or subcriber 

> I've managed to create publishers and subscribers that communicate with other ROS2 objects via micro-ROS agent. I used the micro-ros_publisher and micro-ros_subscriber examples available in the examples folder of this github repository.

I wanted to try your tip with the colcon meta:
But the colcon_verylowmem was already changed to  "-DRMW_UXRCE_MAX_SERVICES=0",  
do I still need rebuild the library?

Acuadros95, 2023-05-22:
> But the colcon_verylowmem was already changed to "-DRMW_UXRCE_MAX_SERVICES=0",
do I still need rebuild the library?

You need to modify the value to the number of services that your program needs, and yes, you will need to rebuild the library to apply this change.

alvarit1, 2023-05-24:
> You need to modify the colcon.meta and rebuild the library:

This indeed seemed to be the problem. Services are working now after modifying the colcon.meta and rebuilding the library.

Thank you very much for help and a fast reply!