micro-ROS/micro_ros_arduino #1393: Compile error Basic publisher with teensy 3.6 ros2 humble

FrGe2016, 2023-05-22:
## Issue template

- Hardware description:  Teensy 3.6
- RTOS: 
- Installation type:  Installation following this link
- Version or commit hash:  Humble, Teensyduino 1.58, IDE 1.8.19

#### Steps to reproduce the issue

               Fresh install of Arduino IDE
               Fresh install of Teensyduino 1.5.8
               Patch installed at 2 locations ( the location required in the tutorial does not exist ) : 
                    $ARDUINO_PATH/hardware/avr/$TEENSYDUINO_VERSION/

                 I have either :
                    ~/.arduino15/packages/teensy/hardware/avr/1.58.1
                    of
                    ~/arduino-1.8.19/hardware/teensy/avr

                  I would have liked to try with Arduino 2.10 but i did not find any location that would look slightlly like the requirement to place     the patch

                  

#### Expected behavior Compile without errors

#### Actual behavior This error message
                                   Arduino: 1.8.19 (Linux), TD: 1.58, Board: "Teensy 3.6, Serial, 180 MHz, Faster, US English"

WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/francois/Arduino/libraries/micro_ros_arduino/src/mk66fx1m0/fpv4-sp-d16-hard
/home/francois/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/francois/Arduino/libraries/micro_ros_arduino/src/mk66fx1m0/fpv4-sp-d16-hard/libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xc0): undefined reference to `__locale_ctype_ptr'
/home/francois/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/francois/Arduino/libraries/micro_ros_arduino/src/mk66fx1m0/fpv4-sp-d16-hard/libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x5e): undefined reference to `__locale_ctype_ptr'
/home/francois/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/francois/Arduino/libraries/micro_ros_arduino/src/mk66fx1m0/fpv4-sp-d16-hard/libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x36): undefined reference to `__locale_ctype_ptr'
/home/francois/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0xa6): undefined reference to `__locale_ctype_ptr'
/home/francois/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0x162): undefined reference to `__locale_ctype_ptr'
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 3.6.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

#### Additional information

Tyce11, 2023-06-29:
Was a solution ever found for this?

closed 2023-07-27 by Acuadros95

Russ76, 2023-08-15:
I am getting these same errors with updated libraries, and Arduino IDE 2.1.1 and Teensy 4.0