micro-ROS/micro_ros_arduino #1404: compile error when tried example publisher into Teensy4.0, ROS2 Humble

aky-u, 2023-06-05:
## Issue template

- Hardware description:  Teensy4.0
- RTOS: 
- Installation type:  micro_ros_setup
- Version or commit hash:  humble [b7d5b96](https://github.com/micro-ROS/micro_ros_arduino/commit/b7d5b962ed7c3bf1c9ef8c633c082f1ba0e990e2)

#### Steps to reproduce the issue

[here](https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/) is what I followed. 
* Installed ArdunoIDE 2.1.0 into ubuntu 22.04, then installed teensyduno ver 1.58
* download zip file from branch humble, then read it from ArduinoIDE 
* I opened basic example publisher, and tried to write it onto Teensy4.0 

#### Expected behavior
Be compiled successfully 

#### Actual behavior
WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Precompiled library in "/home/akiyoshi/Arduino/libraries/micro_ros_arduino/src/mkl26z64" not found
The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.
/home/akiyoshi/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/287110A1220B94566F1BA6C42A587786/sketch/micro-ros_subscriber.ino.cpp.o: in function `setup':
/home/akiyoshi/Arduino/libraries/micro_ros_arduino/src/micro_ros_arduino.h:33: undefined reference to `rmw_uros_set_custom_transport'
/home/akiyoshi/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/287110A1220B94566F1BA6C42A587786/sketch/micro-ros_subscriber.ino.cpp.o: in function `setup':
/tmp/.arduinoIDE-unsaved202356-20412-oqehn1.trba/micro-ros_subscriber/micro-ros_subscriber.ino:46: undefined reference to `rcutils_get_default_allocator'
/home/akiyoshi/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/.arduinoIDE-unsaved202356-20412-oqehn1.trba/micro-ros_subscriber/micro-ros_subscriber.ino:49: undefined reference to `rclc_support_init'
/home/akiyoshi/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/.arduinoIDE-unsaved202356-20412-oqehn1.trba/micro-ros_subscriber/micro-ros_subscriber.ino:52: undefined reference to `rclc_node_init_default'
/home/akiyoshi/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/.arduinoIDE-unsaved202356-20412-oqehn1.trba/micro-ros_subscriber/micro-ros_subscriber.ino:55: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/akiyoshi/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/.arduinoIDE-unsaved202356-20412-oqehn1.trba/micro-ros_subscriber/micro-ros_subscriber.ino:55: undefined reference to `rclc_subscription_init_default'
/home/akiyoshi/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/.arduinoIDE-unsaved202356-20412-oqehn1.trba/micro-ros_subscriber/micro-ros_subscriber.ino:62: undefined reference to `rclc_executor_init'
/home/akiyoshi/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/.arduinoIDE-unsaved202356-20412-oqehn1.trba/micro-ros_subscriber/micro-ros_subscriber.ino:63: undefined reference to `rclc_executor_add_subscription'
/home/akiyoshi/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/287110A1220B94566F1BA6C42A587786/sketch/micro-ros_subscriber.ino.cpp.o: in function `loop':
/tmp/.arduinoIDE-unsaved202356-20412-oqehn1.trba/micro-ros_subscriber/micro-ros_subscriber.ino:68: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

#### Additional information
I searched missing file "micro_ros_arduino/src/mkl26z64" in this repository, but I couldn't.

closed 2023-06-05 by aky-u

Deepspacee12, 2024-10-09:
how was this issue resolved? can you please tell

hippo5329, 2024-10-09:
Please check the wiki, https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki.

Deepspacee12, 2024-10-09:
hey thank you for responding but i found out the problem and solved it