micro-ROS/micro_ros_arduino #1423: How can I confirm that my library was updated after rebuilding?

Tyce11, 2023-06-16:
Hello,

I recently rebuilt my library using, 
`docker pull microros/micro_ros_static_library_builder:humble`
and
`docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:humble`

How can I check that my library was properly rebuilt and updated?

Tyce11, 2023-06-16:
For more context I am wanting to check if it installed correctly after getting this output at the end:   

Summary: 69 packages finished [2min 6s]
  49 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces control_msgs diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs micro_ros_utilities nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_interface rcl_logging_noop rclc rclc_lifecycle rclc_parameter rcutils rmw rmw_implementation rmw_microxrcedds ros2trace rosgraph_msgs rosidl_cli rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs sensor_msgs_py shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs test_tracetools_launch tf2_msgs tracetools_launch tracetools_read tracetools_test tracetools_trace trajectory_msgs unique_identifier_msgs visualization_msgs
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
rsync is already the newest version (3.2.7-0ubuntu0.22.04.2).
0 upgraded, 0 newly installed, 0 to remove and 20 not upgraded.
rsync: [sender] change_dir "/project/src/common_interfaces/common_interfaces" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/libyaml_vendor/libyaml_vendor" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/micro_ros_utilities/micro_ros_utilities" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/microcdr/microcdr" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/microxrcedds_client/microxrcedds_client" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rcl_logging_noop/rcl_logging_noop" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rclc/rclc" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rclc_lifecycle/rclc_lifecycle" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rclc_parameter/rclc_parameter" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rmw_implementation/rmw_implementation" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rmw_implementation_cmake/rmw_implementation_cmake" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rmw_microxrcedds/rmw_microxrcedds" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/ros2trace/ros2trace" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rosidl_adapter/rosidl_adapter" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rosidl_cli/rosidl_cli" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rosidl_cmake/rosidl_cmake" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rosidl_default_generators/rosidl_default_generators" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rosidl_default_runtime/rosidl_default_runtime" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/rosidl_generator_c/rosidl_generator_c" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
[... cut ...]
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/tracetools_test/tracetools_test" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/tracetools_trace/tracetools_trace" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]

pablogs9, 2023-06-19:
It seems that your library is not being copied correctly to your local folder. Where are you running the docker command?

Tyce11, 2023-06-19:
Arduino/libraries/Micro-ros-arduino2.06

Tyce11, 2023-06-30:
Closing this issue. According to @Acuadros95 's reply in #1447 , the rsync errors have no effect on the build. If I'm not wrong you can check to see if a library rebuilt by seeing if the values in `micro_ros_arduino/src/rmw_microxrcedds_c/config.h` were changed to the values you changed in the respective `.meta` file

closed 2023-06-30 by Tyce11

Acuadros95, 2023-07-03:
>  If I'm not wrong you can check to see if a library rebuilt by seeing if the values in micro_ros_arduino/src/rmw_microxrcedds_c/config.h were changed to the values you changed in the respective .meta file

That's correct, and yes, the rsync errors can just be ignored.