micro-ROS/micro_ros_arduino #1447: Error build the precompiled library

LuisYordano, 2023-06-28:
#### Steps to reproduce the issue
git clone -b humble https://github.com/micro-ROS/micro_ros_arduino
cd micro_ros_arduino
docker pull microros/micro_ros_static_library_builder:humble
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:humble

-------------------------

[image]
--------------------------

Finished <<< stereo_msgs [2.99s]
--- stderr: visualization_msgs                                      
CMake Warning at /uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_runtime_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_runtime_c/cmake/rosidl_runtime_cConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/builtin_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/builtin_interfaces/cmake/builtin_interfacesConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

---
Finished <<< visualization_msgs [7.79s]
Starting >>> common_interfaces
Finished <<< common_interfaces [0.69s]                          

Summary: 69 packages finished [1min 35s]
  50 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces control_msgs diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs micro_ros_utilities microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_interface rcl_logging_noop rclc rclc_lifecycle rclc_parameter rcutils rmw rmw_implementation rmw_microxrcedds ros2trace rosgraph_msgs rosidl_cli rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs sensor_msgs_py shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs test_tracetools_launch tf2_msgs tracetools_launch tracetools_read tracetools_test tracetools_trace trajectory_msgs unique_identifier_msgs visualization_msgs
Crosscompiled environment: cleaning path

-----------

I thank you in advance for your answer.

Acuadros95, 2023-06-29:
There is no error on your log, just a warning that is not relevant for this build system.

closed 2023-06-29 by Acuadros95

LuisYordano, 2023-06-29:
Additional info:
The build process starts a loop and the library is never built.
During the build process I get the following error:

-----------------
[image]

-----------------

Acuadros95, 2023-06-29:
Its not a loop, the docker builds the library multiple times to support the different boards.

You can set the specific target you are building the library for like this: 

```bash
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:iron -p <LIBRARY_TARGET>

# Example build for ESP32
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:iron -p esp32
```

Please give it a try with your specific target, so we can focus on just one single board.

Tyce11, 2023-06-29:
@LuisYordano, I am also having the rsync error problem. Please keep the thread updated with any solutions you may find.

Acuadros95, 2023-06-30:
The rsync errors are just verbose and it actually does not affect the build process.

You can just ignore those errors.