micro-ROS/micro_ros_arduino #1449: Teensy 3.6 with ros2 Humble - No topics with publisher example

Matt-Collins-52, 2023-07-03:
I am using a teensy 3.6 with ros2 humble on ubuntu 22. I have installed the teensyduino patches and the code 'micro-ros_publisher.ino' compiles fine and is running on the teensy.
When I source my terminal window and run `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0` I get this output:
`[1688391842.255238] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1688391842.255336] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1688391848.636169] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1688391848.646262] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1688391849.650129] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1688391850.656217] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1688391851.418302] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1688391853.435834] info     | Root.cpp           | create_client            | create                 | client_key: 0x604882F6, session_id: 0x81
[1688391853.435892] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x604882F6, address: 0
[1688391853.446454] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x604882F6, participant_id: 0x000(1)
[1688391853.451422] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x604882F6, topic_id: 0x000(2), participant_id: 0x000(1)
[1688391853.456573] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x604882F6, publisher_id: 0x000(3), participant_id: 0x000(1)
[1688391853.461512] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x604882F6, datawriter_id: 0x000(5), publisher_id: 0x000(3)
`

And then running `ros2 topic list` gives me this:
`/parameter_events
/rosout
`

I have also specified different baud rates and have gotten the same result.

I do not see any topics being published to from the teensy, and the onboard led is on all the time. I am expecting to see a topic that the teensy is publishing to.

I have also used this tutorial to set everything up. I am not using docker, and I have followed it for humble and teensy 3.6 instead of their exact setup.
https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/

Acuadros95, 2023-07-03:
The publisher is being created, as this line details:

`[1688391853.461512] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x604882F6, datawriter_id: 0x000(5), publisher_id: 0x000(3)`

Could you the agent on debug mode? You can do this adding the `-v6` flag:

`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6`

Did you modify any part of the micro-ROS example code? 
Any modifications on your ROS 2 environment? (For example a custom `ROS_DOMAIN_ID` configuration)

Matt-Collins-52, 2023-07-03:
I removed `export ROS_DOMAIN_ID=1` from ~/.bashrc and I was able to see the topic. Is there any way to use a domain id with micro ros?

Acuadros95, 2023-07-03:
Yes, you can configure it on the node, check this tutorial: [Link](https://docs.vulcanexus.org/en/latest/rst/microros_documentation/user_api/user_api_nodes.html)

Closing!

closed 2023-07-03 by Acuadros95