micro-ROS/micro_ros_arduino #1461: Support for Arduino Uno R4

james-yoo, 2023-07-18:
Are there any plans to support a new Arduino UNO R4 (Renesas RA4M1 Arm® Cortex®-M4)?

pablogs9, 2023-07-18:
Hello, there is no plan for integrating Arduino UNO R4, but it is easy to integrate new platforms in this repo, and community contributions are very welcomed.

patrickwasp, 2023-08-17:
hi @pablogs9 can you point me to any documentation for integrating new platforms? Thanks

pablogs9, 2023-08-18:
Hello, there is no official documentation for making new ports, but you can see some PR that have integrated new platforms:

- https://github.com/micro-ROS/micro_ros_arduino/pull/35
- https://github.com/micro-ROS/micro_ros_arduino/pull/1284
- https://github.com/micro-ROS/micro_ros_arduino/pull/680

In general, you will need to:
- Determine which compiler (version specific) uses Arduino for the target platform
- If not available in the building docker, add it [here](https://github.com/micro-ROS/docker/blob/b6c0c8fc7e6cfabb64746d57f5d25da43fb8880e/micro-ROS-static-library-builder/Dockerfile#L5-L27) and PR changes.
- Update the [library_generation.sh script](https://github.com/micro-ROS/micro_ros_arduino/blob/iron/extras/library_generation/library_generation.sh) to use the compiler to build the micro-ROS static library and put it in the correct folder
- Update CI to test your changes

In general, this is not a trivial process, but handling it with care should be possible to do it.

--- 

**IMPORTANT** 

It is very important to check if one of the already precompiled libraries actually works for the platform.

For example, R4 uses a Cortex-M4 and we already have a precompiled Cortex-M4 library: https://github.com/micro-ROS/micro_ros_arduino/tree/iron/src/cortex-m4

Have you check if adding it as [supported plaftorm](https://github.com/micro-ROS/micro_ros_arduino/blob/ea286412a524cb7c0eec5a2f6f92ae283b9bd0df/library.properties#L10) it works out of the box?

gbr1, 2024-03-21:
Hi! I have just checked this issue. I made a [PR](https://github.com/micro-ROS/micro_ros_arduino/pull/1712) to get Arduino UNO R4 working over USB

gbr1, 2024-11-14:
Now it is supported 😃

closed 2024-11-14 by james-yoo

anhdong2194, 2024-12-07:
> Now it is supported 😃

I have tested Micro-ROS on the Arduino UNO R4 WiFi, but it was unsuccessful. I tried uploading the example of a publisher node, and while it could build and load successfully, it automatically jumps into the error loop about a minute later. Could you please help me check this?

hippo5329, 2024-12-07:
@anhdong2194  It might be time out on micro-ros connection. You can try reset R4 and start the micro-ros agent immediately.

anhdong2194, 2024-12-07:
> @anhdong2194 It might be time out on micro-ros connection. You can try reset R4 and start the micro-ros agent immediately.

Thank you for your response. Yes, I tried resetting, but the result remains the same. I have tested with both Docker and micro_ros_agent built from source. Additionally, when debugging, I noticed that changing the baud rate from 115200 to 921600 ( change at "default_transport.cpp" ) causes the error_loop to hang at the publish node creation function. Below are the screenshots for baud rates 115200 and 921600.
[image]
[image]

gbr1, 2024-12-07:
About wifi, just to be sure, did you run the agent with wifi? Did you write the right IP address in your sketch?

hippo5329, 2024-12-07:
The Uno R4 main cpu core is 48MHz, and very limited 256 kB Flash, 32 kB RAM. I would not suggest this board. You should use pico, which has more resources for robotics.

anhdong2194, 2024-12-08:
> About wifi, just to be sure, did you run the agent with wifi? Did you write the right IP address in your sketch?

Yes, I also tried using WiFi, but it couldn't connect. After some time, it timed out, and the LED started blinking (entered the error_loop)
[image]
[image]
[image]

gbr1, 2025-01-03:
@anhdong2194 the IP into your sketch must be the one of your PC not the one of the Arduino board

WinetheLe, 2025-02-26:
@gbr1, hi, I'm having the same problem with Anhdong2194 when I'm trying to test the code. After troubleshooting, it seems the problem lies with the first initialisation. I placed a serial.print function between the RCCcheck and this one seems to have lead to it:

[image]

the serial stoped at the 0 and afterward the LED starts blinking so I think this might be it. though I'm inspecting the code to see what went wrong, as of right now I also dont know what's the cause.

gbr1, 2025-02-26:
@WinetheLe , you are right. I'm investingating on what could be. It seems to happen on USB, too. So probably it is something on how is managed the bytestream.

sdaerasd, 2025-04-22:
How to configure Micro-ROS on Arduino UNO R4 WiFi?