micro-ROS/micro_ros_arduino #1479: Cannot compile micro_ros_arduino on SAM board via arduino-cli running on aarch64 host

henriksod, 2023-07-26:
## Issue template

- Hardware description: Arduino DUE
- RTOS: N/A
- Installation type: arduino-cli
- Version or commit hash: micro_ros_arduino-2.0.7-humble

#### Steps to reproduce the issue

The SAM board is not officially supported on aarch64 systems.
The solution is to compile using a later version of arm-none-eabi-gcc https://github.com/arduino/arduino-cli/issues/1339

On a Raspberry PI 4 model B running 64-bit Raspberry PI OS:
1. Install arduino-cli
2. arduino-cli core update-index --additional-urls https://per1234.github.io/ArduinoCore-sam/package_per1234_samarm64_index.json
3. export ARDUINO_BOARD_MANAGER_ADDITIONAL_URLS=https://per1234.github.io/ArduinoCore-sam/package_per1234_samarm64_index.json
4. arduino-cli core install per1234:sam
5. curl https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/iron/extras/patching_boards/platform_arduinocore_sam.txt > ~/.arduino15/packages/per1234/hardware/sam/1.6.11-arm64/platform.txt
6. arduino-cli lib install --zip-path micro_ros_arduino-2.0.7-humble.zip
7. arduino-cli compile --fqbn per1234:sam micro_ros_publisher.ino -v

#### Expected behavior

Should be able to compile `micro_ros_arduino-2.0.7-humble` using `arm-none-eabi-gcc-7-2017q4`.

#### Actual behavior

Getting errors about `undefined reference to __ctype_ptr__` when compiling an example sketch `micro_ros_publisher` using arduino-cli.

I followed the steps in this ticket https://github.com/micro-ROS/micro_ros_arduino/issues/315, since this is the error I get after having updated the arm-none-eabi-gcc version to `7-2017q4` from `4.8.3-2014q1`

I am not able to compile a new `libmicroros.a` static library file due to errors such as:

```bash
--- stderr: sensor_msgs                                       
CMake Warning at /uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
```

I used the following command on a x86_64 host machine:
```bash
~/dev/micro_ros_arduino$ docker run -it --rm -v /home/henrik/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4:/toolchain -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:humble -p cortex_m3
```

pablogs9, 2023-07-26:
Hello @henriksod I see that you have a solution, could you open a PR with a proposal for fixing this issue?

henriksod, 2023-07-26:
Hi! Maybe I am confusing things here. The solution I mention is not related to micro_ros_arduino. 

I had two issues:
- I want to compile an arduino sketch towards an Arduino DUE using micro_ros_arduino on an aarch64 system. This is not officially supported by arduino since there is no packaged arm-none-eabi-gcc compiler as old as 4_8-2014q1.
- The micro_ros_arduino-2.0.7-humble comes with a compiled static library for cortex_m3 which is compiled using 4_8-2014q1. As a result of the solution solving the previous point, I need to update this to 7-2017q4.

henriksod, 2023-07-26:
***Update:***

So I ran the following command `strings src/cortex-m3/libmicroros.a | grep "GCC: ("` and got this:

```
GCC: (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
```

Seems like it is the correct gcc version used here. Maybe the error messages I got from running the docker container were misleading.

I will try to use this static library and come back,

closed 2023-07-26 by henriksod