micro-ROS/micro_ros_arduino #1490: Teensy 3.2 and 4.0 don't work with Arduino IDE 2.1.1

Russ76, 2023-08-02:
## Issue template

- Hardware description: 
- Ubuntu laptop HP Ryzen 7, Ros Humble
- RTOS: 
- Teensy 3.2 and 4.0
- Installation type: 
- Arduino IDE 2.1.1 with Teensyduino installed
- Version or commit hash: 
- Downloaded latest Microros zip file and loaded library into sketch

#### Steps to reproduce the issue

#### Expected behavior
That program would compile and upload to Teensy

#### Actual behavior
IDE complains of the precompiled libraries. I installed the platform.txt file under /home/.arduino15/packages/teensy/hardware/avr/1.58.1

I am unable to find any Arduino/hardware/sam folder.

WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/russ/Arduino/libraries/micro_ros_arduino/src/mk20dx256
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/russ/Arduino/libraries/micro_ros_arduino/src/mk20dx256/libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xc0): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/russ/Arduino/libraries/micro_ros_arduino/src/mk20dx256/libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x5e): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/russ/Arduino/libraries/micro_ros_arduino/src/mk20dx256/libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x36): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0xa6): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0x162): undefined reference to `__locale_ctype_ptr'
collect2: error: ld returned 1 exit status

#### Additional information

pablogs9, 2023-08-02:
This error is related to building the micro-ROS library with a different version of GCC. micro-ROS for teensy are built using with [this GCC](https://github.com/micro-ROS/docker/blob/b6c0c8fc7e6cfabb64746d57f5d25da43fb8880e/micro-ROS-static-library-builder/Dockerfile#L12) since https://github.com/micro-ROS/micro_ros_arduino/pull/1482.

Could you check:

- It your local GCC match the one used by micro-ROS?
- Does it works if you checkout a commit previous to #1282, for example d993b12433a511a772be246e7382bb610df99f1c ?

Russ76, 2023-08-02:
Thanks. My local GCC version is 11.3.0
I don't see the version number in the Micro-ROS page. 
I tried using the Micro-ROS Humble zip file from May, in the IDE, but it failed uploading with the same errors.

pablogs9, 2023-08-02:
We just updated the Teensy support recently: https://github.com/micro-ROS/micro_ros_arduino/pull/1482

Could you install a fresh Arduino + Teensy environment (latest versions), and recheck?

Russ76, 2023-08-03:
I downloaded the latest Arduino IDE, Teensyduino, and MicroROS but it still comes up with same error when trying to compile the program.

pablogs9, 2023-08-03:
I will take a look tomorrow if I have a slot. Will report here.

pablogs9, 2023-08-07:
I have just tested this locally and it works ok with the latest teensy package: https://www.pjrc.com/teensy/td_download.html

Russ76, 2023-08-08:
I did a complete new install and followed all the rules. I still get this error when trying to compile the program. Could it be because I have Humble installed, that the latest MicroROS is Iron? My Humble install has been updated, more than once. And this Teensy 4.0 loads the Blink program just file.

WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/russ/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/russ/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xbe): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/russ/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x5e): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/russ/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x46): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0xb6): undefined reference to `__locale_ctype_ptr'
/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0x164): undefined reference to `__locale_ctype_ptr'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

pablogs9, 2023-08-08:
Can you try a `/home/russ/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/gcc --version` and paste here the answer?

Russ76, 2023-08-08:
I tried a new computer to do it, too. New hardware, 4 core Celeron unit, new Jammy install, Humble with dev tools, MircoROS installed and it ran PingPong. Then Arduino IDE 2.1.1, Added the Udev rules, then Teensyduino, changed the hidden platform.txt file. Added Iron MicroROS library to a sketch, it loaded. Chose an example from the list, the MicroROS publisher. Tried to compile. Same error!

adityakamath, 2023-11-25:
Hi @Russ76, I have the same issue, with Arduino IDE 2.2.1 and trying to find a solution. Have you managed to fix this issue?

Russ76, 2023-11-26:
Yes, the problem vanished with updated versions. Be sure to use the newest versions, and copy over the text file. You will see the example code come up to choose from.

adityakamath, 2023-12-02:
Thanks! I installed updated versions for everything, but still had issues compiling on Windows. I compiled on WSL2 and it worked fine. Problem solved, for now at least :)

Russ76, 2023-12-04:
This issue is resolved.